espressif/esp-idf #2538: Hello I am trying to use the CMake tools being tested by espressif but when I follow the clone instructions "esp>git clone --branch feature/cmake --recursive https://github.com/espressif/esp-idf.git"  I get the following messages "Cloning into 'esp-idf'..."  and then "fatal: Remote branch feature/cmake not found in upstream origin" Can anyone help please?

embeddedjim, 2018-10-10:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Core (if using chip or module): [ESP32-Wrover-I|ESP32-Wrover|ESP32-PICO-D4|ESP32-SOLO-I|ESP32-WROOM-32U|ESP-WROOM-32D|ESP-WROOM32|ESP32]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:		[Arduino IDE|Make|Eclipse|other]
- Operating System:		[Windows|Ubuntu|MacOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

mahavirj, 2018-10-10:
@embeddedjim `feature/cmake` is already merged in `master` branch, please follow instructions at https://docs.espressif.com/projects/esp-idf/en/latest/get-started-cmake/index.html

mahavirj, 2018-10-10:
@negativekelvin Thanks for notifying, will fix them.

jcabad100, 2019-05-20:
Hello everyone, i still have the same problem and the documention https://docs.espressif.com/projects/esp-idf/en/v3.1/get-started-cmake/index.html#get-started-get-esp-idf-cmake still contains the very same instructions "git clone --recursive https://github.com/espressif/esp-idf.git" and i get the very same error as @embeddedjim i.e. "$ git clone --branch feature/cmake --recursive https://github.com/espressif/esp-idf.git
Cloning into 'esp-idf'... 
fatal: Remote branch feature/cmake not found in upstream origin" ...can someone  please tell me how to solve this issue? @mahavirj i don't know what to do with the information you provided "feature/cmake is already merged in master branch" ...can you be more explicit on how to clone the repository? ...thank you!

jcabad100, 2019-05-20:
@Alvin1Zhang can you please update the documentation on  https://docs.espressif.com/projects/esp-idf/en/v3.1/get-started-cmake/index.html#get-started-get-esp-idf-cmake  indicating how to clone the repository (i am working on Windows 10) and i am deadlocked with "fatal: Remote branch feature/cmake not found in upstream origin" ..thank you.

Alvin1Zhang, 2019-05-20:
@jcabad100 Thanks for the feedback. We will look into this. Thanks.

mahavirj, 2019-05-20:
@jcabad100 If you do not have any dependency on IDF v3.1 release then you can refer to latest release (v3.2) documentation at https://docs.espressif.com/projects/esp-idf/en/v3.2/get-started-cmake/index.html#step-2-get-esp-idf. For IDF v3.1, it seems like fix is not yet backported, will ensure that it will get addressed soon.

jcabad100, 2019-05-20:
@Alvin1Zhang ..i justi solved (on Windows) ,instead of "git clone --recursive https://github.com/espressif/esp-idf.git" i used "git clone --recursive https://github.com/espressif/esp-idf.git " ...and it started/finished cloning...

catherinedodge, 2019-06-28:
This bug still exists in the docs here: https://docs.espressif.com/projects/esp-idf/en/v3.1.3/get-started-cmake/index.html#get-started-get-esp-idf-cmake which are linked to from the Amazon FreeRTOS page. If someone knows where those docs are generated from, happy to submit a pull request to remove the remote branch argument.

krzychb, 2019-07-02:
Hi @catherinedodge thank you for pointing it out. I will get it resolved.

mahavirj, 2019-07-04:
Fixed with https://github.com/espressif/esp-idf/commit/4e25f20c829b32d4cff022fe8c76fc4073672380 (Reflected at https://docs.espressif.com/projects/esp-idf/en/release-v3.1/get-started-cmake/index.html, will also be part of `v3.1.5` documentation with next release)

closed 2019-07-04 by mahavirj