espressif/esp-idf #2539: Setting the BT Classic allowed packet types ignored

jakkra, 2018-10-10:
## Environment

- Development Kit:      [none]
- Core (if using chip or module): [ESP32]
- IDF version bdb65ed066b008fbd2c4a28806b397dce3b2f5c1 (release v3.1)
- Development Env:		[Make]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description
Setting the classic packet type does not seem to stick, i.e ignored by the controller. 

I create a classic connection and provide the HCI command with supported packet types like this:
[image]
Setting for example only DM1 as supported packet type, I still get 1mbps+ throughput, which should not be possible. I can also see by sniffing that other packet types are used.

I found a workaround, for example if I want DM1 only packets I can do the below:
What seems to work is to issue a hci_change_connection_packet_type (DM1 only) after the connection is established, however it's not enough, the connection must also be created with allowed packet type to DM1 only. 

To sum it up, connection must be created with DM1 as only allowed packet type AND after the connection is established, a hci_change_connection_packet_type (DM1 only) needs to be issued. Doing only one of the above is not enough.

### Expected Behavior
Connection uses one of the specified supported classic packet types provided to the create_connection HCI command.
### Actual Behavior
Controller seems to ignore what supported packet types I provide to create_connection command. Also the hci_change_connection_packet_type shall not be required. Setting allowed packet types in the create connection command shall be enough.
### Steps to repropduce
Create a classic connection by sending HCI_Create_Connection and set allowed packet types to DM1 only.
Sniff the connection/test throughput and see that other packet types than DM1 is used.

## Debug Logs

## Other items if possible
We are not using Bluedroid.

jakkra, 2018-10-19:
Found some more info.
Setting the packet_type parameter in HCI_Create_connection does not work as expected.
If I set for example packet_type to 0x0008 (DM1 only) the ESP32 should only send DM1 packets. But if I sniff what packets are used I see it's using 2-DH3 packets. If I set packet_type to allow all packets I normally get 3-DH3 or 3-DH5 packets (depending on size of data I send), meaning setting packet_type to 0x0008 does something, as afterwards 2-DH3 packets are used, instead of 3DH-3/5 as I usually get when allowing all types.

See spec:

When sending HCI
ACL Data Packets the Link Manager shall only use the packet type(s) specified
by the Packet_Type command parameter or the always-allowed DM1 packet
type. Multiple packet types may be specified for the Packet Type parameter by
performing a bit-wise OR operation of the different packet types. The Link
Manager may choose which packet type to be used from the list of acceptable
packet types.

blueMoodBHD, 2019-01-29:
Hi @ jakkra There may be some mistakes about `packet types`. In `Core_v4.2`, it is defined as follow:
[image]

So, if you want to supported  DM1 packet, you should set packet type as `0x330e` instead of `0x0008`. Please change it and test again.

Thanks.

jakkra, 2019-02-05:
@blueMoodBHD Thank you, sorry, my misstake. Works fine now.

closed 2019-02-05 by jakkra