espressif/esp-idf #2541: bluetooth heap corruption

chegewara, 2018-10-10:
@Weijian-Espressif @Yulong-espressif 
Im not sure if this is bt stack heap corruption or it is from my app. Testing environment:
- esp32 devKitC as a central client,
- 4 x esp32 as a peripheral server,
- all 4 peripheral are connecting successfully,
- after connection client writes message to every connected peripheral every 10s on peripheral connection basis (independent),
- when i "disconnect" one peripheral by hard reset i get this error:
```
CORRUPT HEAP: multi_heap.c:474 detected at 0x3ffdea24
abort() was called at PC 0x4008f369 on core 0
0x4008f369: multi_heap_assert at D:/msys32/home/imper/esp/esp-idf/components/heap/multi_heap.c:377
 (inlined by) multi_heap_free_impl at D:/msys32/home/imper/esp/esp-idf/components/heap/multi_heap.c:474

Backtrace: 0x4008f920:0x3ffcff90 0x4008faf1:0x3ffcffb0 0x4008f369:0x3ffcffd0 0x4008297a:0x3ffcfff0 0x40082edd:0x3ffd0010 0x4000bec7:0x3ffd0030 0x4014395e:0x3ffd0050 0x40143bdf:0x3ffd0080 0x401
4187e:0x3ffd00c0 0x40141cc2:0x3ffd00e0 0x4014a801:0x3ffd0100 0x40138a3c:0x3ffd0140 0x40138c72:0x3ffd0160 0x4008d849:0x3ffd0190
0x4008f920: invoke_abort at D:/msys32/home/imper/esp/esp-idf/components/esp32/panic.c:660

0x4008faf1: abort at D:/msys32/home/imper/esp/esp-idf/components/esp32/panic.c:660

0x4008f369: multi_heap_assert at D:/msys32/home/imper/esp/esp-idf/components/heap/multi_heap.c:377
 (inlined by) multi_heap_free_impl at D:/msys32/home/imper/esp/esp-idf/components/heap/multi_heap.c:474

0x4008297a: heap_caps_free at D:/msys32/home/imper/esp/esp-idf/components/heap/heap_caps.c:130

0x40082edd: _free_r at D:/msys32/home/imper/esp/esp-idf/components/newlib/syscalls.c:42

0x4014395e: gatt_cl_send_next_cmd_inq at D:/msys32/home/imper/esp/esp-idf/components/bt/bluedroid/stack/gatt/gatt_cl.c:1042

0x40143bdf: gatt_client_handle_server_rsp at D:/msys32/home/imper/esp/esp-idf/components/bt/bluedroid/stack/gatt/gatt_cl.c:1166

0x4014187e: gatt_data_process at D:/msys32/home/imper/esp/esp-idf/components/bt/bluedroid/stack/gatt/gatt_main.c:977

0x40141cc2: gatt_le_data_ind at D:/msys32/home/imper/esp/esp-idf/components/bt/bluedroid/stack/gatt/gatt_main.c:560

0x4014a801: l2c_rcv_acl_data at D:/msys32/home/imper/esp/esp-idf/components/bt/bluedroid/stack/l2cap/l2c_main.c:267

0x40138a3c: btu_hci_msg_process at D:/msys32/home/imper/esp/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:598

0x40138c72: btu_task_thread_handler at D:/msys32/home/imper/esp/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:598

0x4008d849: vPortTaskWrapper at D:/msys32/home/imper/esp/esp-idf/components/freertos/port.c:403
```
If this is my code related then i would like to apology for wasting your time.

Weijian-Espressif, 2018-10-11:
@chegewara what is the commit id of ESP-IDF ?

chegewara, 2018-10-11:
@Weijian-Espressif 
This exact log :
```
Merge: 1d58429ba b557cca6b
Author: Jiang Jiang Jian <jack@espressif.com>
Date:   Mon Oct 8 15:59:07 2018 +0800

    Merge branch 'bugfix/no_disconnect_event_report_if_no_correct_status_code_report' into 'master'

    WiFi: bugfix of no disconnect event report if no correct status code report

    See merge request idf/esp-idf!3370
```
```
I (62665) BT_GATT: GATTC_ConfigureMTU conn_id=261 mtu=100
I (62665) SampleClient: 6--> 15836
I (62665) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 30aea449a732
I (62685) BT_GATT: GATTC_ConfigureMTU conn_id=262 mtu=100
I (62675) BLEClient: esp_ble_gattc_get_service: 3 services found in cache
E (62675) BLEDevice:  gattc_if: 5 == 3
I (62695) BLEClient: Service: uuid: 00001801-0000-1000-8000-00805f9b34fb, start_handle: 1 0x1, end_handle: 5 0x5
I (62705) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 30aea449a732
CORRUPT HEAP: multi_heap.c:474 detected at 0x3ffddb04
abort() was called at PC 0x4008f349 on core 0
0x4008f349: multi_heap_assert at D:/msys32/home/imper/esp/esp-idf/components/heap/multi_heap.c:377
 (inlined by) multi_heap_free_impl at D:/msys32/home/imper/esp/esp-idf/components/heap/multi_heap.c:474

Backtrace: 0x4008f900:0x3ffd00f0 0x4008fad1:0x3ffd0110 0x4008f349:0x3ffd0130 0x4008297a:0x3ffd0150 0x40082edd:0x3ffd0170 0x4000bec7:0x3ffd0
190 0x4014391a:0x3ffd01b0 0x40143b9b:0x3ffd01e0 0x4014183a:0x3ffd0220 0x40141c7e:0x3ffd0240 0x4014a7bd:0x3ffd0260 0x401389f8:0x3ffd02a0 0x4
0138c2e:0x3ffd02c0 0x4008d829:0x3ffd02f0
0x4008f900: invoke_abort at D:/msys32/home/imper/esp/esp-idf/components/esp32/panic.c:660

0x4008fad1: abort at D:/msys32/home/imper/esp/esp-idf/components/esp32/panic.c:660

0x4008f349: multi_heap_assert at D:/msys32/home/imper/esp/esp-idf/components/heap/multi_heap.c:377
 (inlined by) multi_heap_free_impl at D:/msys32/home/imper/esp/esp-idf/components/heap/multi_heap.c:474

0x4008297a: heap_caps_free at D:/msys32/home/imper/esp/esp-idf/components/heap/heap_caps.c:130

0x40082edd: _free_r at D:/msys32/home/imper/esp/esp-idf/components/newlib/syscalls.c:42

0x4014391a: gatt_cl_send_next_cmd_inq at D:/msys32/home/imper/esp/esp-idf/components/bt/bluedroid/stack/gatt/gatt_cl.c:1042

0x40143b9b: gatt_client_handle_server_rsp at D:/msys32/home/imper/esp/esp-idf/components/bt/bluedroid/stack/gatt/gatt_cl.c:1166

0x4014183a: gatt_data_process at D:/msys32/home/imper/esp/esp-idf/components/bt/bluedroid/stack/gatt/gatt_main.c:977

0x40141c7e: gatt_le_data_ind at D:/msys32/home/imper/esp/esp-idf/components/bt/bluedroid/stack/gatt/gatt_main.c:560

0x4014a7bd: l2c_rcv_acl_data at D:/msys32/home/imper/esp/esp-idf/components/bt/bluedroid/stack/l2cap/l2c_main.c:267

0x401389f8: btu_hci_msg_process at D:/msys32/home/imper/esp/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:598

0x40138c2e: btu_task_thread_handler at D:/msys32/home/imper/esp/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:598

0x4008d829: vPortTaskWrapper at D:/msys32/home/imper/esp/esp-idf/components/freertos/port.c:403
```

Weijian-Espressif, 2018-10-30:
@chegewara Sorry, according to your test steps, we did not have reproduce the issue. here is the test log.
[BTSTK_GATT_50007_30143322_000.html.zip](https://github.com/espressif/esp-idf/files/2528323/BTSTK_GATT_50007_30143322_000.html.zip)

chegewara, 2018-10-30:
@Weijian-Espressif Thanks, i will close it for now and back to it when i have time to do more tests.

closed 2018-10-30 by chegewara