espressif/esp-idf #2543: bluetooth interface get affected and couldn't advertise again after stopping WIFI

NOA-Vernon, 2018-10-11:
Hi,
I enable both Bluetooth and WIFI interface at the same time and attempt to disable the WIFI interface as it did not need to use, however, when I successfully  disable the WIFI interface, the Bluetooth interface seems not work again, below is part of the log from UART, is there anyone happened to this issue? 

[0m
I (349573) wifi: state: run -> init (0)
I (349583) wifi: pm stop, total sleep time: 0 us / 56480465 us
I (349593) wifi: n:3 0, o:3 0, ap:255 255, sta:3 0, prof:1
I (349613) wifi: flush txq
I (349613) wifi: stop sw txq
I (349613) wifi: lmac stop hw txq
[0;32mI (349623) WIFI_IF: SYSTEM_EVENT_STA_DISCONNECTED[0m
[0;32mI (349623) WIFI_IF: esp_wifi_stop() succ[0m
I (349633) wifi: Deinit lldesc rx mblock:10
[0;32mI (349643) WIFI_IF: esp_wifi_deinit() succ[0m
[0;32mI (349653) GATTS_DEMO: ESP_GATTS_CONF_EVT, status 0[0m
[0;33mW (353553) BT_APPL: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0008[0m
[0;33mW (353553) BT_APPL: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0008[0m
[0;33mW (353553) BT_APPL: bta_gattc_conn_cback() - cif=5 connected=0 conn_id=5 reason=0x0008[0m
[0;33mW (353563) BT_APPL: bta_gattc_conn_cback() - cif=6 connected=0 conn_id=6 reason=0x0008[0m
[0;32mI (353573) GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT[0m
[0;32mI (371433) event: station ip lost[0m

TianHao-Espressif, 2018-10-11:
@NOA-Vernon , I'm sorry for this problem.  We will debug it. BTW, could you provide IDF version and the bt/wifi library version printed in logs? It may be helpful.

NOA-Vernon, 2018-10-17:
@TianHao-Espressif could you please provide the API of getting the IDF version and the bt/wifi lib version? thank you.

chegewara, 2018-10-17:
IDF version with `git show`
bt/wifi you can find when you restart esp32:
```
I (83) wifi: wifi driver task: 3ffbd8cc, prio:23, stack:3072, core=1
I (83) wifi: wifi firmware version: 75aab56
```

NOA-Vernon, 2018-10-18:
@chegewara , please see the following log that contains the bt/wifi version. and IDF is v3.1

[0;32mI (322) BTDM_INIT: BT controller compile version [44d04c1]
[0m
[0;32mI (322) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (642) phy: phy_version: 3910, c0c45a3, May 21 2018, 18:07:06, 0, 0

I (42732) wifi: wifi driver task: 3ffea4b0, prio:23, stack:4096, core=0
I (42742) wifi: wifi firmware version: a7a46e9
I (42742) wifi: config NVS flash: enabled
I (42752) wifi: config nano formating: disabled
[0;32mI (42752) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (42762) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (42792) wifi: Init dynamic tx buffer num: 32
I (42792) wifi: Init data frame dynamic rx buffer num: 32
I (42792) wifi: Init management frame dynamic rx buffer num: 32
I (42792) wifi: Init static tx buffer num: 16
I (42802) wifi: Init static rx buffer size: 1600
I (42802) wifi: Init static rx buffer num: 10
I (42802) wifi: Init dynamic rx buffer num: 32
I (42832) wifi: mode : sta (24:0a:c4:95:07:ac)
I (42842) wifi: Set ps type: 0

Wth-Esp, 2020-10-29:
@NOA-Vernon Please get the latest IDF and have a try, thanks.

closed 2021-09-14 by Alvin1Zhang