espressif/esp-idf #2545: Wifi driver alloc esf_buf failed when disconnect previous AP then connect to new AP

outermanle, 2018-10-11:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Core (if using chip or module): [ESP-WROOM-32D]
- IDF version (3276a1316): 
- Development Env:		[Make]
- Operating System:		[Ubuntu]
- Power Supply:         [USB]

## Problem Description
1. esp32 connected with AP;
2. then configure another AP info then call disconnect and reconnect the new AP.
3. WIFI driver error occurs.

### Expected Behavior
Normally disconnect and try to connect another AP.

### Actual Behavior
Wifi driver alloc buf failed.

### Steps to repropduce

1. esp32 connected with AP;
2. then configure another AP info then call disconnect and reconnect the new AP.
3. WIFI driver error occurs.

100% reproduce this issue.

get the same error log after change sdkconfig
```
-CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=10
-CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=32
+CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=20
+CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=64
-CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM=32
+CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM=64
```

## Debug Logs

```
E (820160) wifi: esf_buf: t=2 l=76 max:32, alloc:32 no eb
W (820170) wifi: alloc eb len=76 type=2 fail, heap:165824
W (820170) wifi: m f probe req l=0
E (820300) wifi: esf_buf: t=2 l=76 max:32, alloc:32 no eb
W (820300) wifi: alloc eb len=76 type=2 fail, heap:165840
W (820300) wifi: m f probe req l=0
E (820420) wifi: esf_buf: t=2 l=76 max:32, alloc:32 no eb
W (820420) wifi: alloc eb len=76 type=2 fail, heap:165840
W (820420) wifi: m f probe req l=0
E (820540) wifi: esf_buf: t=2 l=76 max:32, alloc:32 no eb
W (820540) wifi: alloc eb len=76 type=2 fail, heap:165840
W (820540) wifi: m f probe req l=0
E (820670) wifi: esf_buf: t=2 l=76 max:32, alloc:32 no eb
W (820670) wifi: alloc eb len=76 type=2 fail, heap:165840
W (820670) wifi: m f probe req l=0
E (820790) wifi: esf_buf: t=2 l=76 max:32, alloc:32 no eb
W (820790) wifi: alloc eb len=76 type=2 fail, heap:166036
W (820790) wifi: m f probe req l=0
```

outermanle, 2018-10-13:
Any suggestion?

outermanle, 2018-10-13:
I solve this issue. Call wifi api incorrectly may trigger wifi state machine to a wrong state.
Thanks!

MightyPork, 2018-10-17:
@outermanle could you post a code example before/after the fix, so e.g people finding this on google can find how to solve it?

outermanle, 2018-10-18:
@MightyPork 

In my case, I call  esp disable sniffer api before disconnect wifi, then error occurs, 100% reproduce it.
It works after remove disable sniffer api before disconnect wifi.

We should follow the wifi state machine designed by esp, document refer to https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/wifi.html#wi-fi-connect-phase

sometimes we call esp wifi api may trigger bug, then we may test other implement way or we call report bug with details.

closed 2018-10-18 by outermanle