espressif/esp-idf #2547: Wi-fi features stopped and wifi task constantly triggers the watchdog.

tmedicci, 2018-10-11:
## Environment

- Development Kit:      none
- Core (if using chip or module): ESP32-PICO-D4
- IDF version ESP-IDF v3.1-75-gfbcfc86
- Development Env:		Eclipse
- Operating System:		Ubuntu
- Power Supply:         external 3.3V

## Problem Description

Task watchdog being triggered for the first time after ~10h. 5s before task watchdog had triggered, it logged the following error:
E (58930960) wifi: ieee80211_scan.c 1395

Log output shows the following information:

Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 
 - IDLE (CPU 0)
 
Tasks currently running:
CPU 0: wifi
CPU 1: uart_rx_task

All CPU 0 being executed is wifi task. CPU1 task changes among other tasks on each log output. Application runs without wifi features and task watchdog trigger every 5 seconds. 

### Steps to reproduce

I could not reproduce this bug yet, but I am aware it could happen other times as my application should always be powered on.

liuzfesp, 2018-10-22:
Hi @tmedicci, it's our bug, we will debug it.

XinDeng11, 2018-10-22:
there are is a case may cause this issue, that is no beacon operation channel in the beacon or probe response, if it that, station can not judge which channel to follow, so we add assert here, if you want to reporduce the phenmenon, maybe you should find the router which do not have DS information in the beacon

XinDeng11, 2018-10-24:
@tmedicci we have do a fix of this serval days before, if you still remember the commit-id of you IDF, I can verify the root cause of your problem

tmedicci, 2018-10-29:
Hello @XinDeng11 , my IDF commit is:
```
commit fbcfc869e2339be33f5c7258f079aa862efef7a9
Merge: 4107361 bb48077
Author: Jiang Jiang Jian <jack@espressif.com>
Date:   Thu Sep 20 18:44:18 2018 +0800

    Merge branch 'bugfix/fix_the_bug_create_socket_fail_v3.1' into 'release/v3.1'
    
    lwip: fix the bug that failed to create socket (backport v3.1)
    
    See merge request idf/esp-idf!3311

```
Hope this can be helpful. Thank you!

liuzfesp, 2019-03-19:
Hi @tmedicci could you help to check whether this issue still exist in latest release/v3.1?

tmedicci, 2019-03-22:
Hi @liuzfesp , it never happened again. In fact, I have seen this bug only once. I think the issue could be closed. Thank you for your attention.

closed 2019-03-22 by Alvin1Zhang