espressif/esp-idf #2549: flash read err, 1000 when ESP_PD_DOMAIN_RTC_FAST_MEM is OFF

Stasikiii, 2018-10-11:
Wroom-32
Chip is ESP32D0WDQ6 (revision 1)

>rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
>flash read err, 1000
>ets_main.c 371
>ets Jun 8 2016 00:22:57

```cpp
#include "esp_deep_sleep.h"
#include <sys/time.h>
#include "time.h"
struct timeval tv;

void printLocalTime(){
  struct tm timeinfo;
  if(!getLocalTime(&timeinfo)){
    Serial.println("Failed to obtain time");
    return;
  }
  Serial.println(&timeinfo, "%A, %B %d %Y %H:%M:%S");
}
void setup(){
  Serial.begin(115200);
  delay(50);
  gettimeofday(&tv,NULL);
  if (tv.tv_sec < 1539178840){
    tv.tv_sec = 1539178840+7200;
    settimeofday(&tv, NULL);
  }
  
  esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
  esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_SLOW_MEM, ESP_PD_OPTION_OFF);
  esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_FAST_MEM, ESP_PD_OPTION_OFF);
  esp_sleep_enable_timer_wakeup(5000000);
  printLocalTime();
  //Serial.println(tv.tv_sec);
  delay(10);
  esp_deep_sleep_start();
}

void loop(){}
```
if esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_FAST_MEM, ESP_PD_OPTION_ON);
then it's ok.

It's a BUG or i just fon't understand something?

Stasikiii, 2018-10-12:
Many many thx @negativekelvin :+1:

Stasikiii, 2018-10-12:
I did ESP32_DEEP_SLEEP_WAKEUP_DELAY 5000
but still 
> rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> flash read err, 1000
> ets_main.c 371

Pls help!

negativekelvin, 2018-10-12:
ESP32_DEEP_SLEEP_WAKEUP_DELAY only works if ESP_PD_DOMAIN_RTC_FAST_MEM is ESP_PD_OPTION_ON

Your flash chip has 1800us startup time while for example winbond w25q has 20us

Stasikiii, 2018-10-13:
@negativekelvin 
Then i don't understand what is issue for **ESP_PD_DOMAIN_RTC_FAST_MEM is ESP_PD_OPTION_OFF**
and flash **read err, 1000** after deep sleep...
as i said in first message - no problem with **ESP_PD_DOMAIN_RTC_FAST_MEM is ESP_PD_OPTION_ON**

igrr, 2018-10-13:
The feature pointed out by @negativekelvin above (CONFIG_ESP32_DEEP_SLEEP_WAKEUP_DELAY) allows delaying startup, compensating for flash chip startup time. This feature works using deep sleep wake stub, which is placed into RTC Fast memory. If you force power down RTC Fast memory, then this wakeup stub code will be lost, and startup will not be delayed.

In general, there is little need to call esp_deep_sleep_pd_config function in the app, as the default behavior is to power down everything that is not really needed. e.g. if your application does not use RTC slow memory for data storage, this memory will be powered down.

Stasikiii, 2018-10-13:
@igrr 
Then, in my case i just need to change flash chip (on faster) and it's fix's my problem? I do not store either in fast or in slow memory nothing on how much I understand.
Or i'm wrong?
What is issue in my case then? Or i have no possibility to fix this?

igrr, 2018-10-13:
I think you need to remove the call which powers down RTC Fast memory. This will fix the problem, and difference in power consumption will be very small (less than 0.5uA).

Stasikiii, 2018-10-14:
thx @iggr but i think it's not issue for this time :(

closed 2018-10-14 by Stasikiii