espressif/esp-idf #2550: ESP32的flash超出导致软件复位问题

Freepom, 2018-10-12:
用的是最新的IDF，应用开发功能有WiFi，blufi，mqtt，现在是如果把blufi功能加进去，flash就会超出。

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6116
load:0x40078000,len:9404
load:0x40080400,len:7128
entry 0x400807a0
[0;32mI (56) boot: ESP-IDF v3.2-dev-1145-g66fd7e9cc-dirty 2nd stage bootloader[0m
[0;32mI (56) boot: compile time 15:11:25[0m
[0;32mI (58) boot: Enabling RNG early entropy source...[0m
[0;32mI (64) boot: SPI Speed      : 40MHz[0m
[0;32mI (68) boot: SPI Mode       : DIO[0m
[0;32mI (72) boot: SPI Flash Size : 4MB[0m
[0;32mI (76) boot: Partition Table:[0m
[0;32mI (79) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (87) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (94) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (102) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (109) boot: End of partition table[0m
[0;32mI (113) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2d010 (184336) map[0m
[0;32mI (187) esp_image: segment 1: paddr=0x0003d038 vaddr=0x3ffbdb60 size=0x02fd8 ( 12248) load[0m
[0;32mI (192) esp_image: segment 2: paddr=0x00040018 vaddr=0x400d0018 size=0xbaca0 (765088) map[0m
[0;32mI (461) esp_image: segment 3: paddr=0x000facc0 vaddr=0x3ffc0b38 size=0x00694 (  1684) load[0m
[0;32mI (462) esp_image: segment 4: paddr=0x000fb35c vaddr=0x3ffc11cc size=0x00000 (     0) load[0m
[0;32mI (468) esp_image: segment 5: paddr=0x000fb364 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (478) esp_image: segment 6: paddr=0x000fb76c vaddr=0x40080400 size=0x15c2c ( 89132) load[0m
[0;32mI (523) esp_image: segment 7: paddr=0x001113a0 vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (523) esp_image: segment 8: paddr=0x001113a8 vaddr=0x50000000 size=0x00000 (     0) load[0m
[0;31mE (530) esp_image: Image length 1053648 doesn't fit in partition length 1048576[0m
[0;31mE (538) boot: Factory app partition is not bootable[0m
[0;31mE (543) boot: No bootable app partitions in the partition table[0m
ets Jun  8 2016 00:22:57

这个问题是要怎么解决？

redchenjs, 2018-10-12:
修改partitions.csv中的factory分区大小

Alvin1Zhang, 2019-03-15:
Feel free to reopen. Thanks.

closed 2019-03-15 by Alvin1Zhang