espressif/esp-idf #2552: BLE: ESP_GATTS_CONF_EVT params lacks associated attribute handle

mws-rmain, 2018-10-12:
When the ESP_GATTS_CONF_EVT is triggered in a BLE server to confirm a client has received an 'indicate' update of the value of an attribute, the paramaters passed to the server (see below) do not include the 'handle' of the associated attribute. The handle is part of the BLE indicate ACK message, so clearly it is available when the event is created. Passing this information as part of the event parameters only makes sense.

```    /**
     * @brief ESP_GATTS_CONF_EVT
     */
    struct gatts_conf_evt_param {
        esp_gatt_status_t status;       /*!< Operation status */
        uint16_t conn_id;               /*!< Connection id */
        uint16_t len;                   /*!< The indication or notification value length, len is valid when send notification or indication failed */
        uint8_t *value;                 /*!< The indication or notification value , value is valid when send notification or indication failed */
    } conf;                             /*!< Gatt server callback param of ESP_GATTS_CONF_EVT (confirm) */
```
without the handle of the associated attribute, the server has no way to know WHICH attribute update is being confirmed.

I'm using Neil Kolban's fabulous 'esp32-snippets' c++ classes  to implement a BLE server. In his current implementation, on receipt of a 'ESP_GATTS_CONF_EVT' event, the server scans through all associated services, and characteristics, sending the event notification to every one of them (which is horribly inefficient), since it has no idea which attribute update is being confirmed. If the attribute handle was included in the event parameters, the server could directly send the event to the appropriate characteristic.

chegewara, 2018-10-12:
hi @mws-rmain 
could you test library from this repo?
https://github.com/chegewara/esp32-snippets-enchancements-test/tree/multi_connect_test
I have changed few things in code and that might fix it.

mws-rmain, 2018-10-12:
Thanks for the suggestion, @chegewara. I reviewed your changes - they appear to be directed toward multidevice connection, and don't address this issue, although this issue will CERTAINLY affect multi-device connections.

Your code hasn't changed the current handling of ESP_GATTS_CONF_EVT. Any confirmation will blindly clear the m_semaphoreConfEvt, for all attributes, and note that there is one ONE semaphore - not one per connection.

Weijian-Espressif, 2018-10-15:
@mws-rmain 👌, I will add attr handle for ESP_GATTS_CONF_EVT.

mws-rmain, 2018-10-15:
Thanks very much.

closed 2018-10-15 by mws-rmain

chegewara, 2018-11-08:
Hi @Weijian-Espressif it seems that conf event handle is not working:
`W (22205) handleGATTServerEvent: m_handle = 42, conf->handle = 0`
from this code:
`			ESP_LOGW(__func__, "m_handle = %d, conf->handle = %d", m_handle, param->conf.handle);
`