espressif/esp-idf #2555: make monitor in eclipse problem

snikul, 2018-10-13:
## Environment

- Development Kit:      NodeMCU-32S Lua WiFi IOT Development Board v1.1
- Core (if using chip or module): [ESP32S]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    221eced06
- Development Env:		[Eclipse]
- Operating System:		[Windows10]
- Power Supply:         [USB]

## Problem Description
I have just installed esp32_win32_msys2_environment_and_toolchain-20181001.zip for Windows (W10).

No errors with "make monitor" in bash/make environment. I can makeconfig, build, flash and monitor
In Eclipce: "make all" "make flash"targets finished without errors.
# python --version
Python 2.7.15
# git describe --tags --dirty
v3.2-dev-1250-g221eced06
PATH in Eclipse "Environment to set" C:\msys32\usr\bin;C:\msys32\mingw32\bin;C:\msys32\opt\xtensa-esp32-elf\bin
winpty winpty.dll ... in C:\msys32\usr\bin from the box.
and the result with the error. 
I tried hard reading forums with no success...

10:59:04 **** Build of configuration Default for project hello_world ****
python C:/msys32/home/Megauser/esp/esp-idf/tools/windows/eclipse_make.py -j2 monitor 
stdin is not a tty
make: *** [/home/Megauser/esp/esp-idf/components/esptool_py/Makefile.projbuild:106: monitor] Error 1
Running make in 'C:/msys32/home/Megauser/esp/hello_world'
including C:/msys32/home/Megauser/esp/esp-idf/components/app_update/Makefile.projbuild...
including C:/msys32/home/Megauser/esp/esp-idf/components/bootloader/Makefile.projbuild...
including C:/msys32/home/Megauser/esp/esp-idf/components/bootloader_support/Makefile.projbuild...
including C:/msys32/home/Megauser/esp/esp-idf/components/coap/Makefile.projbuild...
including C:/msys32/home/Megauser/esp/esp-idf/components/esp32/Makefile.projbuild...
including C:/msys32/home/Megauser/esp/esp-idf/components/esptool_py/Makefile.projbuild...
including C:/msys32/home/Megauser/esp/esp-idf/components/mbedtls/Makefile.projbuild...
including C:/msys32/home/Megauser/esp/esp-idf/components/nghttp/Makefile.projbuild...
including C:/msys32/home/Megauser/esp/esp-idf/components/partition_table/Makefile.projbuild...
including C:/msys32/home/Megauser/esp/esp-idf/components/ulp/Makefile.projbuild...
Toolchain path: C:/msys32/opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
python C:/msys32/home/Megauser/esp/esp-idf/tools/check_python_dependencies.py
Python requirements from C:/msys32/home/Megauser/esp/esp-idf/requirements.txt are satisfied.
[ -f C:/msys32/home/Megauser/esp/hello_world/build/hello-world.elf ] || echo "*** 'make monitor' target requires an app to be compiled and flashed first."
[ -f C:/msys32/home/Megauser/esp/hello_world/build/hello-world.elf ] || echo "*** Run 'make flash monitor' to build, flash and monitor"
[ -f C:/msys32/home/Megauser/esp/hello_world/build/hello-world.elf ] || echo "*** Or alternatively 'make simple_monitor' to view the serial port as-is."
[ -f C:/msys32/home/Megauser/esp/hello_world/build/hello-world.elf ] || exit 1
winpty python C:/msys32/home/Megauser/esp/esp-idf/tools/idf_monitor.py --baud 115200 --port "COM7" --toolchain-prefix "xtensa-esp32-elf-" --make "make" --print_filter "" C:/msys32/home/Megauser/esp/hello_world/build/hello-world.elf

11:06:15 Build Failed. 1 errors, 0 warnings. (took 7m:11s.88ms)

### Expected Behavior
I expect hello_world response in eclipse console window.
But when I copy/paste
 winpty python C:/msys32/home/Megauser/esp/esp-idf/tools/idf_monitor.py --baud 115200 --port "COM7" --toolchain-prefix "xtensa-esp32-elf-" --make "make" --print_filter "" C:/msys32/home/Megauser/esp/hello_world/build/hello-world.elf
to bash make env window and run then that command I get the hello_world response from myboard ok.
### Actual Behavior
no hello_world response from the board. Stops with the 106 error/

igrr, 2018-10-13:
Eclipse build is not running from a tty, so starting "monitor" there is not possible. Since "monitor" is an interactive application, it needs some kind of console...
You can change the build target to `make flash`, which is able to run without a tty attached. This target will build your app and upload it to the board. Then you can either open a separate console window and run `make monitor` there, or open one within Eclipse.

snikul, 2018-10-13:
So, you mean I cannot use "make flash monitor" or "make monitor"in Eclipse?I have to open MinGW32 console and run
 winpty python C:/msys32/home/Megauser/esp/esp-idf/tools/idf_monitor.py --baud 115200 --port "COM7" --toolchain-prefix "xtensa-esp32-elf-" --make "make" --print_filter "" C:/msys32/home/Megauser/esp/hello_world/build/hello-world.elf

Is that what you meant when wrote "you can either open a separate console window and run make monitor there". But how to evoke a console within Eclipse as you said?

snikul, 2018-10-13:
I guess I got it! Open Teminal -->Serial Terminal. And I got the output of my dev. board Thank you, Игорь!

closed 2018-10-14 by Alvin1Zhang

asmitajha, 2019-03-18:
Hi
I was able to build and flash successfully using eclipse IDE. But the problem is that when I checked with blinking LED example, the LED was not blinking on the board. I also tried by changing its GPIO in menuconfig. But when i flashed using msys command prompt using same examples and menuconfig the led was blinking on the board. So i want to ask that why is it not blinking using when flashed with eclipse?﻿