espressif/esp-idf #2560: DAC header isn't self contained

devsaurus, 2018-10-14:
## Environment

- Development Kit:      [none]
- Core (if using chip or module): [ESP32-Wrover]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 7313e39fd
- Development Env:		[Make]
- Operating System:		[openSuse]
- Power Supply:         [USB]

## Problem Description

The `driver/dac.h` header uses definitions for gpio but doesn't include `driver/gpio.h`.

### Expected Behavior

The code below compiles without errors.

### Actual Behavior

```
sdk/esp32-esp-idf/components/driver/include/driver/dac.h:43:53: error: unknown type name 'gpio_num_t'
 esp_err_t dac_pad_get_io_num(dac_channel_t channel, gpio_num_t *gpio_num);
```

### Steps to repropduce

Compile the code below.

### Code to reproduce this issue

```c
#include "driver/dac.h"

void myfunction(int chn) {
  dac_output_enable( chn );
}
```

## Other items if possible

As a workaround I have to include `driver/gpio.h` before the dac header file even though my source file doesn't use GPIO API.

koobest, 2018-10-15:
hi, @devsaurus, thank you, we will update the dac.h.

xiongyumail, 2018-11-27:
Is there any progress?

pilnikov, 2019-06-25:
Autors, please add new line 
#include "gpio.h"
and correct line 42
 *   - ESP_ERR_INVALID_ARG if channal not valid 
on
 *   - ESP_ERR_INVALID_ARG if chann**_e_**l not valid

Nufflee, 2020-02-15:
Any progress on this? Seems like a very simple change.

Alvin1Zhang, 2021-09-14:
Thanks for reporting, sorry for slow turnaround, fix is available https://github.com/espressif/esp-idf/commit/83573189454e806be83627d6f4059c22591bf184, feel free to reopen.

closed 2021-09-14 by Alvin1Zhang