espressif/esp-idf #2562: ESP32 Simple OTA Example Error.

gyaneshsingh, 2018-10-14:
Hello All, 
I am trying to test ESP32 Simple OTA update example given in "espressif/esp-idf/example/systems/ota/simple_ota_example". on run I am getting following error:

I (333) simple_ota_example: Setting WiFi configuration SSID Gyanesh...
I (413) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 0
I (413) wifi: mode : sta (30:ae:a4:4a:fb:c8)
I (413) simple_ota_example: Starting OTA example...
I (1743) wifi: n:11 0, o:1 0, ap:255 255, sta:11 0, prof:1
I (2733) wifi: state: init -> auth (b0)
I (2733) wifi: state: auth -> assoc (0)
I (2733) wifi: state: assoc -> run (10)
I (2953) wifi: connected with Gyanesh, channel 11
I (2953) wifi: pm start, type: 1

I (3763) event: sta ip: 192.168.0.107, mask: 255.255.255.0, gw: 192.168.0.1
I (3763) simple_ota_example: Connect to Wifi ! Start to Connect to Server....
E (22023) esp-tls: Failed to connnect to host (errno 113)
E (22023) TRANS_SSL: Failed to open a new connection
E (22023) HTTP_CLIENT: Connection failed
E (22023) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (22033) simple_ota_example: Firmware Upgrades Failed

I guess server is not started correctly as I tried to "ping 192.168.0.3" (I am not sure if it is correct way to test) and getting error "destination host not reachable". Although I used command 
"openssl s_server -WWW -key ca_key.pem -cert ca_cert.pem -port 8070"

It would be great if someone helps me to locate the problem and resolve it.

Thanks in advance.
Gyanesh

jitin17, 2018-10-17:
@gyaneshsingh It seems your are not setting `firmware upgrade url endpoint` because `192.168.0.3` is the default value, so can you ensure whether the IP address of the host on which you are setting up the server, by running the command `openssl s_server -WWW -key ca_key.pem -cert ca_cert.pem -port 8070`, is `192.168.0.3`. If it is not `192.168.0.3`, then change it to the actual IP under `make menuconfig` -> `Example Configuration` -> `firmware upgrade url endpoint`

Alvin1Zhang, 2018-10-24:
@gyaneshsingh Could you help share if any updates for this issue? Thanks.

mahavirj, 2018-11-20:
@gyaneshsingh If there are any followup questions please ask on esp32.com forum.

closed 2018-11-20 by mahavirj