espressif/esp-idf #2563: issue with search_cmpl.searched_service_source parsing

chegewara, 2018-10-14:
@Weijian-Espressif 
I have implemented logic where if i cant find cache data then im getting GATT attributes from peer device, when i have more than 2 then get data from cache (its because false reading from cache, not always 0 even after make erase_flash). Here is my logic:
```
	uint16_t count = 10;
	esp_gattc_service_elem_t *result = (esp_gattc_service_elem_t*)calloc(count, sizeof(esp_gattc_service_elem_t));
	esp_err_t errRc;
	clearServices(); // Clear any services that may exist.
	errRc = esp_ble_gattc_get_service(getGattcIf(), getConnId(), NULL, result, &count, 0);
	// ESP_LOGI(LOG_TAG, "esp_ble_gattc_get_service: %d services found in cache", count);
#ifdef CONFIG_GATTC_CACHE_NVS_FLASH
	if(count < 13) {  // <-- for testing purpose, to confirm issue, it should be count == 0, or at most count < 3 to update even if generic services are still in cache
#else
	if(true) {  // <-- if we dont have turned on caching data, then always read from peer device
#endif
	ESP_LOGI(LOG_TAG, "esp_ble_gattc_get_service: %d, getting services from peer device", count);
		errRc = esp_ble_gattc_search_service(
			getGattcIf(),
			getConnId(),
			NULL            // Filter UUID
		);
	
		m_semaphoreSearchCmplEvt.take("getServices");
		if (errRc != ESP_OK) {
			ESP_LOGE(LOG_TAG, "esp_ble_gattc_search_service: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
			return &m_servicesMap;
		}
		// If sucessfull, remember that we now have services.
		m_haveServices = (m_semaphoreSearchCmplEvt.wait("getServices") == 0);
	}else{
		ESP_LOGI(LOG_TAG, "esp_ble_gattc_get_service: %d, getting services from cache", count);

		// result++;
		// result++;
		for(int i=0; i<count;i++){
			esp_gattc_service_elem_t srv = *result;
			esp_gatt_id_t id = {
				.uuid = srv.uuid,
				.inst_id = 0
			};
			BLEUUID uuid = BLEUUID(srv.uuid);
			BLERemoteService* pRemoteService = new BLERemoteService(
				id,
				this,
				srv.start_handle,
				srv.end_handle
			);
			// while(m_servicesMapByInstID.find(pRemoteService) >= 0){
			// 	id.inst_id++;
			// }
			m_servicesMap.insert(std::pair<std::string, BLERemoteService *>(uuid.toString(), pRemoteService));
			// m_servicesMapByInstID.insert(std::pair<BLERemoteService *, uint16_t>(pRemoteService, id.inst_id));
			result++;
			ESP_LOGI(LOG_TAG, "%s", pRemoteService->toString().c_str());
		}
	}
```
To get logs basically i copy/paste this event:
https://github.com/espressif/esp-idf/blob/594dcd525ba745a399849e5df0ea858c21a6836f/examples/bluetooth/gatt_client/main/gattc_demo.c#L153-L159

Here is my log:
```
I (866) BLEClient: esp_ble_gattc_get_service: 2, getting services from peer device  <--- we can see that my logic works, and GATT attributes should be read from peer device
I (874) BLEClient: Get service information from flash  // <--- here we can see that ESP_GATTC_SEARCH_CMPL_EVT is reporting reading data from flash, and in this place my code just stuck

//peer device that did not delete service also is affected
I (2458) BLEClient: esp_ble_gattc_get_service: 3 services from peer device
I (2469) BLEClient: Get service information from flash
```

I have to mention one very important thing. Everything works as smooth as silk until peer device does not delete service and report it to esp32. In such case even after resetting esp32 central this issue persist.

Weijian-Espressif, 2018-10-15:
@chegewara After Bluetooth connection, gattc will find peer service information from flash, If it does not exist, read peer service information from remote device.

chegewara, 2018-10-15:
One more time:
1. `esp_ble_gattc_search_service` should always search services from remote device:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/bluetooth/esp_gattc.html#_CPPv228esp_ble_gattc_search_service13esp_gatt_if_t8uint16_tP13esp_bt_uuid_t
2. in normal usage it does, after getting from remote device indication about changing services configuration (delete service on peer device), `esp_ble_gattc_search_service` is fired, but event is `ESP_GATT_SERVICE_FROM_NVS_FLASH`, and my app stop running, maybe is confused just like i am now because is expecting to get services from remote device and bt stack is telling NO, from flash

Weijian-Espressif, 2018-10-15:
@chegewara  when BLE connection , gattc will load service information from flash. If information does not exist in flash, gattc read service information from remote device, then build service cache. `esp_ble_gattc_search_service `  get service information from service cache.

chegewara, 2018-10-15:
Then documentation is outdated and misleading. Because it clearly states that `esp_ble_gattc_get_service` is to get services from cache and `esp_ble_gattc_search_service` to retrieve from remote device:
> Find all the service with the given service uuid in the gattc cache, if the svc_uuid is NULL, find all the service. Note: It just get service from local cache, won’t get from remote devices. If want to get it from remote device, need to used the esp_ble_gattc_search_service.

https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/bluetooth/esp_gattc.html#_CPPv225esp_ble_gattc_get_service13esp_gatt_if_t8uint16_tP13esp_bt_uuid_tP24esp_gattc_service_elem_tP8uint16_t8uint16_t

Weijian-Espressif, 2018-10-15:
@chegewara Sorry, we will modify the documentation.

chegewara, 2018-10-15:
@Weijian-Espressif No problem. In such case i have some questions. Why do we have 2 function that are getting data from cache. 

Clearly i have to call `esp_ble_gattc_cache_refresh` when i receive service change event. Thanks.

Weijian-Espressif, 2018-10-15:
@chegewara 
1.  `esp_ble_gattc_get_service` just get service from local cache, won't get from remote devices. 
`esp_ble_gattc_search_service` can get service from local cache or get from remote devices
2.  I test service change, here is the log:
I (807) SEC_GATTC_DEMO: Stop scan successfully
I (917) SEC_GATTC_DEMO: EVT 40, gattc if 3
I (917) SEC_GATTC_DEMO: EVT 2, gattc if 3
I (917) SEC_GATTC_DEMO: open success
I (917) SEC_GATTC_DEMO: REMOTE BDA:
I (917) SEC_GATTC_DEMO: 66 70 37 d7 7b 06 
I (997) SEC_GATTC_DEMO: EVT 18, gattc if 3
I (997) SEC_GATTC_DEMO: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 200, conn_id 0
I (997) SEC_GATTC_DEMO: EVT 6, gattc if 3
I (997) SEC_GATTC_DEMO: **Get service information from flash**
E (5787) BT_APPL: bta_gattc_start_discover conn_id=3 p_clcb->p_srcb->state = 0 
I (5797) SEC_GATTC_DEMO: EVT 15, gattc if 3
I (5797) SEC_GATTC_DEMO: **ESP_GATTC_SRVC_CHG_EVT, bd_addr:**
I (5797) SEC_GATTC_DEMO: 66 70 37 d7 7b 06 
I (7387) SEC_GATTC_DEMO: EVT 6, gattc if 3
I (7387) SEC_GATTC_DEMO: **Get service information from remote device**

chegewara, 2018-10-15:
@Weijian-Espressif I dont want to open new issue and this seems to be related. This is my log from first time searching for services, it is searched from remote device:
```
D (16910) BLEUtils: GATT Event: ESP_GATTC_CONNECT_EVT
D (16910) BLEUtils: [conn_id: 0, remote_bda: 30:ae:a4:43:a1:72]
D (16913) BLEUtils: GATT Event: ESP_GATTC_OPEN_EVT
D (16917) BLEUtils: [status: ESP_GATT_OK, conn_id: 0, remote_bda: 30:ae:a4:43:a1:72, mtu: 23]
D (21285) BLEUtils: Received a GAP event: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT
D (21286) BLEUtils: [status: 0, bd_addr: 30:ae:a4:43:a1:72, min_int: 6, max_int: 6, latency: 0, conn_int: 6, timeout: 600]
D (21352) BLEUtils: GATT Event: ESP_GATTC_SEARCH_RES_EVT
D (21353) BLEUtils: [conn_id: 0, start_handle: 1 0x01, end_handle: 5 0x05, srvc_id: uuid: 00001801-0000-1000-8000-00805f9b34fb, inst_id: 1
D (21361) BLEUtils: GATT Event: ESP_GATTC_SEARCH_RES_EVT
D (21365) BLEUtils: [conn_id: 0, start_handle: 20 0x14, end_handle: 28 0x1c, srvc_id: uuid: 00001800-0000-1000-8000-00805f9b34fb, inst_id: 20
D (21379) BLEUtils: GATT Event: ESP_GATTC_SEARCH_RES_EVT
D (21383) BLEUtils: [conn_id: 0, start_handle: 40 0x28, end_handle: 65535 0xffff, srvc_id: uuid: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, inst_id: 40
D (21397) BLEUtils: GATT Event: ESP_GATTC_SEARCH_CMPL_EVT
D (21402) BLEUtils: [status: ESP_GATT_OK, conn_id: 0]
I (21407) BLEClient: Get service information from remote device
D (21414) BLEUtils: GATT Event: ESP_GATTC_CFG_MTU_EVT
D (21426) BLEUtils: GATT Event: ESP_GATTC_REG_FOR_NOTIFY_EVT
D (21426) BLEUtils: [status: ESP_GATT_OK, handle: 42 0x2a]
D (21436) BLEUtils: GATT Event: ESP_GATTC_WRITE_DESCR_EVT
D (21508) BLEUtils: Received a GAP event: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT
D (21509) BLEUtils: [status: 0, bd_addr: 30:ae:a4:43:a1:72, min_int: 125, max_int: 125, latency: 2, conn_int: 125, timeout: 1000]
D (21666) BLEUtils: GATT Event: ESP_GATTC_WRITE_CHAR_EVT
D (21667) BLEUtils: [status: ESP_GATT_OK, conn_id: 0, handle: 42 0x2a, offset: 0]
D (22440) BLEUtils: GATT Event: ESP_GATTC_WRITE_CHAR_EVT
D (22441) BLEUtils: [status: ESP_GATT_OK, conn_id: 0, handle: 42 0x2a, offset: 0]
D (23440) BLEUtils: GATT Event: ESP_GATTC_WRITE_CHAR_EVT
D (23440) BLEUtils: [status: ESP_GATT_OK, conn_id: 0, handle: 42 0x2a, offset: 0]
D (24440) BLEUtils: GATT Event: ESP_GATTC_WRITE_CHAR_EVT
```
and when i reset both devices then my app stuck after this event:
```
D (9574) BLEUtils: GATT Event: ESP_GATTC_CONNECT_EVT
D (9574) BLEUtils: [conn_id: 0, remote_bda: 30:ae:a4:43:a1:72]
D (9576) BLEUtils: GATT Event: ESP_GATTC_OPEN_EVT
D (9579) BLEUtils: [status: ESP_GATT_OK, conn_id: 0, remote_bda: 30:ae:a4:43:a1:72, mtu: 23]
D (9590) BLEUtils: GATT Event: ESP_GATTC_SEARCH_RES_EVT
D (9592) BLEUtils: [conn_id: 0, start_handle: 1 0x01, end_handle: 5 0x05, srvc_id: uuid: 00001801-0000-1000-8000-00805f9b34fb, inst_id: 1
D (9606) BLEUtils: GATT Event: ESP_GATTC_SEARCH_RES_EVT
D (9610) BLEUtils: [conn_id: 0, start_handle: 20 0x14, end_handle: 28 0x1c, srvc_id: uuid: 00001800-0000-1000-8000-00805f9b34fb, inst_id: 20
D (9624) BLEUtils: GATT Event: ESP_GATTC_SEARCH_RES_EVT
D (9628) BLEUtils: [conn_id: 0, start_handle: 40 0x28, end_handle: 65535 0xffff, srvc_id: uuid: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, inst_id: 40
D (9642) BLEUtils: GATT Event: ESP_GATTC_SEARCH_CMPL_EVT
D (9647) BLEUtils: [status: ESP_GATT_OK, conn_id: 0]
I (9652) BLEClient: Get service information from flash
D (10502) BLEUtils: GATT Event: ESP_GATTC_CFG_MTU_EVT
```
Then when i reset peripheral device, central disconnecting it with such logs:
```
W (290962) BT_APPL: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0008
E (290964) BT_OSI: osi_mem_dbg_clean invalid

D (290965) BLEUtils: GATT Event: ESP_GATTC_DISCONNECT_EVT
D (290970) BLEUtils: [reason: ESP_GATT_CONN_TIMEOUT, conn_id: 0, remote_bda: 30:ae:a4:43:a1:72]
I (290979) BLEScan: erase device: 30:ae:a4:43:a1:72
D (290985) BLEUtils: GATT Event: ESP_GATTC_CLOSE_EVT
D (290990) BLEUtils: [status: ESP_GATT_OK, reason:ESP_GATT_CONN_TIMEOUT, conn_id: 0]
I (290998) BLEDevice: remove: 3, GATT role client
E (291003) BT_APPL: bta_gattc_deregister Deregister Failedm unknown client cif
E (291013) BLERemoteService: esp_ble_gattc_get_all_char: ESP_GATT_NOT_FOUND
```
I know and understand what means all messages, only i cant understand is why my app stuck after or on `ESP_GATTC_CFG_MTU_EVT`. 

After writing this i think i may have some issue with semaphores in library. Will update when check it.

chegewara, 2018-10-15:
@Weijian-Espressif Maybe its nothing but in second logs i see its missing conn_update event after open event:
```
D (9574) BLEUtils: GATT Event: ESP_GATTC_CONNECT_EVT
D (9574) BLEUtils: [conn_id: 0, remote_bda: 30:ae:a4:43:a1:72]
D (9576) BLEUtils: GATT Event: ESP_GATTC_OPEN_EVT
D (9579) BLEUtils: [status: ESP_GATT_OK, conn_id: 0, remote_bda: 30:ae:a4:43:a1:72, mtu: 23]
D (9590) BLEUtils: GATT Event: ESP_GATTC_SEARCH_RES_EVT
D (9592) BLEUtils: [conn_id: 0, start_handle: 1 0x01, end_handle: 5 0x05, srvc_id: uuid: 00001801-0000-1000-8000-00805f9b34fb, inst_id: 1
D (9606) BLEUtils: GATT Event: ESP_GATTC_SEARCH_RES_EVT
```

Weijian-Espressif, 2018-10-16:
@chegewara , IDF will update connection params when reading  service information from remote device. if service information load from flash, ESP32 do not update connection params.

Weijian-Espressif, 2018-11-14:
@chegewara  Do you have any question?

Alvin1Zhang, 2019-03-21:
@chegewara Hi, would you help share if any updates for the issue? Thanks.

closed 2019-03-21 by chegewara