espressif/esp-idf #2565: undefined reference to flock, flockfile

siara-cc, 2018-10-15:
## Environment
- Core: ESP-WROOM32
- IDF version: v3.1
- Development Env:  Arduino IDE, make(esp-idf)
- Operating System: MacOS sierra 10.12.6
- Power Supply: USB

## Problem Description
Unable to use functions such as flock, flockfile.  There is no compilation error, but a linker error is thrown such as "undefined reference to `flockfile' collect2: error: ld returned 1 exit status"

### Expected Behaviour
Links without error and binary file generated and file locking possible.

### Actual Behaviour
See problem description. Same behaviour is seen in both Arduino IDE and using make with ESP-IDF 3.1 installed.

### Steps to reproduce
Paste following code in Arduino IDE New sketch with any ESP32 board selected.

### Code to reproduce this issue

```cpp
#include <stdio.h>
#include <fcntl.h>
#include "FS.h"
#include "SD_MMC.h"

void setup() {

    Serial.begin(115200);
    if (!SD_MMC.begin()){
        Serial.println("Card Mount Failed");
        return;
    }

    FILE *fp = fopen("/sdcard/test.txt", "r+");
    flock(fileno(fp), LOCK_EX);
    flock(fileno(fp), LOCK_UN);
    fclose(fp);

}

void loop() {
}
```

siara-cc, 2018-10-15:
Does fcntl() work for file locking?  It seems to link without problems.

igrr, 2018-10-16:
On desktop platforms, flock creates a lock which works between processes. Here in ESP-IDF we have only one process, so an implentation of flock would be a no-op.

flockfile, on the other hand, can be used to lock between threads (tasks in FreeRTOS), and is certainly useful. It is not exposed at the moment but we will certainly provide this definition.

fcntl locking is also between processes, as far as I understand, so it would also be a no-op in ESP-IDF.

luanmm, 2019-07-01:
It is something that would be very useful for a project I'm working on. We may expect it to be done soon?

closed 2024-05-25 by espressif-bot