espressif/esp-idf #2567: pinnedToCore() arg xCoreID should be renamed to xCoreNum

MightyPork, 2018-10-15:
Example: https://github.com/espressif/esp-idf/blob/master/components/freertos/tasks.c#L766

The API uses xCoreID, but core ID is defined in `xtensa_context.h` as 

```
#define CORE_ID_PRO 0xCDCD
#define CORE_ID_APP 0xABAB
```

these do not work e.g. in `xTaskCreatePinnedToCore`, ending with assert.

There are no correct defines for task IDs besides `tskNO_AFFINITY`.

Either

1) The API functions should support the CORE_ID defines in the `xCoreID` argument, or
2) The argument should be renamed to `xCoreNum` or similar, if it's not a core ID

projectgus, 2019-07-31:
Hi @MightyPork ,

Sorry noone got back to you about this for such an unacceptably long time. That's not anywhere near good enough.

We will add some clarification in the API documentation about the acceptable values for "Core ID" values, and also add more comments in the xtensa_context header about how the core ID CPU register values are used.

The reason we won't make either of the suggested fixes is that Core ID (0,1) is used in other widely used APIs too, such as the xPortGetCoreID() macro. We could rename these but it's a massive undertaking and potential disruption for IDF users when nearly all of IDF uses "Core ID" to me 0,1 or 0,1,tskNoAffinity. The exception is the xtensa_context header which has macros for the "raw" register values.

Similarly, we could change the xTaskCreatePinnedToCore() function to accept the register values but this only delays the confusion for a user if the xPortGetCoreID() macro returns a different value later on, and this is harder to detect when debugging.

negativekelvin, 2019-07-31:
@projectgus what about changing
```
#define CORE_ID_PRO 0xCDCD
#define CORE_ID_APP 0xABAB
```
To
```
#define CORE_INTERNALID_PRO 0xCDCD
#define CORE_INTERNALID_APP 0xABAB
```

MightyPork, 2019-07-31:
Another fix I can think of, is to introduce an enum for this "numeric core ID", something like the example below. I believe this will keep binary compatibility and you can still use plain numbers or int.

```
enum esp_core_id {
  ESP_CORE_ANY = -1,
  ESP_CORE_0 = 0,
  ESP_CORE_1 = 1
}
```

closed 2019-08-21 by igrr