espressif/esp-idf #2571: Limit interdependencies

X-Ryl669, 2018-10-16:
In order to get faster build time, and smaller binaries, I've spot many changes to the core file that could save this.

First, the minimal component list (found in your main Makefile) I'm trying to build is:
```
COMPONENTS=freertos esp32 soc heap newlib esptool_py nvs_flash spi_flash log tcpip_adapter lwip main xtensa-debug-module driver bt vfs micro-ecc nghttp wpa_supplicant smartconfig_ack app_trace mbedtls libesphttpd esp_http_client esp-tls pthread cxx
```

Doing so, leads to compilation errors in those files:
```
esp-idf/components/esp32/clk.c:34
  The include for bootloader_clock.h is not required (commenting it solves the error)

esp-idf/components/esp32/cpu_start.c:66
  The include for esp_efuse.h is not required (commenting it solves the error)

esp-idf/components/esp32/event_default_handlers.c:25
  Ethernet should be conditional
(to be continued)
```

Those are inter-module dependencies. Looking at the compilation line, there is no define whatsoever that tell one module what are the other module configuration (or presence). There isn't any shared header either (like sdkconfig.h or something).
I can add `CXXFLAGS := -DSOMETHING` and `CFLAGS:= -DSOMETHING` in my makefile, but it would be a PITA to maintain.

Do you think it would be possible to add in the general makefile rules that each component add a `-D_componentname` to the `CFLAGS` and `CXXFLAGS` to all modules ?
If so, it would be possible to remove/simplify inter component dependencies in the code.

projectgus, 2020-04-30:
With the CMake build system we now use in ESP-IDF, this is achievable but in a differet way. It's possible to declare the minimal list of components that your project uses in the CMakeLists file ,and then the build system will only build these components plus their (recursive) dependencies.

For more information:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#component-requirements
especially
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#including-components-in-the-build

If anyone is trying to build a minimial project with CMake and encountering problems getting the build size down, please open a new issue with the details.

closed 2020-04-30 by projectgus

X-Ryl669, 2020-04-30:
Well, this is a very old issue. In my PR, I've done more thing to actually modify the component themselves to deal with minimal dependencies. Typically, if you don't need Ethernet (most of us don't), then, **in addition to not building ethernet component**, my changes also modified the LWIP, event loop, wifi components to never refer to ethernet interface anywhere, so the code size could be really reduced.

At the time, I did this by making each component defining a `_DECL_component_name` macro when they are present and acting accordingly in other components based on the macros.
I'm not sure this can be done automatically in CMake (I haven't found how, and honestly, our project is too advanced now to switch build system and port the changes). 

Typically, my solution is a surgical solution cutting fat everywhere by little touch **in addition** to what you describe, a nuclear warhead based solution.

I guess, in the future or if you bother to hire someone to do that, a "size" optimization work will be required for esp-idf. It's not obvious there is an immediate benefit for this (compared to always adding new features and fixing bugs), but really there is one for all of us. In my *small* PR, I was able to cut build time by 40%, reduce code binary size by at least 10%. Multiply this by all your user's day work waiting for their CI to finish testing.