espressif/esp-idf #2575: HTTP server over SSL

MightyPork, 2018-10-16:
I have a working patch for the http_server component to make it work with HTTPS (OpenSSL) or any other layer protocol.

I write this instead of a PR, to ask - do you want it?

I need to clean it a bit, so if there's an official version in the works, I won't waste time doing that.
But otherwise it can be here in a few days.

AadiMehta, 2018-10-16:
Hi @MightyPork, can you please upload it to github and paste a link here?

Just a question what is your turnaround time for an https request for your server? In openssl server example, I am getting turnaround time per request to be around 5.5 seconds which is too much as compared to that of an http request(of around 100ms).

MightyPork, 2018-10-16:
@AadiMehta I'll upload it if nobody from espressif responds in 2 days, don't want to do the work twice.

anyway, I have 2.3 seconds RTT when the session is established, running at 240 MHz. I also changed the FreeRTOS clock rate to 1000Hz, but that shouldn't make a difference here.

but keep in mind that the session is not closed immediately when using the http server - same goes for SSL in my implementation. So follow-up requests (favicon etc) are faster, for me it's 500ms - may be even lower if the debug log is disabled - the server spams a lot in LOGD. I'll need to look into that. For my use this is good enough, it'll be just for some config pages.

MightyPork, 2018-10-16:
update - it's 1.8s for the first response, then 50-200 ms for the follow up messages. I think it's very good, considering I use TLSv1.2 (and suspect the OpenSSL library may not be using hardware acceleration properly)

AadiMehta, 2018-10-16:
That's quite a huge difference, I would definitely want to give it a try. Also wondering for what maximum time can we keep the session open?

Also do you happen to try https server implementation using mbedtls, if that has lower turnaround time

AadiMehta, 2018-10-17:
Okay, waiting now to try it. Thanks again for all the info

anurag-kar, 2018-10-17:
@MightyPork That seems promising! We are presently not looking into HTTP over SSL, but it's a desirable addition. So, please proceed with your PR.

MightyPork, 2018-10-17:
@anurag-kar the PR is submitted as https://github.com/espressif/esp-idf/pull/2578

it includes comments and an example, feel free to adjust as needed. moving discussion to the PR

closed 2018-10-17 by MightyPork

AadiMehta, 2018-10-17:
@MightyPork I think its missing "https_server.h" header file.

AadiMehta, 2018-10-17:
My bad! found it @MightyPork

MightyPork, 2018-10-17:
@AadiMehta no it's there. Try building the example, it should work as is (after you menuconfig your wifi)
https://github.com/espressif/esp-idf/pull/2578/files#diff-8f852df8ae12ea60a6f2e0d00fce9205