espressif/esp-idf #2579: Pin 33 related Interrupt/GPI

hvishu, 2018-10-17:
In our project, we have configured Pin 33 as interrupt with "GPIO_INTR_NEGEDGE" type.  It's working fine.  What I would like to understand is that, can I read this pin as GPIO after configured for interrupt?

What's happening is during the ESD testing sometimes we are seeing the ISR getting occurred, but when we read this pin using gpio_get_level(), it's not low.  This kind of helps, as we don't want the interrupt to occur caused by the ESD.

A couple of questions on this
1. I feel that the voltage levels for triggering the NEGEDGE interrupt and GPIO low value could be different thresholds?  Is my understanding correct?  Otherwise how above behavior can be explained.
2. Also, is it ok to read a pin using gpio_get_level() when configured as an interrupt?

stickbreaker, 2018-10-17:
1. Signal was low but returned high.  After the low level was detected by the hardware, the resulting interrupt was dispatched, But, this takes time.  During this finite period 2 to 10 microseconds, the signal went High. You will need an oscilloscope to see this pulse behavior.

2. yes. reading the level will not interfere with the interrupt. 
 
Chuck.

hvishu, 2018-10-21:
Thankyou @stickbreaker

closed 2018-10-21 by hvishu