espressif/esp-idf #2580: mdns sometimes will crash during network change

outermanle, 2018-10-17:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Core (if using chip or module): [ESP-WROOM-32D]
- IDF version (3276a1316): 
- Development Env:		[Make]
- Operating System:		[Ubuntu]
- Power Supply:         [USB]

## Problem Description
mdsn crash during network changing.
mdns.c may has a lot of variable need to check during network changing.

Wifi connected then become disconnected.
1. esp32 connected with AP;
2. then configure another AP info then call disconnect and reconnect the new AP.

or 

1. change station mode to softap+sta mode.

### Expected Behavior
Normally works.

### Actual Behavior
mdns will crash sometime.

### Steps to repropduce

Wifi connected then become disconnected.
1. esp32 connected with AP;
2. then configure another AP info then call disconnect and reconnect the new AP.

or 

1. change station mode to softap+sta mode.

20% reproduce this issue.

test code refer to examples/protocols/mdns/main/mdns_example_main.c

## Debug Logs

wifi disconnect and connect another AP.
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400db2df  PS      : 0x00060c30  A0      : 0x800db7ac  A1      : 0x3ffc10b0  
0x400db2df: _mdns_append_answer at /home/xx/esp-idf/components/mdns/mdns.c:4137

A2      : 0x3ffc1110  A3      : 0x003ffd18  A4      : 0x00000000  A5      : 0x3ffc1110  
A6      : 0x00000000  A7      : 0x0000001c  A8      : 0x8008b76c  A9      : 0x3ffc10d0  
A10     : 0x00000003  A11     : 0x00060223  A12     : 0x00060220  A13     : 0x00000001  
A14     : 0x0000cdcd  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x003ffd1c  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  
Backtrace: 0x400db2df:0x3ffc10b0 0x400db7a9:0x3ffc1100 0x400dd2ed:0x3ffc1140
0x400db2df: _mdns_append_answer at /home/xx/esp-idf/components/mdns/mdns.c:4137
0x400db7a9: _mdns_dispatch_tx_packet at /home/xx/esp-idf/components/mdns/mdns.c:4137
0x400dd2ed: _mdns_tx_handle_packet at /home/xx/esp-idf/components/mdns/mdns.c:4137
 (inlined by) _mdns_execute_action at /home/xx/esp-idf/components/mdns/mdns.c:3758
 (inlined by) _mdns_service_task at /home/xx/esp-idf/components/mdns/mdns.c:3863
```

wifi station mode --> sta+softap mode
```
Backtrace: 0x4008e6b4:0x3ffc4430 0x4008e8ed:0x3ffc4450 0x400df6af:0x3ffc4470 0x40149adf:0x3ffc44a0 0x40149c9e:0x3ffc44c0 0x400df245:0x3ffc44e0 0x400df36c:0x3ffc4520 0x400df3c9:0x3ffc4540 0x4013c1f8:0x3ffc4560
0x4008e6b4: invoke_abort at /home/xx/esp-idf/components/esp32/panic.c:660
0x4008e8ed: abort at /home/xx/esp-idf/components/esp32/panic.c:660
0x400df6af: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
0x40149adf: igmp_lookup_group at /home/xx/esp-idf/components/lwip/lwip/src/core/ipv4/igmp.c:268 (discriminator 1)
0x40149c9e: igmp_joingroup_netif at /home/xx/esp-idf/components/lwip/lwip/src/core/ipv4/igmp.c:498
0x400df245: _udp_join_group at /home/xx/esp-idf/components/mdns/mdns_networking.c:76
0x400df36c: _udp_pcb_init at /home/xx/esp-idf/components/mdns/mdns_networking.c:217
0x400df3c9: _mdns_pcb_init_api at /home/xx/esp-idf/components/mdns/mdns_networking.c:245
0x4013c1f8: tcpip_thread at /home/xx/esp-idf/components/lwip/lwip/src/api/tcpip.c:483
```

david-cermak, 2018-10-18:
Hi @outermanle,

I can reproduce the crashes only when running `mdns_free()` and `mdns_init()` between wifi reconnection. Please let me know if this is also your case?

Thanks,
David

outermanle, 2018-10-19:
Hi @david-cermak ,

It is not my case, ```mdns_init``` is called just one time when system bootup and I never call ```mdns_free```.
I put ```mdns_handle_system_event(ctx, event)``` in the end of ``wifi_event_handler`` refer to examples/protocols/mdns/main/mdns_example_main.c

Error just occurs during network changing, from the code of mdns.c, I thinks there is something to do to protect variable access when netif property changes.

For msg queue used in mdns.c, this is async work to access netif,  async process need to process mutex protect and null check.

AadiMehta, 2018-10-19:
@david-cermak @outermanle Mdns crashes for me also when wifi mode is changed from STA to AP. I have thus used workaround of deleting my mdns task and recreating it when wifi mode is STA again.

outermanle, 2018-10-19:
@AadiMehta I use both in sta and softap mode, even in sta+softap mode.

@david-cermak 
The reproduce probability is about 10-20%.

I just register a http service and do nothing.
Error occurs in the situation of network packets processing in the lwip while network interface property change.

initial one time no matter network changes.
```c
static void initialize_mdns(void)
{
    ESP_ERROR_CHECK( mdns_init() );
    /* set mDNS hostname */
    ESP_ERROR_CHECK( mdns_hostname_set(CONFIG_MDNS_NAME) );
    /* add http web service */
    ESP_ERROR_CHECK( mdns_service_add(NULL, "_http", "_tcp", 80, NULL, 0) );
}
```
register callback in wifi event handler

```c
static esp_err_t event_handler(void *ctx, system_event_t *event)
{
....

    case SYSTEM_EVENT_STA_CONNECTED:
        /* enable ipv6 */
        tcpip_adapter_create_ip6_linklocal(TCPIP_ADAPTER_IF_STA);
        break;
....
    case SYSTEM_EVENT_STA_DISCONNECTED:
        /* This is a workaround as ESP32 WiFi libs don't currently
           auto-reassociate. */
        if (auto_reconnect) {
            esp_wifi_connect();
        }
        xEventGroupClearBits(wifi_event_group, IP4_CONNECTED_BIT | IP6_CONNECTED_BIT);
        break;
    default:
        break;
    }
    mdns_handle_system_event(ctx, event);
    return ESP_OK;
}

```

outermanle, 2018-10-21:
Hi @david-cermak,

I only call 4 API of the mdns as above.
Wifi driver initialize with WIFI_INIT_CONFIG_DEFAULT.
Configure and connect to a non-exist AP and try 5 times then restore the previous one.

david-cermak, 2018-10-23:
Hi @outermanle,

with some luck (+delaying some threads on purpose) I was able to recreate the issue you describe. I am posting a preliminary fix here, so you can check if it solves your crashes as well.
These fixes should be eventually applied to IDF, but it may take some time (reviews, internal pipeline...)  Thank you for taking the time to report the issue!

[0001-mdns-crash-on-net-change-fix-sketch.txt](https://github.com/espressif/esp-idf/files/2505604/0001-mdns-crash-on-net-change-fix-sketch.txt)

outermanle, 2018-10-23:
hi @david-cermak , 

Thanks a lot! 
I will test it soon and give feedback later.

outermanle, 2018-10-31:
hi @david-cermak , so far so good. Thanks again.

commit 097282a8 (2018-11-01): mdns: fixed crashes on network changes

1) two events AP_STOP, AP_START shortly after each other may cause IGMP config on already stopped netif
2) not properly locked sending packets to queue from timer task

closes #2580
closed 2018-10-31 by outermanle