espressif/esp-idf #2583: Actual version does not compile

Reiner1210, 2018-10-18:
v3.2-dev-1350-g18684f53e

/home/reiner/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:33:38: error: unknown type name 'httpd_req_t'
 static esp_err_t common_post_handler(httpd_req_t *req)
                                      ^
/home/reiner/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c: In function 'protocomm_httpd_add_endpoint':
/home/reiner/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:138:30: error: implicit declaration of function 'strlen' [-Werror=implicit-function-declaration]
     char* ep_uri = calloc(1, strlen(ep_name) + 2);

igrr, 2018-10-19:
Are you seeing this with one of the examples, or with a custom app? Which build system are you using (Make/CMake)? If this is a custom app, can you offer the list of components being compiled (make list-components for Make build system, or printed at the start of compilation by CMake)? Attaching  full build output (make V=1 or idf.py -v build) would also help.

Reiner1210, 2018-10-19:
Hello,

Custom app. make under Linux. It works with v3.2-dev-1133-g4d50427e8
make list-components:
<code>
Toolchain path: /home/reiner/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
*******************************************************************************
COMPONENT_DIRS (components searched for here)
/home/reiner/esp/time-switch/components
src
/home/reiner/esp/mycomponents
/home/reiner/esp/esp-idf/components
/home/reiner/esp/time-switch/main
*******************************************************************************
TEST_COMPONENTS (list of test component names)

*******************************************************************************
TEST_EXCLUDE_COMPONENTS (list of test excluded names)
/home/reiner/esp/esp-idf/make/project.mk:570: Warnung: undefinierte Variable „EXCLUDE_COMPONENTS“
/home/reiner/esp/esp-idf/make/project.mk:570: Warnung: undefinierte Variable „EXCLUDE_COMPONENTS“
 
*******************************************************************************
COMPONENT_PATHS (paths to all components):
/home/reiner/esp/esp-idf/components/app_trace
/home/reiner/esp/esp-idf/components/app_update
/home/reiner/esp/esp-idf/components/asio
/home/reiner/esp/esp-idf/components/aws_iot
/home/reiner/esp/esp-idf/components/bootloader
/home/reiner/esp/esp-idf/components/bootloader_support
/home/reiner/esp/esp-idf/components/bt
/home/reiner/esp/esp-idf/components/coap
/home/reiner/esp/mycomponents/components
/home/reiner/esp/esp-idf/components/console
/home/reiner/esp/esp-idf/components/cxx
/home/reiner/esp/esp-idf/components/driver
/home/reiner/esp/esp-idf/components/esp-tls
/home/reiner/esp/esp-idf/components/esp32
/home/reiner/esp/esp-idf/components/esp_adc_cal
/home/reiner/esp/esp-idf/components/esp_http_client
/home/reiner/esp/esp-idf/components/esp_https_ota
/home/reiner/esp/esp-idf/components/esp_ringbuf
/home/reiner/esp/esp-idf/components/esptool_py
/home/reiner/esp/esp-idf/components/ethernet
/home/reiner/esp/esp-idf/components/expat
/home/reiner/esp/esp-idf/components/fatfs
/home/reiner/esp/esp-idf/components/freertos
/home/reiner/esp/esp-idf/components/heap
/home/reiner/esp/esp-idf/components/http_server
/home/reiner/esp/esp-idf/components/idf_test
/home/reiner/esp/esp-idf/components/jsmn
/home/reiner/esp/esp-idf/components/json
/home/reiner/esp/esp-idf/components/libsodium
/home/reiner/esp/esp-idf/components/log
/home/reiner/esp/esp-idf/components/lwip
/home/reiner/esp/esp-idf/components/mbedtls
/home/reiner/esp/esp-idf/components/mdns
/home/reiner/esp/esp-idf/components/micro-ecc
/home/reiner/esp/esp-idf/components/mqtt
/home/reiner/esp/esp-idf/components/newlib
/home/reiner/esp/esp-idf/components/nghttp
/home/reiner/esp/esp-idf/components/nvs_flash
/home/reiner/esp/esp-idf/components/openssl
/home/reiner/esp/esp-idf/components/partition_table
/home/reiner/esp/esp-idf/components/protobuf-c
/home/reiner/esp/esp-idf/components/protocomm
/home/reiner/esp/esp-idf/components/pthread
/home/reiner/esp/esp-idf/components/sdmmc
/home/reiner/esp/esp-idf/components/smartconfig_ack
[... 367 lines cut ...]
                    ^
/home/reiner/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c: In function 'protocomm_httpd_stop':
/home/reiner/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:241:9: error: unknown type name 'httpd_handle_t'
         httpd_handle_t *server_handle = (httpd_handle_t *) pc_httpd->priv;
         ^
/home/reiner/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:241:42: error: 'httpd_handle_t' undeclared (first use in this function)
         httpd_handle_t *server_handle = (httpd_handle_t *) pc_httpd->priv;
                                          ^
/home/reiner/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:241:58: error: expected expression before ')' token
         httpd_handle_t *server_handle = (httpd_handle_t *) pc_httpd->priv;
                                                          ^
/home/reiner/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:242:9: error: implicit declaration of function 'httpd_stop' [-Werror=implicit-function-declaration]
         httpd_stop(*server_handle);
         ^
/home/reiner/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c: At top level:
/home/reiner/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:29:17: warning: 'session_id' defined but not used [-Wunused-variable]
 static uint32_t session_id = PROTOCOMM_NO_SESSION_ID;
                 ^
cc1: some warnings being treated as errors
/home/reiner/esp/esp-idf/make/component_wrapper.mk:285: recipe for target 'src/transports/protocomm_httpd.o' failed
make[1]: *** [src/transports/protocomm_httpd.o] Error 1
make[1]: Verzeichnis „/home/reiner/esp/time-switch/build/protocomm“ wird verlassen
/home/reiner/esp/esp-idf/make/project.mk:505: recipe for target 'component-protocomm-build' failed
make: *** [component-protocomm-build] Error 2
</code>

Reiner1210, 2018-10-19:
Ok. I got it.
I have a own HTTP server which includes a my file named http_server.h - this clashes with your ....
Why do you add stuff I already have :-)

closed 2018-10-19 by Reiner1210