espressif/esp-idf #2585: esp_aes_crypt_ecb() fails

Rki009, 2018-10-18:
## Environment

- Development Kit:      none
- Kit version (for WroverKit/PicoKit/DevKitC): ?
- Core (if using chip or module): ESP-WROOM32
- IDF Version = v3.2-dev-1055-g3276a1316 
- Development Env:		Arduino IDE
- Operating System:		Windows
- Power Supply:         USB

## Problem Description

Attached is a simple Sketch that tests the esp_aes_crypt_ecb() function. A block is encrypted using the same key 1,000,000 times. Then decrypted 1,000,00 times. The result should be the same as the initial data.

When run without WiFi enabled, the test always passes. When run with WiFi enabled the test fails!

There is obviously an issue with sharing the AES hardware between WiFi and the test routine. In the esp_aes_crypt_ecb() routine the is a spinlock that should prevent conflict.

//  esp-idf/components/esp32/hwcrypto/aes.c
int esp_aes_crypt_ecb(...) {
    esp_aes_acquire_hardware();		<= should prevent any conflict
    esp_aes_setkey_hardware(ctx, mode);
    esp_aes_block(input, output);
    esp_aes_release_hardware();
    return 0;
}

Note: The code has been run on two different ESP32-NodeMCU modules; ESP-WROOM-32 and ESP-32S

### Expected Behavior
Data  : 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 50
Encode: da 1a c1 fa 6f 48 87 f8 58 36 21 3f 45 f7 64 4a
Decode: 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 50
All is OK!, iter=1000000
Loops = 133, Faults = 0

### Actual Behavior
Data  : 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 50
Encode: 8d 02 ae 93 98 34 16 a6 05 d0 b4 63 c8 60 9b b6
Decode: 4e c9 f0 1b 85 31 a2 71 84 cd 06 70 e8 7a 0e 30
***Failed!, iter=1000000
Loops = 10, Faults = 10

### Steps to reproduce
Run the Sketch.

### Code to reproduce this issue

```cpp
#include <WiFi.h>
#include <hwcrypto/aes.h>

#define ENABLE_WIFI

void dumpBuf(char* tag, uint8_t data[16]) {
	printf("%s:", tag);
	for(int i=0; i<16; ++i) {
		printf(" %02x", data[i]);
	}
	printf("\n");
}

bool faultTest(void) {
	uint8_t key[32];
	uint8_t buf[16];
	uint8_t buf2[16];
	uint8_t nbits = 128;
	uint8_t data[16];
	memcpy(data, "ABCDEFGHIJKLMNOP", 16);
	memcpy(key, "0123456789abcdef", 16);
	dumpBuf("Data  ", data);

	int iter = 1000000;
	esp_aes_context ctx;
	esp_aes_setkey(&ctx, key, nbits);

	memcpy(buf, data, 16);
	for(int i=0; i<iter; ++i) {
		esp_aes_crypt_ecb(&ctx, ESP_AES_ENCRYPT, buf, buf);
	}
	dumpBuf("Encode", buf);

	for(int i=0; i<iter; ++i) {
		esp_aes_crypt_ecb(&ctx, ESP_AES_DECRYPT, buf, buf);
	}
	dumpBuf("Decode", buf);

	if(memcmp(buf, data, 16)!=0) {
		printf("***Failed!, iter=%d\n", iter);
        return false;
	}
	printf("All is OK!, iter=%d\n", iter);
	return true;
}

//===============================================================
//      Setup
//===============================================================
const char* ssid     = "SSID";
const char* password = "PassWord";

void setup() {
	Serial.begin(115200);
	delay(500);
	printf("\nAES hardware test\n");

	// getChipInfo();
	printf("IDF Version = %s\n", esp_get_idf_version());

#ifdef ENABLE_WIFI
	WiFi.begin(ssid, password);
	for(int i=0; i<25; ++i) {   // timeout to connection to WiFi
		if(WiFi.status() == WL_CONNECTED) {
			break;
		}
		delay(500);
		Serial.print(".");
	}
	Serial.println();
	if(WiFi.status() == WL_CONNECTED) {
		IPAddress ip = WiFi.localIP();
		String ipStr = (String)ip.toString();
		printf("WiFi Ip Address: %s\n", ipStr.c_str());
	}
#endif  // ENABLE_WIFI

}

void loop(void) {
    int faults = 0;
    for(int n=1; ; ++n) {
        printf("----------\n");
        if (faultTest()==false) ++faults;
        printf("Loops = %d, Faults = %d\n", n, faults);
    }
}
```

## Debug Logs
None.

## Other items if possible
None.

projectgus, 2018-10-19:
Thanks for reporting this and providing these details.

This seems like it may somehow be a bug when using the Arduino IDE with arduino-esp32.

Using espressif/arduino-esp32@85032b226 and IDF commit 3276a1316:

* Building this code into an IDF app using arduino-esp32 as a component (and same [sdkconfig as Arduino](https://github.com/espressif/arduino-esp32/commit/96822d783f3ab6a56a69b227ba4d1a1a36c66268)) seems to not reproduce the problem. I got up to `Loops = 30, Faults = 0` on two different boards with WiFi enabled. I also tried building with Release optimization, same result.
* However, I can immediately reproduce the failures when I build the same code inside the Arduino IDE. Disabling WiFi removes the failures, as described in the issue.

We'll continue looking into this.

projectgus, 2021-02-09:
Sorry for the very long delay between updates on this one. The Arduino IDE v1.0.4 release (from 2019, current stable release while v1.0.5 is in RC) fixes this issue, I can't reproduce any more (Loops = 30, Failures = 0).

I think probably this was related to some of the "DPORT issues" fixed in ESP-IDF v3.3 (issue 3.10 in the "ESP32 ECO and Workarounds for Bugs" document). The older Arduino IDE core wouldn't have had the same fixes.

closed 2021-02-09 by projectgus