espressif/esp-idf #2587: 关于AP模式下死机复位的问题.

logozouxun, 2018-10-19:
使用DEMO程序:esp-homekit-sdk\esp-idf\examples\wifi\smart_config smartconfig_main.c,配置下面两个函数:
static void initialise_wifi(void)
{
    tcpip_adapter_init();
    wifi_event_group = xEventGroupCreate();
    ESP_ERROR_CHECK( esp_event_loop_init(event_handler, NULL) );

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();

    ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_AP) );//modify
    ESP_ERROR_CHECK( esp_wifi_start() );
}

static void sc_callback(smartconfig_status_t status, void *pdata)
{
    switch (status) {
        case SC_STATUS_WAIT:
            ESP_LOGI(TAG, "SC_STATUS_WAIT");
            break;
        case SC_STATUS_FIND_CHANNEL:
            ESP_LOGI(TAG, "SC_STATUS_FINDING_CHANNEL");
            break;
        case SC_STATUS_GETTING_SSID_PSWD:
            ESP_LOGI(TAG, "SC_STATUS_GETTING_SSID_PSWD");
            break;
        case SC_STATUS_LINK:
            ESP_LOGI(TAG, "SC_STATUS_LINK");
            wifi_config_t *wifi_config = pdata;
            ESP_LOGI(TAG, "SSID:%s", wifi_config->sta.ssid);
            ESP_LOGI(TAG, "PASSWORD:%s", wifi_config->sta.password);
            ESP_ERROR_CHECK( esp_wifi_disconnect() );
            ESP_ERROR_CHECK( esp_wifi_set_config(ESP_IF_WIFI_AP, wifi_config) ); //modify
            ESP_ERROR_CHECK( esp_wifi_connect() );
            break;
        case SC_STATUS_LINK_OVER:
            ESP_LOGI(TAG, "SC_STATUS_LINK_OVER");
            if (pdata != NULL) {
                uint8_t phone_ip[4] = { 0 };
                memcpy(phone_ip, (uint8_t* )pdata, 4);
                ESP_LOGI(TAG, "Phone ip: %d.%d.%d.%d\n", phone_ip[0], phone_ip[1], phone_ip[2], phone_ip[3]);
            }
            xEventGroupSetBits(wifi_event_group, ESPTOUCH_DONE_BIT);
            break;
        default:
            break;
    }
}
发现当前DEMO程序在路由器多的地方会出现不断的死机复位，在路由器少的地方会正常运行.

liuzfesp, 2018-10-22:
@logozouxun 这个问题好复现么，请帮忙提供一下下面的信息：
1. 复位时的 log
2. 复位时编译的 bin 对应该的 .elf 文件

logozouxun, 2018-10-23:
liu,以下是日志和bin文件.elf文件.
[ssession.log](https://github.com/espressif/esp-idf/files/2505739/ssession.log)

文件用7z压缩的，解压缩时，请使用7Z工具解压,build.zip文件名改成build.7z进行解压。
[build.zip](https://github.com/espressif/esp-idf/files/2505765/build.zip)

logozouxun, 2018-10-26:
liuzfesp,怎么样了，发现问题了吗？

logozouxun, 2018-12-10:
解决的怎么样了，现在大部分产品都是在AP模式下，路由器比较多的地方出现死机。

TimXia, 2018-12-11:
@logozouxun 请勿在 AP 模式下使用 esp-touch 配网，否则会出现意想不到的情况。

liuzfesp, 2018-12-11:
@logozouxun 你 build.zip 中给的是 *.elf 文件吗？ 我解压出来之后是build， 使用 objdump 读取的时候提示无法识别： xtensa-esp32-elf-objdump: build: File format not recognized。

logozouxun, 2018-12-11:
我用7.zip压缩的，请用7.zip解压

发自我的小米手机

logozouxun, 2018-12-11:
liu,文件是用7z压缩的，压缩后改了后缀名。解压之前修改文件，将build.zip改成build.7z，然后用7z解压工具解压，请不要用其他的工具解压，谢谢

发自我的小米手机

logozouxun, 2018-12-11:
你说的是否是ESPTOUCH_DOWN_BIT这个定义？如果不用，如何修改？

发自我的小米手机

TimXia, 2018-12-11:
ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_AP) );//modify
ESP_ERROR_CHECK( esp_wifi_set_config(ESP_IF_WIFI_AP, wifi_config) ); //modify
这两行，不要设成 AP 模式

logozouxun, 2018-12-11:
如何由原来的STA模式切换到AP模式，多谢。

发自我的小米手机

TimXia, 2018-12-11:
@logozouxun 请问你是打算实现什么功能？为什么要切换到 AP 模式？

logozouxun, 2018-12-11:
Tim,我们现在的产品分两种配网模式，其中一项就是AP模式

发自我的小米手机

TimXia, 2018-12-14:
这两行改成：
ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_APSTA) );//modify 
ESP_ERROR_CHECK( esp_wifi_set_config(ESP_IF_WIFI_STA, wifi_config) ); //modify

Alvin1Zhang, 2019-02-14:
@logozouxun Could you help share if any updates for this issue? Thanks

Alvin1Zhang, 2019-03-04:
@logozouxun Hi, hope your issue has been resolved, will close this issue for now, and feel free to reopen, thanks.

closed 2019-03-04 by Alvin1Zhang