espressif/esp-idf #2588: Bluetooth Classic data curruption

jakkra, 2018-10-19:
## Environment

- Development Kit:      [none]
- Core (if using chip or module): [ESP32]
- IDF version f11ac037c4958fc68e67a50134a0ce9a1634008c
- Development Env:		[Make]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description

When running Bluetooth classic SPP we get currupt packets. Random data in packets are wrong, the correct amount of data is sent/received but some packets contain a few bytes which are changed.

In our application we see at least 1-2 packets with wrong data every 1-2GB duplex data. I did some investigation and I can very easily reproduce the problem if I introduce some dampening by using an attenuator between the two ESP32's. By setting the damping to a number which causes the throughput to go down by ~50% I see the packet error after just 1-5 minutes of data TX.

The issue is also reproduceable with the bt_spp_initiator and bt_spp_acceptor. We just added a checksum in the beginning of each packet and in that way we can easily see if the packet was received with an error.

I have tested using Enhanced Retransmission Mode using Bluekitchen stack, and still see the problem. ERTM should catch if there was an error in the air which the 16bit Bluetooth Classic CRC missed. Meaning (assuming ERTM works in Bluekitchen stack) that the error is somewhere in Espressif code. So either the packet is currupted before beeing sent (after leavinig the stack to the BT controller) or that it's currupted after it's received and before it's passed to the stack. 

As I have reproduced the problem with two different stacks, it's not a stack issue. The data is already currupt when the stack on the receiving side gets it. 

My guess from my finding is that there is some issue when there are retransmitions of packets, on receiving side.

Additionally I have placed a sniffer (see image at the end) between the master and the attenuator, meaning I can see exactly what data the master sends. When the slave receives a packet with currupt data I can see all sent packets by the master (including all retransmittions) and I can not see any packet which is invalid here. Which also suggests that the issue is on receiving side. (Assuming ERTM works)

### Expected Behavior
No currupted data received.
### Actual Behavior
Currupted data received.
### Steps to repropduce
Problem is reproduceable without intruducing damping, however much easier if you do. Also the environment in our office is pretty harsh, so possibly it's not possible for you to reproduce without damping (casing retransmitions). Maybe moving two units away from eachother far enough to affect the throughput -50% would also work. I have not tested.
1. Connect two ESP32 with damping between them which cause the max throughput to go down 50+%
2. Run modified bt_spp_initiator and bt_spp_acceptor which also checks for packet errors.
3. After a few minutes you should see that currupt data is received. 

Setup looks like this with two ESP32 with external antenna connected to an attenuator:
[image]

### Question to Espressif
Does Bluedroid support Enhanced Retransmission Mode? If so I could verify with another stack that this is not an issue in the air. By searching in Bluedroid code I find ERTM implementation, how do I enable it?

I'm doing some more investigation, so the issue may update.

mringwal, 2018-10-20:
Bluetooth Link Layers uses CRC-CCITT (Core v5, vol 2, part b, 7.1.2), while L2CAP ERTM FCS uses CRC-16.

In your example, you seems to have two bit errors: 92 -> d2 (sets bit 6) and 93->97 (sets bit 2).

Browsing around, http://www0.cs.ucl.ac.uk/staff/electran/2011/pdf/2011-07.pdf states that both CRC-16 & CRC-CCITT detect all single bit and two bit errors. So, both should have rejected this packet. 

Congrats on finding a quick way to reproduce this quickly.

blueMoodBHD, 2018-10-26:
Hi @jakkra 
We do not support Enhanced Retransmission Mode, now.
And the error may caused by some bit error which CRC-CCITT and CRC-16 cannot detected.

mringwal, 2018-10-26:
Minor update on ERTM here. ERTM can optionally use a frame checksum. With ERTM FCS enabled in BTstack, the combined Link Layer & L2CAP CRC-16 seems to be sufficient to detect all bit errors (resp. @jakkra wasn't able to get another false packet).

closed 2019-03-15 by Alvin1Zhang