espressif/esp-idf #2589: Crash on wifi disconnect

mikaelkanstrup, 2018-10-19:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [-]
- Core (if using chip or module): [ESP32]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 88d4217
- Development Env:		[Make]
- Operating System:		[Ubuntu]
- Power Supply:         [external 5V]

## Problem Description

Occasionally on wifi disconnect we see crashes with the following call stack:

```
(gdb) bt full
#0  0x400910a5 in uxPortCompareSet (set=<synthetic pointer>, compare=3007315967, addr=0x0) at /home/travis/esp-idf/components/freertos/include/freertos/portmacro.h:285
No locals.
#1  vPortCPUAcquireMutexIntsDisabledInternal (timeout_cycles=-1, mux=0x0) at /home/travis/esp-idf/components/freertos/portmux_impl.inc.h:86
        coreID = 52685
        res = 52685
        otherCoreID = 43947
#2  vPortCPUAcquireMutexIntsDisabled (timeout_cycles=-1, mux=0x0) at /home/travis/esp-idf/components/freertos/portmux_impl.h:98
No locals.
#3  vTaskEnterCritical (mux=0x0) at /home/travis/esp-idf/components/freertos/tasks.c:4258
        oldInterruptLevel = 394272
        schedulerRunning = 1
#4  0x400893d8 in wifi_int_disable_wrapper (wifi_int_mux=0x0) at /home/travis/esp-idf/components/esp32/wifi_os_adapter.c:187
No locals.
#5  0x4008ae60 in pp_post ()
No symbol table info available.
#6  0x4013c14d in ieee80211_ioctl ()
No symbol table info available.
#7  0x40131ed6 in esp_wifi_internal_reg_rxcb ()
No symbol table info available.
#8  0x4012fbca in system_event_sta_disconnected_handle_default (event=0x3ffd2c2c) at /home/travis/esp-idf/components/esp32/event_default_handlers.c:243
        __err = 1073556524
        __FUNCTION__ = "system_event_sta_disconnected_handle_default"
#9  0x40130258 in esp_event_process_default (event=0x3ffd2c2c) at /home/travis/esp-idf/components/esp32/event_default_handlers.c:402
No locals.
#10 0x4012ee37 in esp_event_loop_task (pvParameters=<optimized out>) at /home/travis/esp-idf/components/esp32/event_loop.c:54
        ret = <optimized out>
        evt = <unavailable>
```

### Expected Behavior

No crash on wifi disconnect

### Actual Behavior

In our project we currently run with the coredump to flash feature enabled. Most of the crashes found have the above call stack. As we've so far not been able to capture this problem with a device connected to a UART console we've patched ESP-IDF to store the last few KBs of UART console log and the panic string generated from panicPutChar function into two memory buffers. The memory buffers are then stored to the coredump file for post mostem analysis. See the output from printing the two save memory buffers from gdb below.

### Steps to reproduce

No known steps to reproduce. This crash is found quite a lot while analyzing coredumps created in one of our projects using the esp32.

## Debug Logs

```
printf "%s", g_panic_str
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400910a5  PS      : 0x00060233  A0      : 0x800893d8  A1      : 0x3ffd2ac0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffc2d2c  A5      : 0x00000000  
A6      : 0x3ffbec5c  A7      : 0x3ffcec3c  A8      : 0x0000abab  A9      : 0xb33fffff  
A10     : 0x0000cdcd  A11     : 0x00060220  A12     : 0x00060220  A13     : 0x00000001  
A14     : 0x0000cdcd  A15     : 0x3ffd1b00  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400910a5:0x3ffd2ac0 0x400893d5:0x3ffd2ae0 0x4008ae5d:0x3ffd2b00 0x4013c14a:0x3ffd2b40 0x40131ed3:0x3ffd2b60 0x4012fbc7:0x3ffd2b90 0x40130255:0x3ffd2bc0 0x4012ee34:0x3ffd2c20
```

addr2line on this don't reveal much else:
```
xtensa-esp32-elf-addr2line -e ~/Downloads/esp.1.1.0-39-geca7011.elf ~/Downloads/esp32_rom.elf -a 0x400893d5:0x3ffd2ae0 0x4008ae5d:0x3ffd2b00 0x4013c14a:0x3ffd2b40 0x40131ed3:0x3ffd2b60 0x4012fbc7:0x3ffd2b90 0x40130255:0x3ffd2bc0 0x4012ee34:0x3ffd2c20
0x00000000
??:0
0x400893d5
/home/travis/esp-idf/components/esp32/wifi_os_adapter.c:187
0x4008ae5d
??:?
0x4013c14a
??:?
0x40131ed3
??:?
0x4012fbc7
/home/travis/esp-idf/components/esp32/event_default_handlers.c:120
0x40130255
/home/travis/esp-idf/components/esp32/event_default_handlers.c:402
0x4012ee34
/home/travis/esp-idf/components/esp32/event_loop.c:109
```

Console log (from saved coredump memory buffer):
```
I (209) wifi: wifi driver task: 3ffd441c, prio:23, stack:3584, core=0
I (209) wifi: wifi firmware version: 6c35f9d
I (209) wifi: config NVS flash: disabled
I (209) wifi: config nano formating: disabled
I (209) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (209) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (209) wifi: Init dynamic tx buffer num: 32
I (209) wifi: Init data frame dynamic rx buffer num: 32
I (209) wifi: Init management frame dynamic rx buffer num: 32
I (219) wifi: Init static rx buffer size: 1600
I (219) wifi: Init static rx buffer num: 10
I (219) wifi: Init dynamic rx buffer num: 32
I (219) wifi: set country: cc=US  schan=1 nchan=11 policy=1
W (219) wifi: ap start fail

W (219) wifi: sta start fail

D (219) event: SYSTEM_EVENT_STA_STOP
I (219) wifi: set country: cc=EU  schan=1 nchan=13 policy=1

W (219) wifi: sta start fail

I (219) wpa: WPA2: already disabled
D (219) event: SYSTEM_EVENT_STA_STOP
I (219) wifi: Set ps type: 2

D (219) RTC_MODULE: Wi-Fi takes adc2 lock.
D (219) phy_init: loading PHY init data from application binary
D (219) nvs: nvs_open_from_partition phy 0
D (219) nvs: nvs_get cal_version 4
D (219) nvs: nvs_get_str_or_blob cal_mac
D (219) nvs: nvs_get_str_or_blob cal_data
D (219) nvs: nvs_close 3
I (249) wifi: mode : sta (xx:xx:xx:xx:xx:xx)
D (249) event: SYSTEM_EVENT_STA_START
controller rcv pkt ready
host rcv pkt: 040e0405030c00
controller rcv pkt ready
host rcv pkt: 040e0405062000
controller rcv pkt ready
host rcv pkt: 040e0405082000
controller rcv pkt ready
host rcv pkt: 040e04050a2000
controller rcv pkt ready
I (1859) wifi: n:13 0, o:1 0, ap:255 255, sta:13 0, prof:1
I (2419) wifi: state: init -> auth (b0)
I (2419) wifi: state: auth -> assoc (0)
I (2419) wifi: state: assoc -> run (10)
host rcv pkt: 0413050109000000
host rcv pkt: 040e04050a2000
controller rcv pkt ready
I (22309) wifi: state: run -> init (0)
I (22309) wifi: n:13 0, o:13 0, ap:255 255, sta:13 0, prof:1
D (22309) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:NG-Obs, ssid_len:6, bssid:xx:xx:xx:xx:xx:xx, reason:8
D (22309) tcpip_adapter: if0 start ip lost tmr: enter
D (22309) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x3ffb5ae4 interval=120 ip=0
I (22319) wifi: flush txq
I (22319) wifi: stop sw txq
D (22319) phy_init: wifi mac time delta: 70
I (22319) wifi: lmac stop hw txq
D (22319) RTC_MODULE: Wi-Fi returns adc2 lock.
I (22319) wifi: Deinit lldesc rx mblock:10

```

mikaelkanstrup, 2018-10-19:
I realize this is a long shot as I can't provide an easy way to reproduce the problem. Feel free to close this issue if there's too little information for you to be able to conclude anything.

projectgus, 2019-03-15:
Hi @mikaelkanstrup ,

Thanks for being patient while someone got back to you. The EXCVADDR of 0 in the panic handler makes it look like this is a null pointer dereference. Possibly due to memory corruption, a pointer has been overwritten with zero.

Unfortunately without steps to reproduce you're correct that there isn't much we can do, in any case. Memory corruption are insidious because the corruption can come from some totally unrelated part of the system, compared to the crash.

If you're still struggling with this then there are some steps to finding heap corruption (which is not exactly the same as what you're seeing, but not entirely different) here:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/heap_debug.html#finding-heap-corruption

Sorry we can't be of more help.

closed 2019-03-15 by projectgus