espressif/esp-idf #2593: mDNS not sending requests but reading other devices responses

masual, 2018-10-19:
## Environment

- Development Kit:      [none]
- Core (if using chip or module): [ESP32]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 22da5f6de
- Development Env:		[Make]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description

mDNS is not sending requets. After analyzing the network with Wireshark, no frames are being sent by the ESP32 module when calling mdns_query_ptr() or mdns_query_a(). I have a Debian machine responding to mdns requests. In the mDNS example, calling  query_mdns_host("debianXPS13"); and query_mdns_service("_mqtt", "_tcp"); does not produce any response. However if at the same time I run the mdns-scan tool on Debian, the ESP32 gets and shows the responses generated by the requests of the Debian machine.

### Expected Behavior

The calls to the functions should produce the required mDNS requests.

### Actual Behavior

No requests are being sent by the ESP32 module.

### Steps to repropduce

1. Use the mDNS example code.

## Debug Logs

```
I (5182) mdns-test: Query PTR: _mqtt._tcp.local
W (8282) mdns-test: No results found!
I (9282) mdns-test: Query A: debianXPS13.local
W (11382) mdns-test: ESP_ERR_NOT_FOUND: Host was not found!

```

When executing mdns-scan at the same time:

```
I (15482) mdns-test: Query A: debianXPS13.local
I (15902) mdns-test: 192.168.5.40
I (15902) mdns-test: Query PTR: _mqtt._tcp.local
1: Interface: STA, Type: V4
  PTR : debianXPS13
  SRV : debianXPS13.local:8883
  A   : 192.168.5.40
  A   : 254.128.0.0
```

david-cermak, 2018-10-23:
Hi @masual ,

It looks like this issue is related to #2507. You can check applying this change https://github.com/espressif/esp-idf/issues/2507#issuecomment-428704419 to the `components/mdns/mdns.c` file if it resolves your issue. If so, we have already prepared a fix internally.

masual, 2018-10-26:
@david-cermak Thank you very much, that patch indeed fixed my problem. I read the issue #2507 but I did not think it was related to my problem. Anyway I applyed the suggested change and now is working as expected. Please let me know when the patch is applied to master branch.

commit 174b3b89 (2018-11-21): mdns: sending search packets also in probing and announcing state

mdns queries did not work properly when send imeadiately after set_hostname, which cuased reinit of pcb and therefore restarted probing, so search packets were blocked until probing finished
closes #2507, closes #2593
commit e885ff23 (2019-01-11): mdns: sending search packets also in probing and announcing state

mdns queries did not work properly when send imeadiately after set_hostname, which cuased reinit of pcb and therefore restarted probing, so search packets were blocked until probing finished
closes #2507, closes #2593
closed 2018-11-18 by igrr