espressif/esp-idf #2595: Bluetooth Controller: Crash on entering Classic Sniff Mode

mringwal, 2018-10-19:
Summary: when entering sniff mode, an assert in the Bluetooth Controller code fires and it crashes.

To enable Sniff mode:
- I'm sending HCI Write Default Link Policy Settings ( 5 ) during stack startup
- After an incoming connection is established, I try to enable Sniff mode and get the log below.

Is there any special configuration in the sdk config? I've tried to enable 'modem sleep'.

[00:00:08.601] CMD => 03 08 0A 80 00 30 04 00 03 00 01 00 01
[00:00:08.607] EVT <= 0F 04 00 05 03 08
[00:00:08.611] EVT <= 6E 00
[00:00:08.613] CMD => 35 0C 05 01 80 00 02 00
[00:00:08.617] EVT <= 6E 00
[00:00:08.712] EVT <= 14 06 00 80 00 02 30 04
ASSERT_PARAM(131072 0), in rwbt.c at line 311
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40084099  PS      : 0x00060034  A0      : 0x80088264  A1      : 0x3ffbe180
0x40084099: r_assert_param at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054
A6      : 0x3ffbdbbc  A7      : 0x60008050  A8      : 0x80084099  A9      : 0x3ffbe160
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40083fd1  LEND    : 0x40083fd8  LCOUNT  : 0x00000000
0x40083fd1: r_assert_param at ??:?

0x40083fd8: r_assert_param at ??:?

Core 0 was running in ISR context:
EPC1    : 0x400d2f1e  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40084099
0x400d2f1e: esp_vApplicationIdleHook at /Projects/esp32/esp-idf/components/esp32/freertos_hooks.c:88

0x40084099: r_assert_param at ??:?

Backtrace: 0x40084099:0x3ffbe180 0x40088261:0x3ffbe1a0 0x4008899f:0x3ffbe1c0 0x400820ca:0x3ffbe1e0 0x400d2f1b:0x00000000
0x40084099: r_assert_param at ??:?

0x40088261: r_rwbt_isr at ??:?

0x4008899f: r_rwbtdm_isr_wrapper at intc.c:?

0x400820ca: _xt_lowint1 at /Projects/esp32/esp-idf/components/freertos/xtensa_vectors.S:1105

0x400d2f1b: esp_vApplicationIdleHook at /Projects/esp32/esp-idf/components/esp32/freertos_hooks.c:88

================= CORE DUMP START =================
aBQAAAYAAAB0AQAA
vL/7P5C9+z+0v/s/
kL37P1C/+z/6QgAAQBT8P0AU/D+8v/s/OBT8PxgAAADFyzeyAJr17by/+z8AAAAA
AQAAALit+z9tYWluAGJu/p/YqgxjULQAAAAAALS/+z8AAAAAIQAGAAEAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAGR2+z/Mdvs/NHf7PwAAAAAAAAAA
AQAAAAAAAAAQPUA/AAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAEAAAAAk5JS
hB4IQPC/AEAwBAYAkMwIgFC++z8AAAAAIwQGACAEBgAAAAAAAAAAAAMJBwJEEAiA
IL77P/jj+z/6QgAAAAAAAAIAAAAEAAAAgAACMAQAAACcbw6AEL77P/0UAEANFQBA
/////30gCEACAAAApOEIQEig+j8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAgAAAAAAAAAAAAAAAMJBwL22giAYL77PzDk+z/6QgAA
1HcOgIC++z8w5Ps//////3Aa/D+AAAAAAQAAAAAAAACQMg6AoL77P9NNYhDMdvs/
AAAAACwhAAAAAAAAAAAAAF8tDoDQvvs/CgAAAAzPDkAAAAAAAAAAAAAAAAABAAAA
BQAAAAEAAAABAAAAAQAAAHMbDYDwvvs/HCoOQAAAAABAJQAAqHj8PzDR+z8jAAYA
aLoIgBC/+z8AAAAAAAAAAAAAAABgzvs/+OP7PyMABgAAAAAAML/7PzQbDUAAAAAA
IwAGALy/+z8BAAAAAAAAAAAAAABQv/s/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAXL/7PwAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAA=
rM/7PwDO+z+kz/s/
[... 73 lines cut ...]
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAPwf/T8AAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=
QKz7P8B9+z9gf/s/
wH37PwB/+z/J8q/rEHj7P+QT/D9ArPs/3BP8PwEAAAA4e/s/OHv7P0Cs+z8we/s/
GAAAAGR7+z9pcGMwAHkkM2WN/LVW/R4AAAAAAGB/+z8AAAAAIQAGABgAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAGR2+z/Mdvs/NHf7PwAAAAAAAAAA
AQAAAAAAAAAQPUA/AAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAh8gy
hB4IQEDDCEAwAAYAmxoIgIB++z8Me/s/AAAAAFh7+z8AAAAAAQAAAAIAAABAwwiA
YH77PwAAAABAFvw/QBb8P8S/+z9Az/s/GB9BPwAAAAClpaWlpaWlpQAAAAAAAAAA
AAAAAH0gCEDEv/s/pOEIQPhf+j8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAABsGghAAAAAAAAAAABougiAwH77PwAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAP////8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAOB++z9sGghAAAAAACMDBgBArPs/AQAAAAAAAAAAAAAAAH/7PwAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAEIPCICAO/4/yBP8P4QW/D8AAAAAAAAAAAx/+z8AAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=
================= CORE DUMP END =================
E (9987) esp_core_dump: Skipped 1 tasks with bad TCB!
E (9993) esp_core_dump: Crashed task has been skipped!
CPU halted.

blueMoodBHD, 2018-10-23:
Hi @mringwal,
What version of ESP-IDF do you use?  
And if it is convenient, can you provide your code?

mringwal, 2018-10-23:
HI @blueMoodBHD - I'm using the current version of the ESP-IDF master branch (129d32772)

To reproduce:

checkout out BTstack (https://github.com/bluekitchen/btstack/) and change to 'develop' branch.

In btstack/port/esp32, run ./integrate_btstack.py

(This will copy BTstack into esp-idf/components/btstack as well as create example projects at btstack/port/esp32/example. More Info at https://github.com/bluekitchen/btstack/tree/master/port/esp32)

To enable HCI Dump logging:
in esp-idf/components/btstack/main.c, function main(), uncomment the call to hci_dump_open(..)

Please copy the attached [hid_keyboard_demo.c.zip](https://github.com/espressif/esp-idf/files/2505028/hid_keyboard_demo.c.zip) that enters sniff mode into btstack/port/esp32/example/hid_keyboard_demo/main

In btstack/port/esp32/example/hid_keyboard_demo, run make and accept the default answers. Please also have a look at sdkconfig, I'm using the Sparkfun ESP32 Thing.
If make fails, run it again (happens on OS X with stock make).

now, run make flash monitor and connect e.g. from a smartphone.

please let me know if you can follow this instructions and if it crashes for you as well.
Do you know what assert is actually triggered?

blueMoodBHD, 2018-10-23:
Hi @mringwal，
I have reproduced the problem with your code. 
And the reason of the assert is that ATT_NB filed in Control Structure is null.

mringwal, 2018-10-23:
Thanks for the update. I'm glad it reproduces for you.
 Is there anything I (resp. BTstack) needs to do differently?
Good luck with  further analysis!

blueMoodBHD, 2018-10-26:
Hi @mringwal ,
I have find the reason of this issue, but it will tasks some time to fix it.
For now, When you send `HCI_sniff_mode` command, setting `Sniff_Attempt` no more than 0xFF will works OK.

mringwal, 2018-10-26:
Hi @blueMoodBHD Thanks for the update and the suggestion to use sniff_attempt < 0x100. I can confirm that it doesn't crash using 0xff. That's good for now, if it could be fixed fully, that would be excellent, too.

blueMoodBHD, 2018-11-02:
Hi @mringwal ,
The bugfix have been merged into master of ESP-IDF.

Issus:
Sniff_Attempt is a 16-bit parameter, but hardware register only support 8 bit.  So, when entering sniff mode with Sniff_Attempt larger than 0xFF, it will cause some errors. More seriously, when Sniff_Attempt is 0x100, we will write 0 to register, it will cause crash.

Sloution:
We do not support Sniff_Attempt more than 0xFF.
1. When get HCI_Sniff_Mode command with Sniff_Attempt large than 0xFF, issue a warning and change Sniff_Attempt to 0xFF
2. As master, when getting sniff_req with Sniff_Attempt large than 0xFF, issue a warning and sent a sniff_req with new Sniff_Attempt(0xFF)
3. As slave, when getting sniff_req with Sniff_Attempt large than 0xFF, issue a warning and reject it

Alvin1Zhang, 2019-10-12:
@mringwal Thanks for reporting, the fix has been merged. Feel free to reopen if you still have the issue, thanks.

closed 2019-10-12 by Alvin1Zhang