espressif/esp-idf #2597: optimal configuration for https client

xarrange, 2018-10-20:
I have simple web-site base on Apache with SSL based on self-signed certificate. And I want to write simple https client much like the one from examples.

What are the optimal parameters in menuconfig to reduce memory use as much as possible? Because right know SSL subsystem eats about 80K of RAM
(e.g. do I need all versions of TLS protocol, all versions of ciphers, etc..?)

jitin17, 2018-10-22:
Hello @xarrange ! 
Recently, we had a similar requirement to reduce tls memory footprint. I think this [blog](https://tls.mbed.org/kb/how-to/reduce-mbedtls-memory-and-storage-footprint) proves to be very helpful in doing so.
Since you control both server and client, you can reduce SSL frame buffer size which will help you to reduce tls memory requirement significantly. Reducing MPI maximum size helped us in reducing memory requirement, and this might help you as well.  But one thing to note is that all these tweaks comes at the cost of performance.

xarrange, 2018-10-22:
Hi, @jitin17!
Thanks for idea.
However this wasn't much helpful because that blog is related to pure mbedTLS, while we use here ported version.
First of all, there is no option in menuconfig which configures MBEDTLS_MPI_MAX_SIZE. Moreover, this define presents in 3 different files (bignum.h, config-suite-b.h, config-thread.h) and I have no idea which one is really used.
Disabling unused ECP curves also looks strange for me as I have no idea who are they - "unused curves"

So question is what we can optimize only with these parameters:
```
#
# mbedTLS
#
CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=8192
CONFIG_MBEDTLS_DEBUG=
CONFIG_MBEDTLS_HARDWARE_AES=y
CONFIG_MBEDTLS_HARDWARE_MPI=y
CONFIG_MBEDTLS_MPI_USE_INTERRUPT=
CONFIG_MBEDTLS_HARDWARE_SHA=y
CONFIG_MBEDTLS_HAVE_TIME=y
CONFIG_MBEDTLS_HAVE_TIME_DATE=
CONFIG_MBEDTLS_TLS_SERVER_AND_CLIENT=
CONFIG_MBEDTLS_TLS_SERVER_ONLY=
CONFIG_MBEDTLS_TLS_CLIENT_ONLY=y
CONFIG_MBEDTLS_TLS_DISABLED=
CONFIG_MBEDTLS_TLS_CLIENT=y
CONFIG_MBEDTLS_TLS_ENABLED=y

#
# TLS Key Exchange Methods
#
CONFIG_MBEDTLS_PSK_MODES=
CONFIG_MBEDTLS_KEY_EXCHANGE_RSA=y
CONFIG_MBEDTLS_KEY_EXCHANGE_DHE_RSA=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ELLIPTIC_CURVE=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_RSA=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_ECDSA=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDH_ECDSA=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDH_RSA=y
CONFIG_MBEDTLS_SSL_RENEGOTIATION=y
CONFIG_MBEDTLS_SSL_PROTO_SSL3=
CONFIG_MBEDTLS_SSL_PROTO_TLS1=
CONFIG_MBEDTLS_SSL_PROTO_TLS1_1=
CONFIG_MBEDTLS_SSL_PROTO_TLS1_2=y
CONFIG_MBEDTLS_SSL_PROTO_DTLS=
CONFIG_MBEDTLS_SSL_ALPN=y
CONFIG_MBEDTLS_SSL_SESSION_TICKETS=y

#
# Symmetric Ciphers
#
CONFIG_MBEDTLS_AES_C=y
CONFIG_MBEDTLS_CAMELLIA_C=
CONFIG_MBEDTLS_DES_C=
CONFIG_MBEDTLS_RC4_DISABLED=y
CONFIG_MBEDTLS_RC4_ENABLED_NO_DEFAULT=
CONFIG_MBEDTLS_RC4_ENABLED=
CONFIG_MBEDTLS_BLOWFISH_C=
CONFIG_MBEDTLS_XTEA_C=
CONFIG_MBEDTLS_CCM_C=y
CONFIG_MBEDTLS_GCM_C=y
CONFIG_MBEDTLS_RIPEMD160_C=

#
# Certificates
#
CONFIG_MBEDTLS_PEM_PARSE_C=y
CONFIG_MBEDTLS_PEM_WRITE_C=y
CONFIG_MBEDTLS_X509_CRL_PARSE_C=y
CONFIG_MBEDTLS_X509_CSR_PARSE_C=y
CONFIG_MBEDTLS_ECP_C=y
CONFIG_MBEDTLS_ECDH_C=y
CONFIG_MBEDTLS_ECDSA_C=y
CONFIG_MBEDTLS_ECP_DP_SECP192R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP224R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP256R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP384R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP521R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP192K1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP224K1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP256K1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_BP256R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_BP384R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_BP512R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_CURVE25519_ENABLED=y
CONFIG_MBEDTLS_ECP_NIST_OPTIM=y 
```

jitin17, 2018-10-24:
@xarrange 

> Because right know SSL subsystem eats about 80K of RAM
So, do you mean 80K of heap?

And, with the latest configuration that you have sent, what is the memory footprint and the negotiated cipher suite?

If you are okay with not using  `Elliptic Curve Ciphers` then can you calculate the memory footprint by disabling  it under `make menuconfig` -> `Component config` -> `mbedTLS` -> `Elliptic Curve Ciphers` ?
Also, can you disable `Support TLS renegotiation` under `make menuconfig` -> `Component config` -> `mbedTLS` -> Support TLS renegotiation`?

jitin17, 2018-10-27:
@xarrange Now, I think the ball is in your court; you'll have to decide which cipher suite to support. You  may also choose to support some elliptic curve ciphers under `make menuconfig` -> `Component config` -> `mbedTLS` -> `Elliptic Curve Ciphers`.  
Keep us posted if you need any other assistance.

closed 2018-10-27 by xarrange