espressif/esp-idf #2598: esp-idf latest build error in protocomm

buckybots, 2018-10-20:
switched to master branch from v3.1-rc2 and no longer compiles with make.

Here is the broken part :+1: 

C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:33:38: error: unknown type name 'httpd_req_t'
 static esp_err_t common_post_handler(httpd_req_t *req)
                                      ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c: In function 'protocomm_httpd_add_endpoint':
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:138:30: error: implicit declaration of function 'strlen' [-Werror=implicit-function-declaration]
     char* ep_uri = calloc(1, strlen(ep_name) + 2);
                              ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:138:30: warning: incompatible implicit declaration of built-in function 'strlen'
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:138:30: note: include '<string.h>' or provide a declaration of 'strlen'
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:146:5: error: unknown type name 'httpd_uri_t'
     httpd_uri_t config_handler = {
     ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:147:9: error: field name not in record or union initializer
         .uri      = ep_uri,
         ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:147:9: note: (near initialization for 'config_handler')
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:147:21: warning: initialization makes integer from pointer without a cast [-Wint-conversion]
         .uri      = ep_uri,
                     ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:147:21: note: (near initialization for 'config_handler')
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:148:9: error: field name not in record or union initializer
         .method   = HTTP_POST,
         ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:148:9: note: (near initialization for 'config_handler')
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:148:21: warning: excess elements in scalar initializer
         .method   = HTTP_POST,
                     ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:148:21: note: (near initialization for 'config_handler')
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:149:9: error: field name not in record or union initializer
         .handler  = common_post_handler,
         ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:149:9: note: (near initialization for 'config_handler')
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:149:21: error: 'common_post_handler' undeclared (first use in this function)
         .handler  = common_post_handler,
                     ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:149:21: note: each undeclared identifier is reported only once for each function it appears in
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:149:21: warning: excess elements in scalar initializer
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:149:21: note: (near initialization for 'config_handler')
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:150:9: error: field name not in record or union initializer
         .user_ctx = NULL
         ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:150:9: note: (near initialization for 'config_handler')
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:155:5: error: unknown type name 'httpd_handle_t'
     httpd_handle_t *server = (httpd_handle_t *) pc_httpd->priv;
     ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:155:31: error: 'httpd_handle_t' undeclared (first use in this function)
     httpd_handle_t *server = (httpd_handle_t *) pc_httpd->priv;
                               ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:155:47: error: expected expression before ')' token
     httpd_handle_t *server = (httpd_handle_t *) pc_httpd->priv;
                                               ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:156:16: error: implicit declaration of function 'httpd_register_uri_handler' [-Werror=implicit-function-declaration]
     if ((err = httpd_register_uri_handler(*server, &config_handler)) != ESP_OK) {
                ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c: In function 'protocomm_httpd_remove_endpoint':
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:175:30: warning: incompatible implicit declaration of built-in function 'strlen'
     char* ep_uri = calloc(1, strlen(ep_name) + 2);
                              ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:175:30: note: include '<string.h>' or provide a declaration of 'strlen'
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:184:5: error: unknown type name 'httpd_handle_t'
     httpd_handle_t *server = (httpd_handle_t *) pc_httpd->priv;
     ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:184:31: error: 'httpd_handle_t' undeclared (first use in this function)
     httpd_handle_t *server = (httpd_handle_t *) pc_httpd->priv;
                               ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:184:47: error: expected expression before ')' token
[... 34 lines cut ...]
     if ((err = httpd_start((httpd_handle_t *)pc->priv, &server_config)) != ESP_OK) {
                                             ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:218:20: warning: variable 'server_config' set but not used [-Wunused-but-set-variable]
     httpd_config_t server_config   = HTTPD_DEFAULT_CONFIG();
                    ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c: In function 'protocomm_httpd_stop':
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:241:9: error: unknown type name 'httpd_handle_t'
         httpd_handle_t *server_handle = (httpd_handle_t *) pc_httpd->priv;
         ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:241:42: error: 'httpd_handle_t' undeclared (first use in this function)
         httpd_handle_t *server_handle = (httpd_handle_t *) pc_httpd->priv;
                                          ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:241:58: error: expected expression before ')' token
         httpd_handle_t *server_handle = (httpd_handle_t *) pc_httpd->priv;
                                                          ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:242:9: error: implicit declaration of function 'httpd_stop' [-Werror=implicit-function-declaration]
         httpd_stop(*server_handle);
         ^
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c: At top level:
C:/linuxShared/esp/esp-idf/components/protocomm/src/transports/protocomm_httpd.c:29:17: warning: 'session_id' defined but not used [-Wunused-variable]
 static uint32_t session_id = PROTOCOMM_NO_SESSION_ID;
                 ^
cc1.exe: some warnings being treated as errors
make[1]: *** [/c/linuxShared/esp/esp-idf/make/component_wrapper.mk:286: src/transports/protocomm_httpd.o] Error 1
make: *** [C:\linuxShared\esp\esp-idf/make/project.mk:505: component-protocomm-build] Error 2

igrr, 2018-10-21:
Possibly same issue as https://github.com/espressif/esp-idf/issues/2583#issuecomment-431292166?

buckybots, 2018-10-31:
Yes, same exact issue, name conflict with http-server.h.  Was using your quick and dirty http-server, and now that a http-server is part of esp-idf  this component is no longer needed.  Easy fix, just removed my http-server component.

closed 2018-10-31 by buckybots