espressif/esp-idf #2602: Running "make monitor" in windows command prompt or other shells

jamesgz, 2018-10-21:
Hi guys, I am trying to use the command "make monitor" in windows command prompt or other shells other than mingw32.exe. I am using Windows 10 64bit.

Note: I want to use cmd to run a monitor command because I am trying to write a Python script to automate factory testing. The script should use subprocess to run that monitor command and terminate that command when a certain string is observed in the output, and then use that string later. I tried making subprocess open mingw32.exe to run that command but it seems like subprocess cannot live capture the output of an opened terminal properly. I also tried writing a shell script to be ran with mingw32.exe, but that did not work either. 

I kind of messed up my environment variables a while ago, but I think I fixed it. I have the following environment variables set up:
`PROJECT_DIR = C:\Users\james\factory\factory-test\panel-table

IDF_PATH = %PROJECT_DIR%\esp-idf

PATH = 
%SystemRoot%\system32
%SystemRoot%
%SystemRoot%\System32\Wbem
%SystemRoot%\System32\WindowsPowerShell\v1.0\
%SystemRoot%\System32\OpenSSH\
C:\Program Files (x86)\Common Files\Oracle\Java\javapath
C:\Program Files (x86)\GtkSharp\2.12\bin
C:\Program Files\dotnet\
C:\Program Files (x86)\Skype\Phone\
C:\Program Files\Git\cmd
C:\Program Files\PuTTY\
C:\Program Files\CMake\bin
C:\Users\james\AppData\Local\Programs\Python\Python37
C:\Users\james\AppData\Local\Programs\Microsoft VS Code\bin
C:\msys32\usr\bin
C:\MinGW\msys\1.0\bin
C:\Program Files (x86)\mingw-w64\i686-8.1.0-posix-dwarf-rt_v6-rev0\mingw32\bin
C:\MinGW\bin
C:\msys32\opt\xtensa-esp32-elf\bin
`

"make monitor" works fine in mingw32.exe (even though it says Toolchain version is not supported)

However, running "make monitor" in cmd outputs the following:

`
C:\msys32\home\james\esp\hello_world>make monitor
WARNING: Toolchain version is not supported: 1.22.0-73-ge28a011
Expected to see version: 1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
MONITOR
Traceback (most recent call last):
  File "C:/Users/james/factory/factory-test/panel-table/esp-idf/tools/idf_monitor.py", line 754, in <module>
    main()
  File "C:/Users/james/factory/factory-test/panel-table/esp-idf/tools/idf_monitor.py", line 670, in main
    p=serial_instance))
  File "C:/Users/james/factory/factory-test/panel-table/esp-idf/tools/idf_monitor.py", line 72, in yellow_print
    color_print(message, ANSI_YELLOW)
  File "C:/Users/james/factory/factory-test/panel-table/esp-idf/tools/idf_monitor.py", line 69, in color_print
    sys.stderr.write("%s%s%s\n" % (color, message,  ANSI_NORMAL))
  File "C:/Users/james/factory/factory-test/panel-table/esp-idf/tools/idf_monitor.py", line 736, in write
    m = re.match(RE_ANSI_COLOR, self.matched)
  File "C:\Users\james\AppData\Local\Programs\Python\Python37\lib\re.py", line 173, in match
    return _compile(pattern, flags).match(string)
TypeError: cannot use a bytes pattern on a string-like object
make: *** [/c/Users/james/factory/factory-test/panel-table/esp-idf/components/esptool_py/Makefile.projbuild:106: monitor] Error 1
`

Running "make monitor" in git bash outputs the following:
`
C:\Users\james\factory\factory-test\panel-table\esp-idf/make/project.mk:61: esp-idf build system only supports MSYS2 in "MINGW32" mode. Consult the ESP-IDF documentation for details.
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
MONITOR
Traceback (most recent call last):
  File "C:/Users/james/factory/factory-test/panel-table/esp-idf/tools/idf_monito
r.py", line 754, in <module>
    main()
  File "C:/Users/james/factory/factory-test/panel-table/esp-idf/tools/idf_monito
r.py", line 670, in main
    p=serial_instance))
  File "C:/Users/james/factory/factory-test/panel-table/esp-idf/tools/idf_monito
r.py", line 72, in yellow_print
    color_print(message, ANSI_YELLOW)
  File "C:/Users/james/factory/factory-test/panel-table/esp-idf/tools/idf_monito
r.py", line 69, in color_print
    sys.stderr.write("%s%s%s\n" % (color, message,  ANSI_NORMAL))
  File "C:/Users/james/factory/factory-test/panel-table/esp-idf/tools/idf_monito
r.py", line 736, in write
    m = re.match(RE_ANSI_COLOR, self.matched)
  File "C:\Users\james\AppData\Local\Programs\Python\Python37\lib\re.py", line 1
73, in match
    return _compile(pattern, flags).match(string)
TypeError: cannot use a bytes pattern on a string-like object
make: *** [/c/Users/james/factory/factory-test/panel-table/esp-idf/components/esptool_py/Makefile.projbuild:106: monitor] Error 1
`

Any help would be greatly appreciated.

dobairoland, 2018-10-22:
Hi @jamesgz. The error you are getting is cased by Python incompatibility. You are probably using  Python 2 in `mingw32` and Python 3 in `cmd`.

Please note that Python 3 support is experimental in master. I'm guessing based on your log (line numbers don't match the latest `idf_monitor`) that you are using some older version of IDF where Python 3 isn't working yet. Please post the output of `git rev-parse --short HEAD`.

The solution to your problem could be:
1. use the latest master,
2. or install a Python 2 interpreter (by Windows installer, outside `mingw32`) and the Python packages for this interpreter from `$IDF_PATH/requirements.txt`, and use that interpreter in `cmd`.

dobairoland, 2018-10-22:
Hello @jamesgz. Sorry if you receive this twice but my previous comment just got lost somehow.

The problem is caused by Python incompatibility. Python 3 is not yet supported but the latest master contains experimental support.

Probably you are using an older IDF guessing from the line numbers from your log. Please post the output of `git rev-parse --short HEAD`.

Your problem could be solved:
1. by using the latest master (and then IDF monitor should work under Python 3),
2. install a Python 2.7 interpreter and the packages from `$IDF_PATH/requirements.txt` for that interpreter. Maybe you can access directly the python2 interpreter from `mingw32` from inside `cmd` (I'm not sure about this). But probably a Windows installer should work.

jamesgz, 2018-10-25:
> Hello @jamesgz. Sorry if you receive this twice but my previous comment just got lost somehow.
>
> The problem is caused by Python incompatibility. Python 3 is not yet supported but the latest master [...]

Thanks a lot! That fixed my problem. I ended up using pyserial instead to monitor my device.

dobairoland, 2018-10-25:
Good to hear that!

closed 2018-10-25 by dobairoland