espressif/esp-idf #2603: idf.py: error: too few arguments

AGPisland, 2018-10-21:
I read the installation instructions are my steps, proper installation of esp-idf-tools-setup-1.2.exe, I set environment variables, PySerial, future, Cryptography, in python2.7 and response to command idf.py menuconfig is
idf.py: error: too few arguments.

I only have python27 installed on my system, vs code and esp-idf-tools-setup-1.2.

Alvin1Zhang, 2018-10-27:
@AGPisland Could you help share if any updates for this issue? Thanks.

AGPisland, 2018-10-29:
Hi @Alvin1Zhang  unfortunately, i wasn't able to make the idf.py work properly; i think i may have misconfigured the build environment. Becouse of the upcoming due date on my project, I have decided to work with micropython, and if that doesn't work, i'll go back to idf.py

souryavarenya, 2019-03-08:
Hi there!

I wish to re-open this issue.

Win10; 
Toolchain installation instructions for windows 10 (CMake version); 

[image]

Alvin1Zhang, 2019-03-08:
Issue reopened.

souryavarenya, 2019-03-08:
I did some more digging into the issue and I believe the problem is with the `argparse` usage. I haven't been able to figure out where exactly but here's what I did to check it:

I opened `idf.py` and made the `actions` argument optional and passed a default value `"all"`. 

This led to a *KeyError* indicating it couldn't find the key `"a"` in `ACTIONS`. I realized that it was only taking the first character of default value I gave and therefore duplicated some keys with names having single character. 

After doing this, it went ahead with compilation but stopped for another irrelevant bug for which I would open a separate issue.

souryavarenya, 2019-03-08:
At the moment, I have got reasons to believe that native Windows compilation is far from finished. WSL still works fine for me and I was just trying this out of curiosity and I'm really looking forward to good windows support.

projectgus, 2019-03-15:
Hi @souryavarenya,

Glad you found a workaround for now.

If you are still set up to reproduce the Windows situation, can you please let me know:

* Are you use Windows Command Prompt or some other terminal?
* If you run `python %IDF_PATH%\tools\idf.py --help` or `python %IDF_PATH\tools\idf.py menuconfig`, does this work?
* What is the output of `python --version`?
* If you type `ftype | findstr -i python`, what does it show? ([context](https://stackoverflow.com/a/8196691/1006619))

Thanks,

Angus

souryavarenya, 2019-03-15:
Hello @angus,

Here are the details you requested:
* I was using CMD when this error occurred. Powershell throws the same issue.
* **Yes** `python %IDF_PATH%\tools\idf.py --help` (or `python $env:IDF_PATH\tools\idf.py --help` in PowerShell) works. I still wonder why it's different from directly calling `idf.py --help`
* The issue is the same, both with 2.7 and 3.6. Tried both. 
* Here's what it shows as of now. I don't suspect any issue with the file type associations. 

```
C:\Users\varen>ftype | findstr -i python

Python.ArchiveFile="C:\WINDOWS\py.exe" "%L" %*
Python.CompiledFile="C:\Python27\python.exe" "%1" %*
Python.File="C:\Python27\python.exe" "%1" %*
Python.NoConArchiveFile="C:\WINDOWS\pyw.exe" "%L" %*
Python.NoConFile="C:\Python27\pythonw.exe" "%1" %*
```

From what I am able to catch here, calling the python script directly (`idf.py menuconfig`) is causing trouble while `python %IDF_PATH%\tools\idf.py menuconfig` is working fine. Seems like just a small issue with the way cmd/powershell/windows treats arguments passed by directly calling the python script.

souryavarenya, 2019-03-15:
As I concluded on my previous comment, I have been able to isolate the issue to the way these commands are treated in CMD/PowerShell/Windows. I have done this my making an `argparse` test script and it throws the same issue when `test.py my_args` is used, which otherwise doesn't show up while executing `python test.py my_args`.

souryavarenya, 2019-03-15:
Hello,

After some more research, I have found no solution to this issue, which seems to be due to the way `argparse` works with windows. 

[Here](https://stackoverflow.com/a/12421946/7944565)'s a StackOverflow thread that talks about this exact issue but in the solution suggests that it could be due to an incomplete setup of Python on Windows. There's no further progress from here.

[This](https://stackoverflow.com/questions/37828739/run-python-program-without-python-in-windows) thread here talks about not including `python` before the script name. My sample python scripts run without having to prefix `python` in the command **with an exception of `argparse`**. So the described problem is not what I am facing, although I have also tried the following:
* Added `.PY` in the global `PATHEXT` variable
* Checked for file associations and have tried both 2.7 and 3.4
---
I guess this is the end of the road for this issue.

souryavarenya, 2019-03-15:
UPDATE: idf.py somehow works now when used from PowerShell, unlike before where it used to throw the error. It still doesn't work with CMD however. 

Amidst this bizarre exploration, I find myself totally unsure of what might have got this to work. (I am guessing it could be because of the `PATHEXT` variable)

projectgus, 2019-03-18:
> I guess this is the end of the road for this issue.

From our point of view, we'd like `idf.py` to work for all Windows users. So it would be good to figure out what the root cause is here. Although you're not obliged to keep helping us, of course. :)

Windows Command Prompt runs idf.py fine on my Windows machine (standard Python installation), and colleagues' Windows machines. So there is some particular Windows configuration which is breaking it.

If you run the `assoc` command [as mentioned here](https://stackoverflow.com/a/8196691/1006619), do you see anything unusual? There should be something like `.py=Python.file` in the output.

A second thing, if you have a test Python program that uses argparse can you please put the following code in it and see what it says:

```
import sys
print(repr(sys.argv))
```
... this will show us exactly what command line arguments are being passed to the python interpreter.

Here it is on my machine (also running `idf.py clean` in the wrong directory, to prove argparse can work!)

[image]

souryavarenya, 2019-03-18:
Hi Angus, 

I can't express how much I appreciate your efforts in addressing the community with such diligence. Your suggestions have helped me resolve this bit of the issue finally. 

---
Here's what it shows up on checking the associations: 
[image]
Safe to conclude that the issue isn't with this. 

The second test you suggested is what takes us ahead and has links to [this issue](https://stackoverflow.com/questions/2640971/windows-is-not-passing-command-line-arguments-to-python-programs-executed-from-t) and the given [solution](https://stackoverflow.com/a/2641185/7944565) works. The issue is indeed with the argument-handling in CMD and how a python registry dictates its functioning.  

It's also important to understand the root cause of this issue and one of the comments on the above SO thread suggests the follows:

> On a side note: The root cause of this might be due to an initial file association with .py files prior to the Python install. Just ran into this issue, files of type .py had been associated with Notepad and then Python was installed and this ugly problem reared its ugly head. On Windows 10 BTW. _– g4m3c0d3r May 24 '16 at 22:32_

The issue is _potentially_ because of an improper/overlapping installation or association of `.py` files.

---

To consolidate the solution for the visitors in future, here it is:

- [x] Check in system path if the desired Python directory is present there in the highest order
- [x] Uninstall irrelevant Python installations that came bundled with anything else
- [x] Check the associations of `,py` files. [More info here](https://stackoverflow.com/questions/11472843/set-up-python-on-windows-to-not-type-python-in-cmd).
- [x] Add `.PY` to `PATHEXT` global variable
- [x] Check the registry entry to be as follows in order to accept arguments. [More info here](https://stackoverflow.com/questions/2640971/windows-is-not-passing-command-line-arguments-to-python-programs-executed-from-t). 
[image]

NOTE: The directory would depend on the default installation of python being used or even the OS itself. 

---

This issue can be closed now!

projectgus, 2019-03-26:
@souryavarenya Thanks for the comprehensive explanation. I'm glad you figured this out!

It's unfortunate that because our entry point is in Python it's not easy for us to check any of this at runtime to provide helpful hints (chicken & egg problem). This is a very useful reference for anyone who runs into this in the future, though.

closed 2019-03-26 by projectgus