espressif/esp-idf #2605: SPIRAM chip wrong assigment/recognition

chegewara, 2018-10-21:
In master we have in addition 8MB spiram chip support, but chips are not properly assigned and/or recognized:
I have board with 4MB spiram (at least according to seller), and before it has been working. Now im getting this error:
`E (405) spiram: Expected 4096KiB chip but found 8192KiB chip. Bailing out..`

This is my board:
https://www.seeedstudio.com/ESP32-CAM-Development-Board-with-camer-p-3153.html

chegewara, 2018-10-21:
When i select `ESP-PSRAM64 or LY68L6400` chip is recognized and 4MB is added to pool:
` spiram: Adding pool of 4096K of external SPI memory to heap allocator`

negativekelvin, 2018-10-21:
Can you read the marking on the chip?

>but for some reason it was detected before update as 4MB
Before there was no attempt to detect the chip, just hardcoded support for 4mb

chegewara, 2018-10-21:
I should have check it before, but I didn't expect it will be 64Mb SPIRAM. IPS6404L. Thanks.

closed 2018-10-21 by chegewara