espressif/esp-idf #2606: Documentation regarding wifi_config_t initialization to zeros

dereks, 2018-10-21:
## Environment

- Development Kit:      None
- Core (if using chip or module): ESP-WROOM32
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
11b444b

## Problem Description

The Wifi examples all have a trivial, hard-coded, one-time initialization of the `wifi_config_t` struct. For example,

```
wifi_config_t wifi_config = {
    .sta = {
        .ssid = DEFAULT_SSID,
        .password = DEFAULT_PWD,
        .listen_interval = DEFAULT_LISTEN_INTERVAL,
    },
}
```

I am working on an app that does a little more than that: it scans for APs, does some bluetooth stuff to let the user choose which AP they want, then tries connecting to that AP.

I had a bug where I could (a) connect with a hard-coded SSID, or (b) scan and find my SSID. But if I scanned and then tried to re-config with the selected SSID, I would keep getting this confusing error:

```
D (79063) network: Disconnect: WIFI_REASON_NO_AP_FOUND (201) ssid:LinuxRules, ssid_len:10, bssid:00:00:00:00:00:00
```

Why NO_AP_FOUND? It sees it fine when I scan! It sees it fine if I just connect! So 4 hours later... I find the bug. Here is where I init the wifi config with my SSID and password, right before trying to `esp_wifi_connect()`:

```
static void configure_wifi(const char *ssid, const char *password)
{
    esp_err_t err;
    wifi_config_t wifi_config;

    // wifi_config.sta.ssid[32] and .password[64] are uint8_t arrays, not pointers.
    strncpy((char *)wifi_config.sta.ssid, ssid, SSID_BUFFER_SIZE);
    strncpy((char *)wifi_config.sta.password, password, SSID_PASSWORD_BUFFER_SIZE);

    ESP_LOGI(TAG, "Setting WiFi configuration SSID %s...", wifi_config.sta.ssid);
    err = esp_wifi_set_config(WIFI_IF_STA, &wifi_config);

    ESP_ERROR_CHECK( esp_wifi_set_config(WIFI_IF_STA, &wifi_config) );
}
```

If I just ran this after boot, I got lucky and it worked. But after doing a scan and some other BLE stuff, the memory at `&wifi_config` was garbage, and it broke my connection attempt. Here was my fix:

```
wifi_config_t wifi_config = {0};   // Clear struct to all zeros.
```

Although this is just simple C initialization (always init your variables) the some-times working nature wasted time on the bug hunt.

Please either call out this `wifi_config_t` initialization requirement in the documentation and examples. Or even better, create a handy initialization macro like `WIFI_INIT_CONFIG_DEFAULT()` is to `wifi_init_config_t` that just zeros it out.

Thanks!

liuzfesp, 2018-10-22:
@dereks, good catch! We will document it!

liuzfesp, 2019-03-19:
HI @dereks we already updated wifi document, refer section <ESP32 Wi-Fi API Parameter Initialization> in https://github.com/espressif/esp-idf/blob/master/docs/en/api-guides/wifi.rst

closed 2019-03-19 by Alvin1Zhang