espressif/esp-idf #2608: About the SD card example I can't download to esp32

TobeChan, 2018-10-22:
I had follow readme to  pullup both GPIOs 15, 2, 4, 12, 13,But when i connect with PC to download the code,the monitor show:
Flashing binaries to serial port COM54 (app at offset 0x10000)...
esptool.py v2.5.0
Serial port COM54
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
MAC: 30:ae:a4:17:93:48
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 22864 bytes to 13498...

A fatal error occurred: Timed out waiting for packet content
make: *** [/home/Administrator/esp/esp-idf/components/esptool_py/Makefile.projbuild:63：flash] 错误 2

So i follow sd_pullup_requirements.rst to connect GPIO0 with GPIO2, but this error still exist.
May i miss something?

jkearins, 2018-10-22:
I got the button between GPIO0 and GPIO2. GPIO2 is pulled-up to 3.3v with 2k2 resistor. For flashing I press the button to connect GPIO0 and GPIO2 and press the EN button on esp32 devkit. I press and hold the buttons till flashing begins, then I release the buttons.

TobeChan, 2018-10-23:
@jkearins Thank for your suggest.I had fix this problem.
I just pull up GPIOs 15, 4, 13,and then esp32 running perfect.

closed 2018-10-23 by Alvin1Zhang

AkhileshThorat, 2018-12-04:
@TobeChan I just connected GPIO's 15,4,13 to pull up resistors, i get following error
E (350) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (350) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.

when i connect GPIO's to 15,4,13,2,12 i get following message while flashing
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Flash params set to 0x0220
Compressed 23024 bytes to 13650...

A fatal error occurred: Timed out waiting for packet content
make: *** [/home/AK/esp/esp-idf/components/esptool_py/Makefile.projbuild:63: flash] Error 2