espressif/esp-idf #2609: esp_ble_gatts_get_attr_value() returns invalid length if it's zero

dereks, 2018-10-22:
esp_ble_gatts_get_attr_value(attr_handle, &length, &value);
## Environment

- Core (if using chip or module): ESP-WROOM32
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
11b444b

- Operating System:		Ubuntu
- Power Supply:         [Battery]

## Problem Description

I have a GATT attribute that gets set by the user. Initially in the firmware, it has length zero, and I set that as the initial length value in my `esp_gatts_attr_db_t` entry that I use to register it (as the second-to-last argument in the struct, as per the docs).

Later when I want to see if the user has sent it to me with BLE, I call this:

```
esp_ble_gatts_get_attr_value(attr_handle, &length, &value);
```

If the user hasn't set it yet, and thus it still has length zero, I get this error log from the library:

```
E (23921) BT_GATT: gatts_get_attribute_value failed:the value length is 0
```

That error seems wrong, zero should not be an error. But the worst part is, my variable `length` whose address I passed in is set to 1, so I can't just check it like I want to.

- If the charactertic's length is zero, please just populate *length with zero. Not 1. 

- Please don't return error on a zero length. It is not a special case, it just means the user hasn't set it yet. Now I must dig into the source code to see what error code is returned, as it's not documented. Or, just treat all errors as if it was just a zero length, which masks other more serious errors.

Weijian-Espressif, 2018-10-30:
@dereks,  I will modify the code, thanks.

Weijian-Espressif, 2018-11-05:
@dereks we have fixed the issue, please update to the latest IDF and try, thanks.

Alvin1Zhang, 2018-11-09:
@dereks Hi, could you please help share if any updates for this issue? Thanks.

Alvin1Zhang, 2019-02-14:
@dereks Feel free to reopen this issue, thanks.

closed 2019-02-14 by Alvin1Zhang