espressif/esp-idf #2613: About the example of sdcard can't download into esp32.

TobeChan, 2018-10-22:
I had follow readme to  pullup both GPIOs 15, 2, 4, 12, 13,But when i connect with PC to download the code,the monitor show:
Flashing binaries to serial port COM54 (app at offset 0x10000)...
esptool.py v2.5.0
Serial port COM54
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
MAC: 30:ae:a4:17:93:48
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 22864 bytes to 13498...

A fatal error occurred: Timed out waiting for packet content
make: *** [/home/Administrator/esp/esp-idf/components/esptool_py/Makefile.projbuild:63：flash] error 2

So i follow sd_pullup_requirements.rst to connect GPIO0 with GPIO2, but this error still exist.
Maybe i miss something?

TobeChan, 2018-10-22:
test

negativekelvin, 2018-10-22:
@TobeChan please close all your duplicate issues

TobeChan, 2018-10-23:
@negativekelvin OK, I am sorry,my network was block  at that time

closed 2018-10-23 by TobeChan