espressif/esp-idf #2614: esp32 uart tx random loss 4-7bytes when sending data to esp8266

outermanle, 2018-10-22:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Core (if using chip or module): [ESP-WROOM-32D]
- IDF version (3276a1316): 
- Development Env:		[Make]
- Operating System:		[Ubuntu]
- Power Supply:         [USB]

- ESP8266_RTOS_SDK master version (e409678ccc6399a1b1ca4656267f507f6a116fc5)

## Problem Description
Esp8266 ota throught uart0, which sending by esp32 uart1.
Random loss data in esp8266 uart_read_bytes, the buffer is 1024, 100% reproduce it.
The esp8266 firmware size is about 330KB.

I am not sure it is esp32 uart driver issue or esp8266 driver issue.
I have merge the patch 508fb79a26a9b62e9b3007b98169a2bc2005594d but it not works. @igrr 
```
 uart: fix uart_tx_wait_idle to wait for fifo empty

In some cases, when data was just written into UART FIFO, transmitter
state could be still zero while the FIFO did contain some data.
This resulted in uart_tx_wait_idle occasionally returning before all
the data was sent out.
Fix by checking both UART transmitter state and TX FIFO count.
```
I have no oscillocope to captrue the signal.

uart config 
```c
    uart_config_t uart_config = {
        .baud_rate = 115200,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_ODD,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE
    };
```

Any help will be greatly appreciated.
Thanks.

### Expected Behavior
Normally works and no data loss in esp8266.

### Actual Behavior
random loss data in esp8266 uart rx.

### Steps to repropduce

esp8266 ota throught uart, sending by esp32.
100% reproduce this issue.

## Debug Logs

#### 1. esp32 uart1 tx data then wait for esp8266 response and continue.
```c
I (82892) ota: Wrote [1024] bytes-----------tx 1024
I (83102) uart-rx: cmd ret=[ota=1020], len=8 ---- but rx 1020 loss 4 char.
I (83102) uart-rx: buf=[ota=1020] ,ret=[ota], ret_cmd=[1020]
W (83102) ota: ota result=[1020]
E (83102) ota: write=1024, return=1020
E (83102) httpd: esp8266_ota_write error ESP_FAIL!
I (83112) httpd: 0x3ffdabd0   5f 69 74 65 6d 5f 68 61  73 68 5f 6c 69 73 74 2e  |_item_hash_list.|
I (83122) httpd: 0x3ffdabe0   63 70 70 00 00 00 66 61  6c 73 65 20 26 26 20 22  |cpp...false && "|
I (83132) httpd: 0x3ffdabf0   69 74 65 6d 20 73 68 6f  75 6c 64 20 68 61 76 65  |item should have|
...
I (83692) httpd: 0x3ffdaf70   61 64 61 70 74 65 72 20  68 61 73 20 6e 6f 20 6d  |adapter has no m|
I (83702) httpd: 0x3ffdaf80   65 6d 6f 72 79 0a 00 73  74 61 30 00 54 43 50 49  |emory..sta0.TCPI|
I (83712) httpd: 0x3ffdaf90   50 20 61 64 61 70 74 65  72 20 62 69 6e 64 20 6e  |P adapter bind n|
I (83722) httpd: 0x3ffdafa0   65 74 20 63 61 72 64 20  25 73 20 65 72 72 6f 72  |et card %s error|
I (83732) httpd: 0x3ffdafb0   0a 00 61 70 30 00 00 00  00 00 96 30 07 77 2c 61  |..ap0......0.w,a|
I (83742) httpd: 0x3ffdafc0   0e ee ba 51 09 99 19 c4  6d 07 8f f4 6a 70 35 a5  |...Q....m...jp5.|
```

#### 2.esp8266 uart0 rx and write flash then respsonse to esp32
 This case missing [ bin] 4 char. It should be 'adapter bind' instead of 'adaptend'.
```c
I (82822) ota: Read [1024] bytes
I (82824) ota: total size = 336928, write size 98982, current size = 1024
I (82839) ota: [CRET:{ota=1024};END] ------previos is normal.

I (83149) ota: Read [1020] bytes  ---- read 1024, but esp32 send 1024! loss 4 char [ bin]
I (83142) ota: total size = 336928, write size 100006, current size = 1020
I (83145) ota: [CRET:{ota=1020};END]
I (85255) slave: 0x40108354   5f 69 74 65 6d 5f 68 61  73 68 5f 6c 69 73 74 2e  |_item_hash_list|
I (85255) slave: 0x40108364   63 70 70 00 00 00 66 61  6c 73 65 20 26 26 20 22  |cpp...false && |
I (85264) slave: 0x40108374   69 74 65 6d 20 73 68 6f  75 6c 64 20 68 61 76 65  |item should hav|
...
I (85819) slave: 0x401086f4   61 64 61 70 74 65 72 20  68 61 73 20 6e 6f 20 6d  |adapter has no |
I (85818) slave: 0x40108704   65 6d 6f 72 79 0a 00 73  74 61 30 00 54 43 50 49  |emory..sta0.TCP|
I (85828) slave: 0x40108714   50 20 61 64 61 70 74 65  6e 64 20 6e 65 74 20 63  |P adaptend net | ------> lost
I (85837) slave: 0x40108724   61 72 64 20 25 73 20 65  72 72 6f 72 0a 00 61 70  |ard %s error..a|
I (85846) slave: 0x40108734   30 00 00 00 00 00 96 30  07 77 2c 61 0e ee ba 51  |0......0.w,a...|
I (85855) slave: 0x40108744   09 99 19 c4 6d 07 8f f4  6a 70 35 a5              |....m...jp5.|
```

outermanle, 2018-10-22:
I also compare the data, esp32 tx differs from esp8266 rx.
The baudrate is too high?

#### esp32 tx
```c
I (89232) ota: Wrote [1024] bytes
I (89242) ota: uart rx timeout
I (89342) ota: uart rx timeout
I (89442) uart-rx: cmd ret=[ota=1013], len=8
I (89442) uart-rx: buf=[ota=1013] ,ret=[ota], ret_cmd=[1013]
W (89442) ota: ota result=[1013]
E (89442) ota: write=1024, return=1013
E (89442) httpd: esp8266_ota_write error ESP_FAIL!
I (89452) ota: send ota ret=1013 succeed.
I (89452) httpd: 0x3ffda4d0   38 c3 0c 0c d7 63 0f 31  83 fd 0c 64 01 f5 c5 c0  |8....c.1...d....|
I (89462) httpd: 0x3ffda4e0   00 00 0c 13 20 c3 83 08  31 2d 0c c8 21 12 c1 10  |.... ...1-..!...|
I (89472) httpd: 0x3ffda4f0   0d f0 19 94 fe 3f 1c 94  fe 3f f4 55 22 40 ec 55  |.....?...?.U"@.U|
I (89482) httpd: 0x3ffda500   22 40 12 c1 e0 31 8d f9  09 71 c9 61 d9 51 32 03  |"@...1...q.a.Q2.|
I (89492) httpd: 0x3ffda510   35 16 93 09 30 63 00 41  3a ff 2a 44 52 04 00 8c  |5...0c.A:.*DR...|
I (89502) httpd: 0x3ffda520   65 30 e6 13 0c 03 86 1f  00 0c 15 52 44 00 30 e6  |e0.........RD.0.|
I (89512) httpd: 0x3ffda530   13 60 63 00 31 35 ff 42  03 00 1b 54 50 50 74 52  |.`c.15.B...TPPtR|
I (89522) httpd: 0x3ffda540   43 00 b6 55 08 42 43 00  0c 03 46 07 00 00 41 e9  |C..U.BC...F...A.|
I (89532) httpd: 0x3ffda550   ff 31 e9 ff 52 04 00 30  35 b0 1b 55 50 50 74 f6  |.1..R..05..UPPt.|
I (89542) httpd: 0x3ffda560   45 03 86 00 00 00 0c 05  52 44 00 60 e6 13 39 01  |E.......RD.`..9.|
I (89552) httpd: 0x3ffda570   cc 43 21 e2 ff 86 0a 00  29 03 21 d6 c2 38 02 21  |.C!.....).!..8.!|
...

```

#### esp8266 rx wrong data 
``` c
I (91594) slave: 0x4010875c   38 c3 0c 0c d7 63 0f 31  83 fd 0c 64 01 f5 c5 c0  |8....c.1...d....|
I (91594) slave: 0x4010876c   00 00 0c 13 20 c3 83 08  31 2d 0c c8 21 12 c1 10  |.... ...1-..!...|
I (91603) slave: 0x4010877c   0d f0 19 94 fe 3f 1c 94  fe 3f f4 55 22 40 ec 55  |.....?...?.U"@.U|
I (91612) slave: 0x4010878c   22 40 12 c1 e0 31 8d f9  09 71 c9 61 d9 51 32 03  |"@...1...q.a.Q2.|
I (91621) slave: 0x4010879c   35 16 93 09 30 63 00 41  3a ff 2a 44 52 04 00 8c  |5...0c.A:.*DR...|
I (91630) slave: 0x401087ac   65 30 e6 13 0c 03 86 1f  00 0c 15 52 44 00 30 e6  |e0.........RD.0.|
I (91640) slave: 0x401087bc   13 60 63 00 31 35 ff 42  03 00 1b 54 50 50 74 52  |.`c.15.B...TPPtR|
I (91659) slave: 0x401087cc   43 00 b6 55 08 42 43 00  0c 03 46 07 00 00 41 e9  |C..U.BC...F...A.|
**I (91668) slave: 0x401087dc   55 50 50 74 f6 45 03 86  00 00 00 0c 05 52 44 00  |UPPt.E.......RD.|**differ
I (91677) slave: 0x401087ec   60 e6 13 39 01 cc 43 21  e2 ff 86 0a 00 29 03 21  |`..9..C!.....).!|
I (91687) slave: 0x401087fc   d6 c2 38 02 21 0d fd c2  23 19 38 83 d8 02 22 a0  |..8.!...#.8...".|
...
```

negativekelvin, 2018-10-23:
How are you doing flow control?

outermanle, 2018-10-23:
No hardware flow control.
With flow control between  packets transfer.

No software flow control during one transfer but set rx-buffer to max packet len for buffering.
max packet length 1024 bytes, get errors even set to 256bytes, need to do flow control?

negativekelvin, 2018-10-23:
If esp8266 is dropping bytes at high baud it sounds like you need to lower rx fifo full threshold

outermanle, 2018-10-23:
@negativekelvin

It works well after adjust the rx fifo full threshold.
Thank you so much! 

And I also want to get the error rate of baudrate from 78400 to 921600 about esp32(40MHz xtal) and esp8266(26Mhz xtal), do you have this specification or formula about clock division?

negativekelvin, 2018-10-23:
Both will usually use source clock of 80mhz but esp32 has fractional divider and esp8266 has integer divider