espressif/esp-idf #2616: How to use SDCard in SPI mode & spi_master at the same time on the M5Stack(esp32 core)?

Abol-Wang, 2018-10-23:
It looks like spi_master and sd card in spi mode (sdpi_host) cannot be used at the same time.

I've tested combining the "iot_lcd.cpp" from <esp-iot-solution> and sd card example and I cannot get both working.

/**********************************************GPIO**********************************************/
My LCD GPIO:
RST-GPIO23 DC-GPIO27 MOSI-GPIO23 SCK-GPIO18 CS-GPIO14

My SD SPI GPIO:
CS-GPIO4 MOSI-GPIO23 SDDO-GPIO19 CLK-GPIO18

/****************************************Test configuration:************************************/

SDcard and display use the different pins for spi interface.
Display uses HSPI_HOST and DMA channel 1.
SDCard uses VSPI_HOST an DMA channel 2.
Depending on which one is accessed first, either sdcard or display does not work.

When the sdcard is mounted first, sd card is detected, card info is correct, file can be opened, read, written...
Then the display is initialized.
After that, trying to open the file fails with:

E (2800) sdspi_host: sdspi_host_start_command: cmd=24 error=0x107
D (2800) sdmmc_cmd: sdmmc_req_run returned 0x107
E (2800) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
E (2810) diskio_sdmmc: sdmmc_write_blocks failed (263)
Display work with no problem.

When the display is intialized first, after the initialization the first display test works as expected, display is initialized and the pattern drawn.
Then the sdcard is mounted and works as expected, card info is correct, file can be opened, read, written...
After the sdcard intialization the display no longer works.

/***********************************************My Code****************************************************/
static CEspLcd* lcd_obj = NULL;
static const char *TAG = "sdcard";

#define PIN_NUM_MISO 19
#define PIN_NUM_MOSI 23
#define PIN_NUM_CLK  18
#define PIN_NUM_CS   4

#define SDSPI_HOST_M5STACK() {\
    .flags = SDMMC_HOST_FLAG_SPI, \
    .slot = VSPI_HOST, \
    .max_freq_khz = SDMMC_FREQ_DEFAULT, \
    .io_voltage = 3.3f, \
    .init = &sdspi_host_init, \
    .set_bus_width = NULL, \
    .get_bus_width = NULL, \
    .set_card_clk = &sdspi_host_set_card_clk, \
    .do_transaction = &sdspi_host_do_transaction, \
    .deinit = &sdspi_host_deinit, \
    .io_int_enable = NULL, \
    .io_int_wait = NULL, \
    .command_timeout_ms = 0, \
}

#define SDSPI_SLOT_CONFIG_M5STACK() {\
    .gpio_miso = PIN_NUM_MISO, \
    .gpio_mosi = PIN_NUM_MOSI, \
    .gpio_sck  = PIN_NUM_CLK, \
    .gpio_cs   = PIN_NUM_CS, \
    .gpio_cd   = SDSPI_SLOT_NO_CD, \
    .gpio_wp   = SDSPI_SLOT_NO_WP, \
    .dma_channel = 2 \
}
void sdcard_main(void)
{
    ESP_LOGI(TAG, "Initializing SD card");

    ESP_LOGI(TAG, "Using SPI peripheral");

    sdmmc_host_t host = SDSPI_HOST_M5STACK();
    sdspi_slot_config_t slot_config = SDSPI_SLOT_CONFIG_M5STACK();
    slot_config.gpio_miso = PIN_NUM_MISO;
    slot_config.gpio_mosi = PIN_NUM_MOSI;
    slot_config.gpio_sck  = PIN_NUM_CLK;
    slot_config.gpio_cs   = PIN_NUM_CS;
    // This initializes the slot without card detect (CD) and write protect (WP) signals.
    // Modify slot_config.gpio_cd and slot_config.gpio_wp if your board has these signals.

    // Options for mounting the filesystem.
    // If format_if_mount_failed is set to true, SD card will be partitioned and
    // formatted in case when mounting fails.
    esp_vfs_fat_sdmmc_mount_config_t mount_config = {
        .format_if_mount_failed = false,
        .max_files = 5,
        .allocation_unit_size = 16 * 1024
    };

    // Use settings defined above to initialize SD card and mount FAT filesystem.
    // Note: esp_vfs_fat_sdmmc_mount is an all-in-one convenience function.
    // Please check its source code and implement error recovery when developing
    // production applications.
    sdmmc_card_t* card;
    esp_err_t ret = esp_vfs_fat_sdmmc_mount("/sdcard", &host, &slot_config, &mount_config, &card);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            ESP_LOGE(TAG, "Failed to mount filesystem. "
                "If you want the card to be formatted, set format_if_mount_failed = true.");
        } else {
            ESP_LOGE(TAG, "Failed to initialize the card (%s). "
                "Make sure SD card lines have pull-up resistors in place.", esp_err_to_name(ret));
        }
        return;
    }

    // Card has been initialized, print its properties
    sdmmc_card_print_info(stdout, card);

    // Use POSIX and C standard library functions to work with files.
    // First create a file.
    ESP_LOGI(TAG, "Opening file");
    FILE* f = fopen("/sdcard/hello.txt", "w");
    if (f == NULL) {
        ESP_LOGE(TAG, "Failed to open file for writing");
        return;
    }
    fprintf(f, "Hello %s!\n", card->cid.name);
    fclose(f);
    ESP_LOGI(TAG, "File written");

    // Check if destination file exists before renaming
    struct stat st;
    if (stat("/sdcard/foo.txt", &st) == 0) {
        // Delete it if it exists
        unlink("/sdcard/foo.txt");
    }

    // Rename original file
    ESP_LOGI(TAG, "Renaming file");
    if (rename("/sdcard/hello.txt", "/sdcard/foo.txt") != 0) {
        ESP_LOGE(TAG, "Rename failed");
        return;
    }

    // Open renamed file for reading
    ESP_LOGI(TAG, "Reading file");
    f = fopen("/sdcard/foo.txt", "r");
    if (f == NULL) {
        ESP_LOGE(TAG, "Failed to open file for reading");
        return;
    }
    char line[64];
    fgets(line, sizeof(line), f);
    fclose(f);
    // strip newline
    char* pos = strchr(line, '\n');
    if (pos) {
        *pos = '\0';
    }
    ESP_LOGI(TAG, "Read from file: '%s'", line);

    // All done, unmount partition and disable SDMMC or SPI peripheral
    esp_vfs_fat_sdmmc_unmount();
    ESP_LOGI(TAG, "Card unmounted");
}

extern "C" void app_main(void *pvParameter)
{
    /*Initialize LCD*/
    lcd_conf_t lcd_pins = {
        .lcd_model    = LCD_MOD_ILI9341,
        .pin_num_miso = GPIO_NUM_19,
        .pin_num_mosi = GPIO_NUM_23,
        .pin_num_clk  = GPIO_NUM_18,
        .pin_num_cs   = GPIO_NUM_14,
        .pin_num_dc   = GPIO_NUM_27,
        .pin_num_rst  = GPIO_NUM_33,
        .pin_num_bckl = GPIO_NUM_32,
        .clk_freq     = 40 * 1000 * 1000,
        .rst_active_level = 0,
        .bckl_active_level = 1,
        .spi_host = HSPI_HOST,
        .init_spi_bus = true,
    };

    if (lcd_obj == NULL) {
        lcd_obj = new CEspLcd(&lcd_pins);
    }

    lcd_obj->setRotation(1);
    lcd_obj->fillScreen(COLOR_ESP_BKGD);           
    
    //LCD APIS test is ok!!
    sdcard_main();
    //SD write Read file is ok!!

    //ALL LCD APIS test is invalid.
    lcd_obj->fillScreen(COLOR_BLUE);

    vTaskDelay(2000 / portTICK_PERIOD_MS);
}

Abol-Wang, 2018-10-23:
Do you have a suitable solution? 
This question really bothers me to use esp32 on the M5Stack.

igrr, 2018-10-23:
Closing as duplicate of https://github.com/espressif/esp-idf/issues/1597. Please follow the link to this issue and click "Subscribe" to receive further update.

closed 2018-10-23 by igrr