espressif/esp-idf #2618: SYSTEM_EVENT_AP_STADISCONNECTED event is not triggered while running a OTA to a ESP32 running softAP

agmuntianu, 2018-10-23:
## Environment
- Development Kit:      		[none]
- Core (if using chip or module): [ESP-WROOM-32D]
- IDF version:   			22da5f6de ( v3.1-rc2 tag)
- Development Env: 		[Make|Eclipse]
- Operating System:		[Ubuntu]
- Power Supply:         [external 3.3V]

## Problem Description
If using the ESP32 softAP to upload a new firmware using a POST / OTA, I get a lot of failed flashing attempts if I don't wait for several seconds between initializing the OTA update and begin the flashing itself. Below is the log for my workaround, but I believe a bug in softAP might be the root cause ( something related to issue [2375](https://github.com/espressif/esp-idf/issues/2375)  ):
- here I turn the access point on
```
I (12446) AP: turn on AP
I (12446) wifi: mode : softAP (24:0a:c4:1e:06:e5)
I (12451) wifi: Access point started
```
- here I connect my PC to the access point 
```
I (26715) wifi: n:3 1, o:3 0, ap:3 1, sta:255 255, prof:3
I (26716) wifi: station: 90:2e:1c:a7:81:68 join, AID=1, bgn, 40U
I (26719) wifi: Access point connected
I (26743) tcpip_adapter: softAP assign IP to station,IP is: 192.168.10.2
```
-here I begin the POST - OTA firmware upload
```
I (35681) ota: http_request HTTP/1.1
Host: 192.168.10.1
Connection: keep-alive
Content-Length: 1331934 length 487
I (35682) post: content_len 1331934
I (35686) ota: Running partition type 0 subtype 17 (offset 0x00620000)
I (35693) ota: Writing to partition subtype 16 at offset 0x320000
I (35700) ota: number of connected stations: 1
I (35713) ota: number of connected stations: 1
I (35713) ota: Waiting 10 sec because of WiFi soft-AP
I (39864) wifi: n:3 1, o:3 1, ap:3 1, sta:255 255, prof:3
I (39865) wifi: station: 90:2e:1c:a7:81:68 join, AID=1, bgn, 40U
I (39866) wifi: Access point connected
I (45713) ota: number of connected stations: 1
I (45713) ota: Begin OTA
```
Note:  The event `SYSTEM_EVENT_AP_STACONNECTED` is called after `esp_ota_begin` function is called (see snippet bellow )

OTA over POST code snippet:
```cpp
    ESP_LOGI("ota", "http_request %s length %d", http_request, http_request_len);
    ESP_LOGI("post", "content_len %d", content_len);

    esp_err_t err;
    const esp_partition_t *configured = esp_ota_get_boot_partition();
    const esp_partition_t *running = esp_ota_get_running_partition();
    if (configured != running) {
        ESP_LOGW("ota", "Configured OTA boot partition at offset 0x%08x, but running from offset 0x%08x",
                 configured->address, running->address);
        ESP_LOGW("ota", "(This can happen if either the OTA boot data or preferred boot image become corrupted somehow.)");
    }
    ESP_LOGI("ota", "Running partition type %d subtype %d (offset 0x%08x)",
             running->type, running->subtype, running->address);

    update_partition = esp_ota_get_next_update_partition(NULL);
    ESP_LOGI("ota", "Writing to partition subtype %d at offset 0x%x",
             update_partition->subtype, update_partition->address);
    assert(update_partition != NULL);
    wifi_sta_list_t stations;
    ESP_ERROR_CHECK(esp_wifi_ap_get_sta_list(&stations));
    ESP_LOGI("ota","number of connected stations: %d",stations.num);
    err = esp_ota_begin(update_partition, OTA_SIZE_UNKNOWN, &update_handle);
    if (err != ESP_OK) {
        ESP_LOGE("ota", "esp_ota_begin failed (%s)", esp_err_to_name(err));
        return ERR_ABRT;
    }
    ESP_ERROR_CHECK(esp_wifi_ap_get_sta_list(&stations));
    ESP_LOGI("ota","number of connected stations: %d",stations.num);

    EventBits_t uxBits;
    uxBits = xEventGroupWaitBits(event_group, STA_CONNECTED_BIT, false, false, (TickType_t)0);
    if ((uxBits & STA_CONNECTED_BIT) == STA_CONNECTED_BIT) {
        //we use wifi so we need to wait 10 seconds for connection re-establish (wifi soft ap workaround)
        ESP_LOGI("ota", "Waiting 10 sec because of WiFi soft-AP");
        vTaskDelay(10000 / portTICK_PERIOD_MS);
    }
    ESP_LOGI("ota", "Begin OTA");
```
### Expected Behavior
If indeed the station disconnects from the softAP after `esp_ota_begin` is called, I would expect at least the event `SYSTEM_EVENT_AP_STADISCONNECTED` to be triggered , but it does not.

### Actual Behavior
Without the 10 seconds delay, the flashing fails with `esp_image: invalid segment length 0xffffffff` error, because the `SYSTEM_EVENT_AP_STACONNECTED` event gets triggered sometimes during the flashing.

BR. Andrei.

agmuntianu, 2018-10-23:
Another thing I noticed : If I have the access point started and the computer connected, but I program the ESP32 using ethernet , I still get that `SYSTEM_EVENT_AP_STACONNECTED ` event.

negativekelvin, 2018-10-23:
Problem is long running flash erase cycle with no breaks cause ap issues

agmuntianu, 2018-10-23:
so basically the `esp_ota_begin` takes too long ?

outermanle, 2018-10-24:
I think it should be done some optimization to ota procedure, maybe just add some delays, as @negativekelvin said " Problem is long running flash erase cycle with no breaks cause ap issues".

I also encounter this issue ocassionally, during ota procedure, uart maybe drop data and other things maybe stuck or trigger watchdog sometimes in the ota process etc...

Do less thing during ota procedure.

agmuntianu, 2018-10-24:
@outermanle I actually tried some "workarounds":
1. I erased the ota partition in batches of 32k/64k and added a small delay between. This worked fine,  but the erase procedure ended up taking a lot of time (~10 seconds, instead of the ~2 seconds when doing the erase in one go).
2. I increased the beacon interval for softAP to 1.5 seconds, which basically hides to any station connected to the esp32 the fact the controller is _dead_ . I will use this workaround.

Also between the writing of the OTA data, I wait 1ms to prevent the watchdog from triggering , like you mention.

As it's clearly a hardware limitation, I guess I will just close the issue, as it cannot be fixed.

closed 2018-10-24 by agmuntianu