espressif/esp-idf #2619: Simultaneous WiFi AP + BT Classic slave causes stations to disconnect

jakkra, 2018-10-23:
## Environment

- Development Kit:      [none]
- Core (if using chip or module): [ESP32]
- IDF version f11ac037c4958fc68e67a50134a0ce9a1634008c
- Development Env:		[Make]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description
Accesspoint loses connection to its stations if a device connects to it over SPP while it has stations connected to it.

Afterwards the station cannot connect to the AP until the ESP32 AP disconnects the SPP link.

### Expected Behavior
- ESP32 AP does not lose connection to stations when the ESP32 is also Classic Slave.
- WiFi stations can connect to an ESP32 AP with an active SPP connection (where the AP also is BT classic slave)
### Actual Behavior
- ESP32 AP loses connection to its stations when someone connects to it over BT Classic/SPP 
- Stations can't connect to ESP32 AP which also has a classic connection, in which it is classic slave.
### Steps to repropduce

Device1: AP and SPP server
Device2: Wifi station
Device3: SPP client

1. Set up a device1 to AP and also start a Bluetooth Classic SPP server.
2. Connect as Wifi station with device2 to device1.
3. Connect to device1 from device3 over Bluetooth classic SPP.
4. Wait ~5 seconds
5. Now device2 will lose connection to the AP(device1)

In above example, if device1 is AP + SPP client instead and device3 is SPP server and the SPP connection is initiated from device1 instead, it does not lose the connection to its stations.

## Debug Logs
Only print I get:
`wlan:01884: esp event: SYSTEM_EVENT_AP_STADISCONNECTED
`
## Other items if possible

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/2505123/sdkconfig.zip)

TianHao-Espressif, 2018-11-02:
@jakkra , Unfortunately, as far as now,  ESP32 doesn't support ESP32 softap mode and Classic BT coexistence. We are evaluating the coexistence of them.

jakkra, 2018-11-02:
@TianHao-Espressif Thank you. I see, is there a list with what is supported and not in coexistance? Since we are building an application with coexist, it would be very helpful knowing what is supported and not.

TianHao-Espressif, 2018-11-07:
@jakkra , As far as now, ESP32 support:
*  WiFi STA +  BLE
* WiFi STA + Classic BT
* Ethernet + Any  (Ethernet do not use wireless RF)

closed 2020-04-27 by jakkra