espressif/esp-idf #2620: OTA failed when using ETH.

welinkio, 2018-10-23:
Download will fail after receive some data. 
test code on commit 129d32772e5e5eafe88be5b9eb34687e84a6f8b8

suda-morris, 2018-10-24:
Hi, @welinkio Can you show me more info about this issue?like your sdkconfig,partition table, log output?

welinkio, 2018-10-25:
I am using LAN8720 PHY chip, attached is sdkconfig and partition info，see ota_fail_log.txt.

[ota_fail_log.txt](https://github.com/espressif/esp-idf/files/2515478/ota_fail_log.txt)

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/2515128/sdkconfig.zip)

welinkio, 2018-10-25:
[native_example_ota_fail.txt](https://github.com/espressif/esp-idf/files/2515560/native_example_ota_fail.txt)

using native ota function log. @suda-morris

suda-morris, 2018-10-30:
hi @welinkio What if you check on the 
`Component config->Ethernet->Enable copy between Layer2 and Layer3` ?

suda-morris, 2018-10-31:
@welinkio Thanks again, the bug has been fixed and will be merged into the master branch soon.

closed 2018-11-18 by igrr