espressif/esp-idf #2625: ESP32 OTA failure

iw2lsi, 2018-10-24:
Hi all,

  I'm in trouble trying to run the ESP32 OTA examples (both native and simple)... my ESP32_DevKitC v4
 can succesfully connect to the WIFI and to the web-server using SSL, but then it fails in retrieving the image and/or while flashing it...

Below is the logfile produced by "make monitor"

```
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from C:/esp32_msys2/msys32/home/Giampaolo/esp/esp-idf/requirements.txt are satisfied.
MONITOR
--- idf_monitor on COM24 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6268
load:0x40078000,len:10084
load:0x40080400,len:6476
entry 0x40080764
I (28) boot: ESP-IDF v3.2-dev-1250-g221eced06 2nd stage bootloader
I (29) boot: compile time 12:24:41
I (29) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (65) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (73) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  3 factory          factory app      00 00 00010000 00100000
I (88) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (95) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (103) boot: End of partition table
I (107) boot: Defaulting to factory image
I (111) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1d9d0 (121296) map
I (163) esp_image: segment 1: paddr=0x0002d9f8 vaddr=0x3ff80000 size=0x00000 (     0) load
I (163) esp_image: segment 2: paddr=0x0002da00 vaddr=0x3ff80000 size=0x00000 (     0) load
I (170) esp_image: segment 3: paddr=0x0002da08 vaddr=0x3ffb0000 size=0x02608 (  9736) load
I (182) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x84ec4 (544452) map
0x400d0018: _stext at ??:?

I (378) esp_image: segment 5: paddr=0x000b4ee4 vaddr=0x3ffb2608 size=0x00e00 (  3584) load
I (380) esp_image: segment 6: paddr=0x000b5cec vaddr=0x3ffb3408 size=0x00000 (     0) load
I (385) esp_image: segment 7: paddr=0x000b5cf4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/esp32_msys2/msys32/home/Giampaolo/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

I (394) esp_image: segment 8: paddr=0x000b60fc vaddr=0x40080400 size=0x10da4 ( 69028) load
I (431) esp_image: segment 9: paddr=0x000c6ea8 vaddr=0x400c0000 size=0x00000 (     0) load
I (432) esp_image: segment 10: paddr=0x000c6eb0 vaddr=0x50000000 size=0x00000 (     0) load
I (438) esp_image: segment 11: paddr=0x000c6eb8 vaddr=0x50000000 size=0x00000 (     0) load
I (457) boot: Loaded app from partition at offset 0x10000
I (458) boot: Disabling RNG early entropy source...
I (459) cpu_start: Pro cpu up.
I (463) cpu_start: Starting app cpu, entry point is 0x40081014
0x40081014: call_start_cpu1 at C:/esp32_msys2/msys32/home/Giampaolo/esp/esp-idf/components/esp32/cpu_start.c:234

I (0) cpu_start: App cpu up.
I (473) heap_init: Initializing. RAM available for dynamic allocation:
I (480) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (486) heap_init: At 3FFB95D8 len 00026A28 (154 KiB): DRAM
I (492) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (499) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (505) heap_init: At 400911A4 len 0000EE5C (59 KiB): IRAM
I (511) cpu_start: Pro cpu start user code
I (194) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (279) wifi: wifi driver task: 3ffc1104, prio:23, stack:3584, core=0
I (279) wifi: wifi firmware version: 75aab56
I (279) wifi: config NVS flash: enabled
I (279) wifi: config nano formating: disabled
I (289) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (299) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (329) wifi: Init dynamic tx buffer num: 32
I (329) wifi: Init data frame dynamic rx buffer num: 32
I (329) wifi: Init management frame dynamic rx buffer num: 32
I (329) wifi: Init static rx buffer size: 1600
I (339) wifi: Init static rx buffer num: 10
I (339) wifi: Init dynamic rx buffer num: 32
I (339) simple_ota_example: Setting WiFi configuration SSID MIC extended...
I (409) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 0
I (419) wifi: mode : sta (84:0d:8e:0c:f8:d4)
I (419) simple_ota_example: Starting OTA example...
I (539) wifi: n:1 1, o:1 0, ap:255 255, sta:1 1, prof:1
I (1519) wifi: state: init -> auth (b0)
I (1529) wifi: state: auth -> assoc (0)
I (1529) wifi: state: assoc -> run (10)
I (1559) wifi: connected with MIC extended, channel 1
I (1559) wifi: pm start, type: 1

I (2779) event: sta ip: 192.168.1.159, mask: 255.255.255.0, gw: 192.168.1.254
I (2779) simple_ota_example: Connect to Wifi ! Start to Connect to Server....
I (5119) esp_https_ota: Starting OTA...
I (5119) esp_https_ota: Writing to partition subtype 16 at offset 0x110000
I (5139) esp_https_ota: esp_ota_begin succeeded
I (5139) esp_https_ota: Please Wait. This may take time
E (5139) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
E (5149) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
I (5149) esp_https_ota: Connection closed,all data received
I (5159) esp_image: segment 0: paddr=0x00110020 vaddr=0x3f400020 size=0x07c2c ( 31788) map
E (5189) esp_image: invalid segment length 0xffffffff
E (5189) esp_https_ota: Error: esp_ota_end failed! err=0x5379. Image is invalid
E (5199) simple_ota_example: Firmware Upgrades Failed

```

On the server side, 

```
$ winpty openssl s_server -WWW -key ca_key.pem -cert ca_cert.pem -port 8070
Enter pass phrase for ca_key.pem:
Using default temp DH parameters
ACCEPT
FILE:hello-world.bin

```

   anyone is having the same problem ?

I'm using windows 10 with latest IDF...

               thanks a lot

                    Giampaolo

iw2lsi, 2018-10-24:
Hi all,

   one more info... if I try to download the firmware image directly either using curl or a browser, the downloaded image seems to be smaller comparing to the original one (1k instead of 150k).

This will probably explain why the native_ota_example fails to flash the image... but I really cannot figure out what's the problem with the openssl server... and disabling windows firewall did not help.

           Giampaolo

jitin17, 2018-10-24:
@iw2lsi There seems to some problem in openssl server, so can you try setting up the server with this python script.
```
import BaseHTTPServer, SimpleHTTPServer
import ssl
import argparse
import os

parser = argparse.ArgumentParser()
parser.add_argument('-p', '--port', dest='port', type= int,
    help= "Server Port", default= 8000)
args = parser.parse_args()

httpd = BaseHTTPServer.HTTPServer(('', args.port),
        SimpleHTTPServer.SimpleHTTPRequestHandler)

httpd.socket = ssl.wrap_socket (httpd.socket,
        keyfile= "ca_key.pem",
        certfile= "ca_cert.pem", server_side=True)
print "Serving on 0.0.0.0 port ", args.port
httpd.serve_forever()

```
Copy this to a `.py` file in the directory containing the image and run the python script.
Then try OTA example again.

And also change the log output level to `debug` under `make menuconfig` -> `Component config` -> `Log output` ->  `Default log verbosity`

iw2lsi, 2018-10-24:
Hi jitin17,

   yes... you are right... I've just made a small test using a raspberry pi-zero-w as SSL server and... it works!

        thanks for your help,

              Giampaolo

closed 2018-10-25 by Alvin1Zhang