espressif/esp-idf #2627: is UART wakeup on arbitrary pins possible?

bearsh, 2018-10-24:
## Environment

- Development Kit:      none
- Kit version (for WroverKit/PicoKit/DevKitC): -
- Core (if using chip or module): ESP32
- IDF version: v3.2-dev-1385-g129d32772
- Development Env: Eclipse|other
- Operating System: Ubuntu
- Power Supply: USB

## Problem Description

I want to use the UART wakeup feature on UART1 but with non native uart pins (I use pin 21 and 33 routed through GPIO matrix).

Is this supposed to work? If not, please note it in the documentation.

negativekelvin, 2018-10-25:
@koobest v3.2-dev-1385-g129d32772 is master from last week.  Have you verified uart wake on any pin?

bearsh, 2018-10-25:
> @bearsh, v3.2 does not support uart wakeup feature, please try with master branch.

yes I know, therefor I am on master, see mentioned IDF version...

> refer to the API
> 
> [esp-idf/components/esp32/include/esp_sleep.h](https://github.com/espressif/esp-idf/blob/7c29a39d6f9f2dfbefc49d34d34e9267afc7200d/components/esp32/include/esp_sleep.h#L231)
> 
> Line 231 in [7c29a39](/espressif/esp-idf/commit/7c29a39d6f9f2dfbefc49d34d34e9267afc7200d)
> 
>  esp_err_t esp_sleep_enable_uart_wakeup(int uart_num); 

I tried this API and it also works using the console example but not with my code where I use UART1 on routed GPIOs. So I fear this feature only works on _native_ pins.

koobest, 2018-10-25:
Hi, I have confirmed with our hardware teem that this feature only works on native pins.

bearsh, 2018-10-25:
@koobest thanks for the confirmation!

closed 2018-11-28 by igrr