espressif/esp-idf #2628: I2S in the ADC mode return incorrect values

alxkbr, 2018-10-24:
Hi all,

I got different raw values after reading data directly from ADC and from I2S in the ADC mode.
Real constant voltage measured with a multimeter - 1690mV.

Below is my source code

```cpp
#include "freertos/FreeRTOS.h"
#include "esp_wifi.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_log.h"
#include "esp_event_loop.h"
#include "nvs_flash.h"
#include "driver/gpio.h"

#include "sdkconfig.h"
#include "driver/adc.h"
#include "driver/i2s.h"
#include "esp_adc_cal.h"

#define ADC_INT_VREF    1100
#define ADC_CHANNEL     ADC1_CHANNEL_0
#define ADC_WIDTH       ADC_WIDTH_BIT_12
#define ADC_ATTEN       ADC_ATTEN_DB_11
#define ADC_UNIT        ADC_UNIT_1
#define I2S_NUM         I2S_NUM_0

static const char *TAG = "main";

void adc_read_raw()
{
  esp_adc_cal_characteristics_t adc_chars;

  adc1_config_width(ADC_WIDTH);
  adc1_config_channel_atten(ADC_CHANNEL, ADC_ATTEN);
  esp_adc_cal_characterize(ADC_UNIT, ADC_ATTEN, ADC_WIDTH, ADC_INT_VREF, &adc_chars);

  for(int i = 0; i < 10; i++)
  {
    int raw = adc1_get_raw(ADC_CHANNEL);
    ESP_LOGI(TAG, "adc sample %d: raw %d, voltage %d mV", i, raw, esp_adc_cal_raw_to_voltage(raw, &adc_chars));
  }
}

void adc_dma_i2s_read_raw()
{
  esp_adc_cal_characteristics_t adc_chars;
  int16_t buf[10] = {0};
  size_t n;

  esp_adc_cal_characterize(ADC_UNIT, ADC_ATTEN, ADC_WIDTH, ADC_INT_VREF, &adc_chars);

  i2s_config_t i2s_config = {
    .mode = I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN,
    .sample_rate = 16000,
    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
    .channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT,
    .communication_format = I2S_COMM_FORMAT_I2S_MSB,
    .intr_alloc_flags = 0,
    .dma_buf_count = 2,
    .dma_buf_len = 1024,
  };
  ESP_ERROR_CHECK(i2s_driver_install(I2S_NUM, &i2s_config, 0, NULL));
  ESP_ERROR_CHECK(i2s_set_adc_mode(ADC_UNIT, ADC_CHANNEL));

  i2s_adc_enable(I2S_NUM);
  {
    i2s_read(I2S_NUM, buf, sizeof(buf), &n, portMAX_DELAY);
  }
  i2s_adc_disable(I2S_NUM);

  for(uint32_t i = 0; i < (n / sizeof(buf[0])); i++)
  {
    int raw = buf[i] & 0xfff;
    ESP_LOGI(TAG, "adc dma i2s sample %d: raw %d, voltage %d mV", i, raw, esp_adc_cal_raw_to_voltage(raw, &adc_chars));
  }
}

void app_main()
{
  adc_read_raw();
  adc_dma_i2s_read_raw();
}
```

Output

```
I (0) cpu_start: App cpu up.
I (106) heap_init: Initializing. RAM available for dynamic allocation:
I (113) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (119) heap_init: At 3FFB2F08 len 0002D0F8 (180 KiB): DRAM
I (125) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (131) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (138) heap_init: At 400896A8 len 00016958 (90 KiB): IRAM
I (144) cpu_start: Pro cpu start user code
I (162) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (165) main: adc sample 0: raw 1939, voltage 1704 mV
I (169) main: adc sample 1: raw 1961, voltage 1722 mV
I (174) main: adc sample 2: raw 1990, voltage 1745 mV
I (180) main: adc sample 3: raw 1983, voltage 1740 mV
I (186) main: adc sample 4: raw 1937, voltage 1703 mV
I (191) main: adc sample 5: raw 1950, voltage 1713 mV
I (197) main: adc sample 6: raw 1939, voltage 1704 mV
I (203) main: adc sample 7: raw 1967, voltage 1727 mV
I (209) main: adc sample 8: raw 1954, voltage 1716 mV
I (214) main: adc sample 9: raw 1983, voltage 1740 mV
I (221) I2S: DMA Malloc info, datalen=blocksize=2048, dma_buf_count=2
I (227) I2S: PLL_D2: Req RATE: 16000, real rate: 1004.000, BITS: 16, CLKM: 83, BCK: 60, MCLK: 83.333, SCLK: 32128.000000, diva: 64, divb: 21
I (241) I2S: PLL_D2: Req RATE: 16000, real rate: 1004.000, BITS: 16, CLKM: 83, BCK: 60, MCLK: 83.333, SCLK: 32128.000000, diva: 64, divb: 21
I (318) main: adc dma i2s sample 0: raw 2127, voltage 1856 mV
I (318) main: adc dma i2s sample 1: raw 2176, voltage 1895 mV
I (320) main: adc dma i2s sample 2: raw 2131, voltage 1859 mV
I (327) main: adc dma i2s sample 3: raw 2100, voltage 1834 mV
I (333) main: adc dma i2s sample 4: raw 2151, voltage 1875 mV
I (339) main: adc dma i2s sample 5: raw 2154, voltage 1877 mV
I (346) main: adc dma i2s sample 6: raw 2112, voltage 1843 mV
I (352) main: adc dma i2s sample 7: raw 2159, voltage 1881 mV
I (359) main: adc dma i2s sample 8: raw 2150, voltage 1874 mV
I (365) main: adc dma i2s sample 9: raw 2168, voltage 1889 mV
```

ESP-IDF commit 129d32772e5e5eafe88be5b9eb34687e84a6f8b8 HEAD -> master

[esp-adc-dma-i2s.zip](https://github.com/espressif/esp-idf/files/2510903/esp-adc-dma-i2s.zip)

v3xation, 2019-02-01:
From my testing it appears that the values returned over the i2s dma channel are bitwise negated. Taking i2s sample 0 from your output as an example: 2127 -> 0x84f --(negate)--> 0x7b0 --> 1968.

alxkbr, 2019-02-06:
@v3xation Thanks for the test. You're right, easy way to fix it:

```cpp
int16_t raw = ~buf[i] & 0xfff;
```

Please correct the function 'example_disp_buf' in the example 'esp-idf/examples/peripherals/i2s_adc_dac/main/app_main.c'.

closed 2019-02-06 by alxkbr