espressif/esp-idf #2635: calling tcpip_adapter_init() while light sleep is active increase current consumption

bearsh, 2018-10-26:
## Environment

- Development Kit: none
- Core (if using chip or module): ESP32
- IDF version: v3.2-dev-1385-g129d32772
- Operating System: Ubuntu
- Power Supply: USB

## Problem Description

Once `tcpip_adapter_init()` has been called and light sleep mode is active, the current consumption increases massively.

### Expected Behavior

No significant change in current consumption

### Actual Behavior

In light sleep, the current consumption increases from ~1.3mA to almost 4mA. There's periodic activity all ~100ms although the tcpip adapter is not used by the software.

The following image shows the current after tcpip_adapter_init() has been called but the software idles
[image]

igrr, 2018-10-27:
Calling tcpip_adapter_init initialized LwIP stack. The stack registers some timers, which execute even if there is no network traffic. So it seems like the expected behavior.

Michael2196, 2019-06-11:
Hello,
I'm having the same problem with the tcip_adapter_init() function. Did you find any solution ?
Thank you

igrr, 2019-11-18:
In IDF 4.0 and above, there is a new option to schedule LwIP timers on demand: https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/lwip.html#on-demand-timers.
(added in commit https://github.com/espressif/esp-idf/commit/ae675973beacdb250900c066510d3bac806c6f02).

closed 2019-11-18 by igrr