espressif/esp-idf #2637: Reboot with server BLE and WIFI

gigirizzi, 2018-10-26:
hi, i try to utilize  BLE and Wifi server, but ig i start BLE server  i have a loop of reset withi this error.

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
Help :(

negativekelvin, 2018-10-27:
That is not an error, you have to post the log that came before that.

me-no-dev, 2018-11-01:
Maybe your application got too big and you did not select proper partition scheme? Try one that supports large apps :)

Alvin1Zhang, 2019-03-15:
@gigirizzi Hi, sorry to get back to you so late. Does your issue still exist? Thanks.

Alvin1Zhang, 2019-05-20:
@gigirizzi Thanks for reporting the issue. Does your issue still exist? Thanks.

Alvin1Zhang, 2019-06-13:
@gigirizzi Thanks for reporting the issue, but will close now due to not enough details. Feel free to reopen if the issue happens. Thanks.

closed 2019-06-13 by Alvin1Zhang