espressif/esp-idf #2638: esp_http_client issue - Wrong content length set

AadiMehta, 2018-10-27:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (DevKitC): [v3]
- Core (if using chip or module): [ESP32-Wrover|ESP-WROOM32|ESP32]
- IDF version : 11b444b

## Problem Description
esp_http_client_open() API should set "Content-Length" header to the passed size. But the content-length header is either not set or it is set to 0. 

I am trying to send a file from esp32 as attachment to a server. Here is the same code for the same.

### Code to reproduce this issue

```cpp
FILE *f = fopen(filename, "r");
   if (f == NULL) {
      ESP_LOGE(TAG, "Failed to open file for sending");
      return -1;
   }

   esp_http_client_handle_t client = esp_http_client_init(&config);

   esp_http_client_set_header(client, "User-Agent", "abc-Node");
   esp_http_client_set_header(client, "Content-Disposition", "attachment; filename=chip.log");
   esp_http_client_set_url(client, url);
   esp_http_client_set_method(client, HTTP_METHOD_POST);

   esp_err_t err;
   if ((err = esp_http_client_open(client, size)) != ESP_OK) {
      ESP_LOGE(TAG, "Failed to open HTTP connection: %s", esp_err_to_name(err));
      free(buffer);
      return -1;
   }

   bzero(buffer, MAX_HTTP_RECV_BUFFER);
   printf("Total size : %d\n", size);

   while ((ret = fread(&buffer[0], sizeof(char), MAX_HTTP_RECV_BUFFER, f)) > 0) {
      esp_http_client_write(client, buffer, ret);
      bzero(buffer, MAX_HTTP_RECV_BUFFER);
   }
   int content_length =  esp_http_client_fetch_headers(client);
   int total_read_len = 0, read_len;
   if (total_read_len < content_length && content_length <= MAX_HTTP_RECV_BUFFER) {
      read_len = esp_http_client_read(client, buffer, content_length);
      if (read_len <= 0) {
         ESP_LOGE(TAG, "Error read data");
      }
      buffer[read_len] = 0;
      ESP_LOGI(TAG, "read_len = %d", read_len);
   }
   ESP_LOGI(TAG, "HTTP Status = %d, content_length = %d",
                       esp_http_client_get_status_code(client),
                       esp_http_client_get_content_length(client));
   esp_http_client_close(client);
   esp_http_client_cleanup(client);
   free(buffer);
   fclose(f);
```

## Debug Logs from server

```
{
	"CONTENT_LENGTH": "0",
	"SERVER_PROTOCOL": "HTTP/1.1",
	"REQUEST_METHOD": "POST",
	"PATH_INFO": "/logs/api/submit",
	"REMOTE_ADDR": "192.168.2.61",
	"CONTENT_TYPE": "text/plain",
	"HTTP_USER_AGENT": "abc-Node",
	"HTTP_HOST": "192.168.2.16",
	"HTTP_CONTENT_DISPOSITION": "attachment; filename=chip.log",
}
```

negativekelvin, 2018-10-28:
You are right, the value is not used at all.  It should do
`client->post_len=write_len;`
https://github.com/espressif/esp-idf/blob/3977d4b56555289f828a4d5f5d756b7b69783d53/components/esp_http_client/esp_http_client.c#L1085-L1095

mahavirj, 2019-03-05:
Fixed with https://github.com/espressif/esp-idf/commit/2e3f06ece8bffeda95ab9546d4356ed4d4b72ca8, closing.

closed 2019-03-05 by mahavirj