espressif/esp-idf #2639: mbedtls allocated in spiram issue

chegewara, 2018-10-27:
I am trying to work with new menuconfig->mbedTLS option to allocate from spiram, but i am having problems. 
I have free in spiram:
```
Heap summary for capabilities 0x00000400:
  At 0x3f800000 len 4194303 free 4071380 allocated 122880 min_free 4071380
    largest_free_block 4071380 alloc_blocks 1 free_blocks 1 total_blocks 2
  Totals:
    free 4071380 allocated 122880 min_free 4071380 largest_free_block 4071380
```
but mbedTLS cant create new connection even if certificate is verified:
```
E (64702) esp-tls: mbedtls_ssl_handshake returned -0x7780
I (64702) esp-tls: Certificate verified.
I (64702) camera_demo: waiting ....
E (64711) esp-tls: Failed to open new connection
E (64711) TRANS_SSL: Failed to open a new connection
E (64716) HTTP_CLIENT: Connection failed, sock < 0
E (64722) HTTP_CLIENT: HTTP POST request failed: ESP_ERR_HTTP_CONNECT
```
.
All works fine with option `allocation strategy -->internal memory`.
This is heap summary internal and spiram memory usage with spiram config `Make RAM allocatable using heap_caps_malloc`:
```
Heap summary for capabilities 0x00000800:
  At 0x3ffae6e0 len 6432 free 0 allocated 6284 min_free 0
    largest_free_block 0 alloc_blocks 29 free_blocks 0 total_blocks 29
  At 0x3ffb9d30 len 156368 free 3568 allocated 151912 min_free 0
    largest_free_block 3420 alloc_blocks 210 free_blocks 4 total_blocks 214
  At 0x3ffe0440 len 15296 free 7316 allocated 7844 min_free 5084
    largest_free_block 7260 alloc_blocks 24 free_blocks 2 total_blocks 26
  At 0x3ffe4350 len 113840 free 113804 allocated 0 min_free 113804
    largest_free_block 113804 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x40096ca4 len 37724 free 37688 allocated 0 min_free 37680
    largest_free_block 37688 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 162376 allocated 166040 min_free 156568 largest_free_block 113804  
Heap summary for capabilities 0x00000400:
  At 0x3f800000 len 4194303 free 4071380 allocated 122880 min_free 4071380
    largest_free_block 4071380 alloc_blocks 1 free_blocks 1 total_blocks 2
  Totals:
    free 4071380 allocated 122880 min_free 4071380 largest_free_block 4071380  //<--- memory allocated by my app only
```
This is heap summary internal and spiram memory usage with spiram config `Make RAM allocatable using malloc() as well`:
```
Heap summary for capabilities 0x00000800:
  At 0x3ffbd32c len 67999 free 61292 allocated 6596 min_free 61292
    largest_free_block 61292 alloc_blocks 18 free_blocks 1 total_blocks 19
  At 0x3ffae6e0 len 6432 free 0 allocated 6276 min_free 0
    largest_free_block 0 alloc_blocks 31 free_blocks 0 total_blocks 31
  At 0x3ffb9d88 len 156280 free 2980 allocated 152672 min_free 0
    largest_free_block 1996 alloc_blocks 145 free_blocks 4 total_blocks 149
  At 0x3ffe0440 len 15296 free 0 allocated 15228 min_free 0
    largest_free_block 0 alloc_blocks 9 free_blocks 0 total_blocks 9
  At 0x3ffe4350 len 113840 free 59664 allocated 53868 min_free 54044
    largest_free_block 59608 alloc_blocks 67 free_blocks 2 total_blocks 69
  At 0x40096dc4 len 37436 free 37400 allocated 0 min_free 37392
    largest_free_block 37400 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 161336 allocated 234640 min_free 152728 largest_free_block 61292 
Heap summary for capabilities 0x00000400:
  At 0x3f800000 len 4194303 free 4071380 allocated 122880 min_free 4071380
    largest_free_block 4071380 alloc_blocks 1 free_blocks 1 total_blocks 2
  Totals:
    free 4071380 allocated 122880 min_free 4071380 largest_free_block 4071380 //<--- memory allocated by my app only
```
This is heap summary with mbedTLS internal memory allocation strategy (normal usage):
```
Heap summary for capabilities 0x00000800:
  At 0x3ffbd32c len 32767 free 26060 allocated 6596 min_free 26060
    largest_free_block 26060 alloc_blocks 18 free_blocks 1 total_blocks 19
  At 0x3ffae6e0 len 6432 free 0 allocated 6276 min_free 0
    largest_free_block 0 alloc_blocks 31 free_blocks 0 total_blocks 31
  At 0x3ffb9d88 len 156280 free 3060 allocated 152560 min_free 0
    largest_free_block 1956 alloc_blocks 149 free_blocks 8 total_blocks 157
  At 0x3ffe0440 len 15296 free 16 allocated 15144 min_free 0
    largest_free_block 16 alloc_blocks 25 free_blocks 1 total_blocks 26
  At 0x3ffe4350 len 113840 free 94892 allocated 18716 min_free 87620
    largest_free_block 94832 alloc_blocks 47 free_blocks 3 total_blocks 50
  At 0x40096dc4 len 37436 free 37400 allocated 0 min_free 37392
    largest_free_block 37400 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 161428 allocated 199292 min_free 151072 largest_free_block 94832  // <--- more free RAM because only 32kB reserved for DMA, in first log no DMA reserved (no option), 2nd example about 64kB reserved
Heap summary for capabilities 0x00000400:
  At 0x3f800000 len 4194303 free 4071380 allocated 122880 min_free 4071380
    largest_free_block 4071380 alloc_blocks 1 free_blocks 1 total_blocks 2
  Totals:
    free 4071380 allocated 122880 min_free 4071380 largest_free_block 4071380
```

mahavirj, 2018-10-28:
@chegewara Is it possible for you to provide minimal application with `sdkconfig` that reliably reproduces this issue?

chegewara, 2018-10-28:
@mahavirj With sdkconfig there is no issue, i can provide copy of mine. There is problem with code, i am working on commercial app and app is connecting to amazon servers. I have to think how to do it.

sdkconfig (just one not working version):
https://gist.github.com/chegewara/a98ed02e54b8b5ecf204fea79b46645d

chegewara, 2018-10-28:
@mahavirj I think it may have something to do with this issue
https://github.com/espressif/esp-idf/issues/2228#issuecomment-408112657
when i am using option like @igrr is suggesting i cant make big allocations with calloc in spiram i have to use instead `heap_caps_calloc` instead with `MALLOC_CAP_SPIRAM`. If i use calloc (malloc probably too) then allocation is in internal memory.

negativekelvin, 2018-10-29:
Did you try disabling hw encryption

mahavirj, 2018-10-29:
> @mahavirj I think it may have something to do with this issue
#2228 (comment)
when i am using option like @igrr is suggesting i cant make big allocations with calloc in spiram i have to use instead heap_caps_calloc instead with MALLOC_CAP_SPIRAM. If i use calloc (malloc probably too) then allocation is in internal memory.

That will also depend on config option `SPIRAM_MALLOC_ALWAYSINTERNAL` which has default value of 16384, essentially means all allocations less than that size will happen from internal memory. 

Also for mbedTLS we are using `heap_caps_calloc` API itself, please refer at https://github.com/espressif/esp-idf/blob/bd90f534fb1cdf7d4406767f9067398d51a12305/components/mbedtls/port/esp_mem.c#L21

I suspect something else happening because of error `mbedtls_ssl_handshake returned -0x7780` from your log, let me once you can get something that can reproduce this issue.

chegewara, 2018-10-29:
> That will also depend on config option `SPIRAM_MALLOC_ALWAYSINTERNAL` which has default value of 16384, essentially means all allocations less than that size will happen from internal memory.
I meant really big allocations, like 100-200kB.

> Also for mbedTLS we are using `heap_caps_calloc` API itself, please refer at
> 
> [esp-idf/components/mbedtls/port/esp_mem.c](https://github.com/espressif/esp-idf/blob/bd90f534fb1cdf7d4406767f9067398d51a12305/components/mbedtls/port/esp_mem.c#L21)
> 
> Line 21 in [bd90f53](/espressif/esp-idf/commit/bd90f534fb1cdf7d4406767f9067398d51a12305)
> 
>  IRAM_ATTR void *esp_mbedtls_mem_calloc(size_t n, size_t size) 
> I suspect something else happening because of error `mbedtls_ssl_handshake returned -0x7780` from your log, let me know once you can get something that can reproduce this issue.

I will try to see whats going on here. Thanks.

chegewara, 2018-10-29:
> Did you try disabling hw encryption

Disabling hw encryption leads to many errors like this one, but thanks for suggestion:
```
/esp-idf/components/esp32/hwcrypto/aes.c:434:5: error: conflicting types for 'esp_aes_xts_setkey_enc'
 int esp_aes_xts_setkey_enc( mbedtls_aes_xts_context *ctx,
```

projectgus, 2018-10-30:
@chegewara 
> Disabling hw encryption leads to many errors like this one, but thanks for suggestion:

Thanks for reporting this, it's a bug in the current master branch. Will fix.

chegewara, 2018-11-18:
Thanks, will reopen if encounter any issues.

closed 2018-11-18 by chegewara