espressif/esp-idf #2643: WIFI_PROTOCOL_LR in APSTA mode?

skitsanos, 2018-10-29:
Is there any way to use WIFI_PROTOCOL_LR in APSTA mode?

I've tried to do the following:

```
ESP_ERROR_CHECK(esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_LR));
ESP_ERROR_CHECK(esp_wifi_set_protocol(WIFI_IF_AP, WIFI_PROTOCOL_LR));
```

and it fails on each line... Am i missing anything?

liuzfesp, 2018-11-06:
@skitsanos everything looks good on my side, could you let me know the return value of the API?

liuzfesp, 2018-11-06:
Thanks @skitsanos for pointing this bug, we will fix it.

skitsanos, 2018-11-06:
i have err code 0 that comes together with 

```
wifi: sta start fail
```

and

```
wifi: ap start fail
```

Maybe I'm doing things in a wrong order? Here is my func:

```
    tcpip_adapter_init();

    ESP_ERROR_CHECK(esp_event_loop_init(wifi_event_handler, NULL));
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg))

    ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));

    ESP_LOGI(TAG, "Initializing WIFI in dual-mode...");

    ESP_LOGI(TAG, "WIFI LR STA err code:%d", esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_LR));
    ESP_LOGI(TAG, "WIFI LR AP err code:%d", esp_wifi_set_protocol(WIFI_IF_AP, WIFI_PROTOCOL_LR));

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));

    wifi_config_t wifi_ap_config = {
            .ap = {
                    .ssid = ap_ssid,
                    .ssid_len = (uint8_t) strlen(ap_ssid),
                    .password = ap_password,
                    .max_connection=4,
                    .authmode = WIFI_AUTH_WPA_WPA2_PSK
            }
    };

    wifi_config_t wifi_sta_config = {
            .sta = {
                    .ssid = "",
                    .password = ""
            }
    };

    strcpy((char *) wifi_sta_config.sta.ssid, (char *) ssid);
    strcpy((char *) wifi_sta_config.sta.password, (char *) password);

    if (strlen(ap_password) == 0)
    {
        wifi_ap_config.ap.authmode = WIFI_AUTH_OPEN;
    }

    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_ap_config));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_sta_config));

    ESP_ERROR_CHECK(esp_wifi_start());

    xTaskCreate(task_espnow_init, "espnow task", 2048, NULL, 10, NULL);
```

if i dont set this protocol into LR, it all works just fine, i have proper APSTA mode and ESP-NOW functioning perfectly

liuzfesp, 2018-11-20:
HI skitsanos, just ignore following log:
*  wifi: sta start fail
*  wifi: ap start fail

The log is caused by API esp_wifi_set_protocol(), this API:
*  stop Wifi
*  configure the protocol
* start wifi
Since the WiFi is still not started via esp_wifi_start() when esp_wifi_set_protocol() is called, so the log is showed. Just ignore it.

We will remove this log in IDF v3.3

skitsanos, 2019-03-10:
Ok, after some while, i've revisited the issue ;) this time porting my whole framework into c++, so following code is c++

```cpp
Log::i("Starting intranet mode");
ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));

esp_err_t lr_result = esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_LR);
std::ostringstream ss;
ss << "WIFI LR STA err code: #" << lr_result
   << " (" << esp_err_to_name(lr_result) << ")";
Log::i(ss.str());
```

Seems, trick was to set WIFI_PROTOCOL_LR after i call esp_wifi_set_mode. So now it gives ESP_OK and not err:238 as it used to.

closed 2019-03-14 by Alvin1Zhang