espressif/esp-idf #2644: RMT IR reception while using ETH PHY (Waveshare LAN8720) results in RMT error: `mem_full` flag set, `wait` state

johannesmaibaum, 2018-10-30:
- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- Development Kit:      [none] Generic "NodeMCU-32"-Module
- Kit version (for WroverKit/PicoKit/DevKitC): [n/a]
- Core (if using chip or module): [ESP-WROOM32]
- IDF version: The one used in Arduino ESP 32 Core Release v1.0.0: https://github.com/espressif/arduino-esp32/releases/tag/1.0.0
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]
- Power Supply:         [USB and external 5V]

## Problem Description

Based on the ESP-IDF example code for the RMT IR reception/transmission (https://github.com/espressif/esp-idf/tree/ee0ebc1/examples/peripherals/rmt_nec_tx_rx) I have created an Arduino style library to send out and receive NEC IR messages. All works fine until I want to add a Waveshare LAN8720 PHY board to the mix (eventually, I want to have the option to switch between WiFi and Wired LAN for my project). As soon as I'm trying to enable the IR reception part, I'm getting a rmt: RMT[0] ERR message with a status word that, decoded according to info provided in espressif/esp-idf#1175 (comment), means:

- `mem_full` flag set
- `wait` state
- at `mem_waddr_ex = 0x40`

It works seamlessly when I'm using standard WiFi mode. Is the low-level ETH code not compatible with reception of IR messages using the RMT module at the same time?

Also, it's only the reception part which is not working when using the ETH PHY, i.e. IR sending works like a charm both with ETH and WiFi.

For reference, I have also created an ESP32.com forum thread for this issue (no answers yet over there): https://esp32.com/viewtopic.php?p=31964

And I have also posted an issue at https://github.com/espressif/arduino-esp32/issues/1993 (@lbernstone suggested to report here as well)

### Expected Behavior

IR reception should work no matter if I'm using WiFi or ETH, using any of the RMT channels.

### Actual Behavior

IR reception only works with WiFi. Using ETH I get rmt: RMT[0] ERR with status `0x14000040` reported (similar on all other RMT channels), i.e.:

- `mem_full` flag set
- `wait` state
- at `mem_waddr_ex = 0x40`

### Steps to reproduce

Wire up a Wireshark LAN8720 PHY module to your ESP-32. I used these instructions: https://sautter.com/blog/ethernet-on-esp32-using-lan8720/

Wire a 38kHz IR receiver module to input pin 33. I used this one: https://www.reichelt.de/ir-receiver-modules-tssp4p38-38-khz-tssp-4p38-p124102.html?GROUPID=3045&START=0&OFFSET=16&SID=95W9gQg6wQAT0AABD1A9w5b13fd67b01043de8d62a431e02213f5&LANGUAGE=EN&&r=1

Use the code from the below test repository I have created.

### Code to reproduce this issue

See: https://github.com/johannesmaibaum/ESP32-LAN8720-RMT-test

## Debug Logs from Arduino IDE (verbose level)

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 20 - ETH_START
ETH Started
Trying to enable IR reception:
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 22 - ETH_CONNECTED
ETH Connected
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 24 - ETH_GOT_IP
[D][WiFiGeneric.cpp:425] _eventCallback(): ETH IP: 192.168.5.234, MASK: 255.255.255.0, GW: 192.168.5.1
ETH MAC: 30:AE:A4:24:0A:73, IPv4: 192.168.5.234, FULL_DUPLEX, 100Mbps
E (9428) rmt: RMT[0] ERR
E (9428) rmt: status: 0x14000040
```

negativekelvin, 2018-10-30:
Do you have a pull-up resistor on the ir input?

negativekelvin, 2018-10-30:
Yes.  The Ethernet pins are right next to gpio33.

johannesmaibaum, 2018-11-01:
I added 10k to 3.3V from the IR input. Behaviour (and error message) is still the same as before, i.e. IR input doesn't work.

koobest, 2018-12-11:
Have you tried changing the defiation?
```
#define RMT_RX_ACTIVE_LEVEL 0  -----> define RMT_RX_ACTIVE_LEVEL 1
```

johannesmaibaum, 2018-12-11:
Hi @koobest,

I will try this next and let you know.

Yet, since it works with `#define RMT_ACTIVE_LEVEL 0` when using WiFi instead of ethernet, I don't see why changing this should make a difference.

negativekelvin, 2018-12-11:
@johannesmaibaum suggest you to use the rmt input filter like the example

```
rmt_rx.rx_config.filter_en = true;
rmt_rx.rx_config.filter_ticks_thresh = 100;
```

johannesmaibaum, 2018-12-12:
@negativekelvin I will try, once I find time. But I don't think that this will make a difference, since it works without the rmt input filter using WiFi. In fact, the IR receiver I am using (Vishay TSOP32238, https://www.vishay.com/docs/82489/tsop322.pdf) already has a built-in input filter.

negativekelvin, 2018-12-12:
The fact that tsop has a filter on the optical side won't help you if the noise is being coupled from the Ethernet interface via wiring or traces or breadboard.

johannesmaibaum, 2018-12-14:
> Have you tried changing the defiation?
> 
> ```
> #define RMT_RX_ACTIVE_LEVEL 0  -----> define RMT_RX_ACTIVE_LEVEL 1
> ```

Hi @ koobest,

Switching to `#define RMT_RX_ACTIVE_LEVEL 1` did not help.

johannesmaibaum, 2018-12-14:
Hi @negativekelvin ,

> @johannesmaibaum suggest you to use the rmt input filter like the example
> 
> ```
> rmt_rx.rx_config.filter_en = true;
> rmt_rx.rx_config.filter_ticks_thresh = 100;
> ```

> The fact that tsop has a filter on the optical side won't help you if the noise is being coupled from the Ethernet interface via wiring or traces or breadboard.

Thanks for the hint and the explanation. Turns out that turning on the input filter on the ESP made it work.

I consider this issue resolved and will close it now. Thanks for all your kind help! :)

closed 2018-12-14 by johannesmaibaum