espressif/esp-idf #2647: ESP32 Wifi Station mode failed to reconnect after first time

HuyKhac, 2018-10-31:
## Environment
- Development Kit:      [ESP32-Devkit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Core (if using chip or module): [ESP-WROOM-32]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): bd90f534fb1cdf7d4406767f9067398d51a12305 
- Development Env:		[Make]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description
Hi all,

Recently, I updated ESP IDF to latest commit and got the issue that ESP32 fails to reconnect to AP network after first time successful connection when I do "make flash_erase".
I use "esp-idf/examples/wifi/getting_started/station" example to test
So the steps are:
 - Make the example: make
 - Erase all flash:  make flash_erase
 - Flash the example and monitor:  make  flash monitor
  This time ESP32 can connect to my AP network
 - Press Reset button on Devkit 
  This time ESP32 cannot connect to my AP network, the reason is Authentication failed
  I try to reset many times, the result is 90% same

 - If I re-erase flash and run again, ESP32 can connect to my AP network but failed to reconnect after reset

I notice one difference is that when it runs with erased flash, it will do full RF calibration, while after reset, it does not do RF calibration.

### Expected Behavior
ESP32 should reconnect to AP network after reset reliably

### Actual Behavior
ESP32 cannot reconnect to AP network afte reset. Have to do erase_flash to have it connects one time

## Debug Logs

This is the log message:
[image]

HuyKhac, 2018-10-31:
Here is the sdkconfig file of the example
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/2532843/sdkconfig.txt)

HuyKhac, 2018-10-31:
I doubt if ESP32 forgets authentication after reset so that it will fails authentication when AP device still requires previous authentication based on MAC address of ESP32

HuyKhac, 2018-11-01:
Here is what I found:
- For successful connecting, my AP device performs 2 steps: authentication and association. While in failed connecting, my AP device only performs authentication. Maybe it still uses last association data but ESP32 does not handle that properly.  My work-around is to add a rule to AP device (hostapd.conf) to periodically poll all station devices and dis-associate which stations not respond; and in ESP32 side, I add a 20s delay before joining Wifi network. 
 I think normal station device like PCs or smartphones, if they resets suddenly, the up time is also more than 20s so that AP device can clean their association data. But with embedded devices like ESP32, the up time is too short.
 However, when I try with a commercial router device, ESP32 can reset and reconnect fine (without the work-around). So I cannot conclude whether ESP32 or my AP device need a fix

smnd96, 2018-11-02:
Surely the AP ( the OrangePi) as you mentioned:
-If you reset the Pi, things work
-If you connect to any commercial devices, the esp works.

liuzfesp, 2018-11-06:
Hi @HuyKhac, it's a bug in AP side. 

# keep alive between AP and STA
Generally, on STA side, it will start to send keep alive frame (NULL frame) to AP to keep the connection. On AP side, it will start a keep alive timer,  it will reset the timer once it receives any **data** frame from STA or receives keep alive frame from STA.  AP will disconnect the STA if it doesn't receive any data frames from STA after the keep alive timer expired.

# normal disconnect
The normal disconnect process is:
1. The station sends de-auth or dis-association frame to AP
2. The AP disconnect the station when it receives the de-auth or dis-association frame

# reset/power down scenario
The power off/reset scenario:
1. STA reset or power off
2. AP disconnects the STA after it's keep alive timer expired

# reconnect scenario
The correct reconnect process on AP side should be:
1. AP receives the AUTH request frame from STA
2. If the STA who sends the AUTH request frame  already connects to the AP,  the AP should firstly disconnect current connected STA and then accept the AUTH request frame.

According to your description, I think the problem is in AP side: the AP rejects WiFi reconnect from the connected STA.

Alvin1Zhang, 2018-11-07:
@HuyKhac Hi, would you mind if we close this issue for the time being? You are free to reopen this issue if the issue pops up again. Thanks.

ivan-baldin, 2019-03-31:
I'm having the same issue when I use android phone as AP. My laptop can connect and disconnect just fine. ESP32 however won't authenticate after the first connection.

Is there any workaround that DOESN'T involve any modifications to AP?

liuzfesp, 2019-04-01:
HI @ivan-baldin what's your IDF commit id? Could you help to check whether it exists in latest IDF?

liuzfesp, 2019-04-02:
@ivan-baldin do you call "git submodule update" to update the submodule after change your release from f03382b to 2f8b6cf?

ivan-baldin, 2019-04-02:
Yes, of course.

I've found workaround... kinda. https://forums.oneplus.com/threads/wifi-hotspot-not-working-after-android-9-pie-update.912563/ suggested enabling tethering hardware acceleration. After doing that, I'm getting first `WIFI_REASON_AUTH_FAIL` followed by `WIFI_REASON_CONNECTION_FAIL`, and success on 3rd `esp_wifi_connect`. Meh, I'll take it.

My guess is that problem is with android 9 or phone hardware. I tested using an old android 6.0 with no problems.